Ysurac/openmptcprouter #2974: Current stable UEFI image will not boot, image not recognized as UEFI device, other images boot as expected.

ioogithub, 2023-09-26:
## Expected Behavior
I expect OMR images will boot work as expected, especially the current stable image.

## Current Behavior
This image will not boot, it is not recognized as a UEFI device in bios:
[openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined-efi.img.gz](https://download.openmptcprouter.com/release/v0.59.1/x86_64/targets/x86/64/openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined-efi.img.gz)

## Possible Solution
Image needs to be remade, there is something nonstandard about it .
## More info
I have tried the following
- booting from 3 different usb devices- fail
- booting from m.2 internal sata drive- fail
- booting from m2. external sata enclosure- fail
- writing the image with dd, etcher and mkubs, same result: the image will not boot- fail
- I can boot various other live linux images successfully.
- tried troubleshooting with gpt, no luck here. If the bios doesn't recognize the image as a uefi device it will not boot.

I also tried the following:

1. recognized by the BIOS as UEFI and boot as expected but OMR says my wan devices do not support MPTCP (they do)
[openmptcprouter-v0.59.1-5.15-r0+20029-3c06a344e9-x86-64-generic-ext4-combined-efi.img.gz](https://download.openmptcprouter.com/release/v0.59.1-5.15/x86_64/targets/x86/64/openmptcprouter-v0.59.1-5.15-r0+20029-3c06a344e9-x86-64-generic-ext4-combined-efi.img.gz)

2. These images bring up a grub menu but OMR will not start with "error disk: 'hd0.gtp1' not found.
[openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-squashfs-combined.img.gz](https://download.openmptcprouter.com/release/v0.59.1/x86_64/targets/x86/64/openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-squashfs-combined.img.gz)

[openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-squashfs-combined-efi.img.gz](https://download.openmptcprouter.com/release/v0.59.1/x86_64/targets/x86/64/openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c-x86-64-generic-squashfs-combined-efi.img.gz)

3. This image boots but OMR glorytun crashes and I could not recover, even after rebooting 3 or 4 times.
[openmptcprouter-v0.60beta1-6.1-r0+23789-ce6ad123e7-x86-64-generic-ext4-combined-efi.img.gz](https://download.openmptcprouter.com/release/v0.60beta1-6.1/x86_64/targets/x86/64/openmptcprouter-v0.60beta1-6.1-r0+23789-ce6ad123e7-x86-64-generic-ext4-combined-efi.img.gz)

I really need the current boot image to work so I can test omr-bypass. All the other images I tried boot as expected except the current stable uefi image, it is not recognized by the bios. If the UEFI image does not boot, how can I write the generic ext4 image so it will boot? Please help.

## Specifications
  - OpenMPTCProuter platform: <x86_64
  - Hardware: generic 4 gig ethernet port jasper lake x86 box with AMI bios.

Ysurac, 2023-09-26:
There is a know issue on current stable release when another mass storage is available on the computer so it can't find the disk.
For the 6.1 image, I need log.

ioogithub, 2023-09-26:
> There is a know issue on current stable release when another mass storage is available on the computer so it can't find the disk. For the 6.1 image, I need log.

Thank you, I spent several hours on this and I could not figure out what the problem was. Is there  work around? I would like to boot a clean image on my router to test omr-bypass.

Why did the 5.15 version say that MPTCP was being blocked my wan devices were being blocked, they work find with other version is this a known issue as well?

For 6.1 yes I set up a new vps and booted the router with this image, everything went well until tun0 stopped, it kept saying invalid key, i rebooted 3 or 4 times but it was never able to reconnect to the vps again. I will post the log if I still have it.

Ysurac, 2023-09-26:
5.15 version need version using 5.15 kernel. In any case it's a deprecated version and will not be supported.
Did you check glorytun key ? Did you try to reboot the VPS ?

ioogithub, 2023-09-26:
yes it worked just fine for 1 hour, key was right then tun0 crashed and it never reconnected, even after reboot it can't reconnect. I worked on it for a few hours yesterday but I could not recover from this.

But anyway, right out of the box with no custom config I tested omr-bypass and it did not work and that is what i needed to work now. traceroute from router always goes to tunnel for first hop. I need it go go where I specified in the omr-bypass screen.

In real world, omr without bypass is limited in application due to IP addressed offered from VPS providers being blocked.

closed 2023-12-31 by github-actions[bot]