Ysurac/openmptcprouter #2975: WANs on native VLAN not working, system log filled with errors (v0.60beta1-6.1)

peterxxl, 2023-09-28:
## Expected Behavior
new install with vanilla config with minor changes (configure 3 WANs, timezone, LAN IP)
0.61 with new kernel on x86_64
VPS upgraded according to the tutorial

this is new machine I recently purchased from Amazon so I am trying it with the new version
[Amazon link to the machine](https://www.amazon.com/gp/product/B0BM3QY2P8/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

I am running the stable version on RPi4 for half a year now without hiccup - I have acquired the new x86_64 hardware to bump up the performance

## Current Behavior
limited testing, but experiencing this so far:

WANs don't work on native VLAN, neither DHCP not static IP. 
it only works when VLAN is anything but 1 (or empty - default), the unit has 6 physical ports

kernel log gets filled with hundreds of these messages
```
[ 1334.929371] MPTCP: addr_signal error, rm_addr=4
[ 1334.929468] MPTCP: addr_signal error, rm_addr=4
[ 1334.929509] MPTCP: addr_signal error, rm_addr=4
[ 1334.929550] MPTCP: addr_signal error, rm_addr=4
[ 1334.929590] MPTCP: addr_signal error, rm_addr=4
[ 1334.929646] MPTCP: addr_signal error, rm_addr=4
[ 1334.929686] MPTCP: addr_signal error, rm_addr=4
[ 1334.929726] MPTCP: addr_signal error, rm_addr=4
[ 1334.929778] MPTCP: addr_signal error, rm_addr=4
[ 1334.929817] MPTCP: addr_signal error, rm_addr=4
[ 1334.929884] MPTCP: addr_signal error, rm_addr=4
[ 1334.929942] MPTCP: addr_signal error, rm_addr=4
[ 1335.094446] MPTCP: addr_signal error, rm_addr=4
[ 1335.094585] MPTCP: addr_signal error, rm_addr=4
[ 1335.094635] MPTCP: addr_signal error, rm_addr=4
[ 1335.094680] MPTCP: addr_signal error, rm_addr=4
[ 1335.094764] MPTCP: addr_signal error, rm_addr=4
[ 1335.094805] MPTCP: addr_signal error, rm_addr=4
[ 1335.094866] MPTCP: addr_signal error, rm_addr=4
[ 1335.095119] MPTCP: addr_signal error, rm_addr=4
[ 1335.095167] MPTCP: addr_signal error, rm_addr=4
[ 1335.095348] MPTCP: addr_signal error, rm_addr=4
[ 1335.095390] MPTCP: addr_signal error, rm_addr=4
[ 1335.095592] MPTCP: addr_signal error, rm_addr=4
```

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta1-6.1 r0+23789-ce6
  - OpenMPTCProuter VPS version: 6.1.0-12-amd64 0.1029-test
  - OpenMPTCProuter VPS provider: Grupo ZGH
  - OpenMPTCProuter platform: x86_64
  - Country: Chile

Kernel Version：6.1.46
Model：YANLING YL-CLU6L-V1
Architecture：Intel(R) Core(TM) i3-10110U CPU @ 2.10GHz

Ysurac, 2023-09-28:
First, we can check the configuration: When you have VLAN set, what do you have, using SSH on the router, in ```uci show network``` and ```ip a``` ?
There is many ports on this computer, sometimes VLANs are used internally for each port of a internal switch so they can't be set again.

peterxxl, 2023-09-28:
Hello

Here are the results of those commands. 

eth1 to eth4 (WAN1 to WAN4) don't need VLAN (each is one cable with untagged traffic plugged in physical port on the device), but without it, it doesn't work, so I have to go thru switch first to tag the traffic on them.

`root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='8'
network.globals=globals
network.globals.ula_prefix='fd68:d97c:4c46::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_subflows='8'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_pm_type='0'
network.globals.mptcp_disable_initial_config='0'
network.globals.mptcp_force_multipath='1'
network.globals.mptcpd_enable='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.proto='static'
network.lan.ipaddr='10.2.0.1'
network.lan.netmask='255.255.255.0'
network.lan.device='eth0'
network.lan.ifname='eth0'
network.lan.metric='9'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.txqueuelen='2000'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.label='LAN'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan2=interface
network.wan2.device='eth2.52'
network.wan2.proto='dhcp'
network.wan2.ip4table='wan'
network.wan2.multipath='on'
network.wan2.defaultroute='0'
network.wan2.metric='11'
network.wan2.peerdns='0'
network.wan2.ipv6='0'
network.wan2.label='52'
network.wan3=interface
network.wan3.device='eth3.53'
network.wan3.proto='dhcp'
network.wan3.ip4table='wan'
network.wan3.multipath='on'
network.wan3.defaultroute='0'
network.wan3.metric='12'
network.wan3.peerdns='0'
[... 149 lines cut ...]
       valid_lft forever preferred_lft forever
167: eth2.52@eth2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN group default qlen 1000
    link/ether 00:e0:97:1d:4f:c6 brd ff:ff:ff:ff:ff:ff
168: eth3.53@eth3: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN group default qlen 1000
    link/ether 00:e0:97:1d:4f:c7 brd ff:ff:ff:ff:ff:ff
169: eth4.54@eth4: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN group default qlen 1000
    link/ether 00:e0:97:1d:4f:c8 brd ff:ff:ff:ff:ff:ff
174: eth1.51@eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN group default qlen 1000
    link/ether 00:e0:97:1d:4f:c5 brd ff:ff:ff:ff:ff:ff
189: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 500
    link/none 
    inet 10.255.255.2 peer 10.255.255.1/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::536f:c1ab:f5f3:ac0f/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
192: ifb4tun0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether be:80:b5:a4:9d:40 brd ff:ff:ff:ff:ff:ff
205: eth5.55@eth5: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN group default qlen 1000
    link/ether 00:e0:97:1d:4f:c9 brd ff:ff:ff:ff:ff:ff
206: eth5.56@eth5: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN group default qlen 1000
    link/ether 00:e0:97:1d:4f:c9 brd ff:ff:ff:ff:ff:ff
207: eth5.61@eth5: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN group default qlen 1000
    link/ether 00:e0:97:1d:4f:c9 brd ff:ff:ff:ff:ff:ff
208: eth5.62@eth5: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN group default qlen 1000
    link/ether 00:e0:97:1d:4f:c9 brd ff:ff:ff:ff:ff:ff`

Ysurac, 2023-09-28:
I don't really understand what you mean. It's not working with and without VLANs no ?
And why do you have 4 VLANs on eth5 ?

Ysurac, 2023-09-28:
I also see it's a I225-V, know to have many issues. Some are fixed by setting "Force link" on the interface, check https://github.com/Ysurac/openmptcprouter/issues/2584

peterxxl, 2023-09-28:
With VLANs (VLAN ID 2 or more) it seems to work... it doesn't work when I leave the VLAN field empty (default VLAN) or fill in VLAN 1

in this case it only shows the error in the log which I sent you
`[ 1334.929942] MPTCP: addr_signal error, rm_addr=4`

when I use native VLANs, the interface just goes up and down every 10 seconds or so and never connects

I am just plying with the configurations now, that's why there are 4 VLANs on eth5.... I have 8 internet lines, on rPi4 thy are all on 1 physical interface tagged as VLANs but the performance is not the best ... so the point of setting this device up is there will be 1 WAN per per physical port on ports eth1 till eth4 and the rest will be on VLANs similar to what I have on rPi4

peterxxl, 2023-09-28:
> I also see it's a I225-V, know to have many issues. Some are fixed by setting "Force link" on the interface, check #2584

I am reading thru this now... seems like the same issue....

Ysurac, 2023-09-28:
VLAN 1 is default VLAN of all ports.
I hope Force link will solve your issue... Else if you find another workaround tell me, I22x-V chipset is on a lot of mini PCs but it's broken by design (I22x not -V are fixed).

peterxxl, 2023-09-28:
So after few hours of testing, I have it running on my main network for real world stress test.
The config is same as posted earlier + the "Force Link"
Eth 0 is LAN
Eth 1 thru 4 are 4 WANs, each on native VLAN  with "Force Link" enabled
Eth 5 has 4 VLANs which are another 4 WANs and I have enabled "Force Link" on each just to be sure, even thou it seemed to work without it

The log is still full of `MPTCP: addr_signal error, rm_addr=4` but it seem to be working

My main issue now is that I don't see the performance gains I was hoping for as opposed to rPi4. 

The speeds on each WAN are quite variable, but usually between 150 and 250 Mbps on download and 20 to 50 Mbps on upload when tested directly. 

When running thru OMR, the max. I usually get is 120 down and 80 up. These result are comparable to rPi4 with just one LAN port with 9 VLANs on it.
I am thinking VPS might be the slow one here. 

I am still using OMR for now instead of using just one WAN ... mainly for redundancy and stability. I was not expecting 1 Gbps+ speeds, but at least 500 Mbps+.

I have tried autorate, but seems to be performing slightly better without it. 

How can I test my VPS speed? Is there command on the VPS to test against the same servers like the router to find out where the bottle neck is?

Any recommended settings to increase speed in the 0.60beta1 with 6.1 kernel version? I see lot of the settings I played with in the old release are gone now. Right now, everything is default. 

Thanks for all the work you put into this project... it's been a lifesaver in the past year or so...

peterxxl, 2023-09-29:
Another update

So the speeds were definitely problem of the VPS... I am on my 6th VPS now. 
I haven't found many great options in Chile, but in the end the one I am using now was actually [on the list](https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS). 
Netflix doesn't seem to be working at all and it's bit expensive, but the speeds can't compare with the previous ones I tried.

I am getting close to 600 Mbps down now on the new hardware... much better.

closed 2024-01-02 by github-actions[bot]