Ysurac/openmptcprouter #2977: No NIC's show up / driver issue

MrLimoAK, 2023-09-29:
Ok so I have a new unknow mfg x86 N5105 Jasper lake, with what I think are 2.5Gb NIC's.  Board is labeled FM107 N5105 JK 4L-V02 CBOX

I used the x64 efi and board boots fine just will not bring up and interfaces.  I used an Ubuntu iso and it will bring up interfaces and pass traffic.  If I attache a Belkin USB-C Ethernet it will bring up the 192.168.100.1 interface and I can setup LAN/WAN.  So the base OS is fully functional.

How can I identify my missing ethernet hardware on the motherboard, what can I do to assist with possibly helping to find a suitable driver(s).

Ysurac, 2023-09-29:
Please use template when you open an issue.

Did you check with 6.1 beta1 if it's supported or not ?

MrLimoAK, 2023-09-29:
GetHub is very confusing to me.  I can't find a supported HW list or a 6.1 beta download, or have the skill set to pull a specific version and compile it from scratch.  Done a couple of make apps before the GH really confuses me.  Sorry.

Ysurac, 2023-09-29:
It's a pinned issue (and also on openmptcprouter.com homepage): https://github.com/Ysurac/openmptcprouter/issues/2961
There is no HW list (too difficult to list all network card and all drivers available as package or already on the image...).

closed 2024-01-02 by github-actions[bot]