Ysurac/openmptcprouter #2978: Router Rebooting or restarting services every 30-60 seconds

jtiller-sw, 2023-09-29:
## Expected Behavior
OpenMPTCPRouter is currently not configured to bond connections. There is 1 WAN connection and 1 LAN connection.

## Current Behavior
Router is rebooting, or system precesses are restarting every 30 to 60 seconds.
Symptoms: 

- web gui reloads and logs user out
- ssh connections are dropped
- no connection to internet
-  Sever stops responding to ping

## Specifications

  - OpenMPTCProuter version: 0.59.1-54 r0
  - OpenMPTCProuter VPS version: NA
  - OpenMPTCProuter VPS provider:NA
  - OpenMPTCProuter platform: x86 Big IP 2000 by F5 Networks

jtiller-sw, 2023-09-29:
SYSLOG

Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.214979] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input1
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.215027] ACPI: Power Button [PWRF]
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.215771] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.236390] 00:01: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.257043] 00:02: ttyS1 at I/O 0x2f8 (irq = 3, base_baud = 115200) is a 16550A
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.257852] Non-volatile memory driver v1.3
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.257984] Linux agpgart interface v0.103
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.259462] loop: module loaded
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.259500] Guest personality initialized and is inactive
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.259546] VMCI host device registered (name=vmci, major=10, minor=63)
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.259552] Initialized host personality
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.259724] Loading iSCSI transport class v2.0-870.
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.259933] VMware PVSCSI driver - version 1.0.7.0-k
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.259962] hv_vmbus: registering driver hv_storvsc
Thu Sep 28 12:08:37 2023 kern.debug kernel: [    0.260365] ahci 0000:00:1f.2: version 3.0
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.260597] ahci 0000:00:1f.2: AHCI 0001.0300 32 slots 4 ports 3 Gbps 0x10 impl SATA mode
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.260605] ahci 0000:00:1f.2: flags: 64bit ncq sntf ilck pm led clo pmp pio slum part ems apst
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.261146] scsi host0: ahci
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.261350] scsi host1: ahci
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.261500] scsi host2: ahci
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.261609] scsi host3: ahci
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.261712] scsi host4: ahci
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.261777] ata1: DUMMY
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.261780] ata2: DUMMY
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.261782] ata3: DUMMY
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.261784] ata4: DUMMY
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.261791] ata5: SATA max UDMA/133 abar m2048@0xfe705000 port 0xfe705300 irq 28
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262029] VMware vmxnet3 virtual NIC driver - version 1.4.17.0-k-NAPI
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262050] hv_vmbus: registering driver hv_netvsc
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262056] Fusion MPT base driver 3.04.20
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262059] Copyright (c) 1999-2008 LSI Corporation
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262068] Fusion MPT SPI Host driver 3.04.20
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262083] Fusion MPT SAS Host driver 3.04.20
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262114] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262119] ehci-pci: EHCI PCI platform driver
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262266] ehci-pci 0000:00:1d.0: EHCI Host Controller
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262275] ehci-pci 0000:00:1d.0: new USB bus registered, assigned bus number 1
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.262296] ehci-pci 0000:00:1d.0: debug port 2
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.266286] ehci-pci 0000:00:1d.0: cache line size of 64 is not supported
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.266303] ehci-pci 0000:00:1d.0: irq 23, io mem 0xfe706000
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.289936] ehci-pci 0000:00:1d.0: USB 2.0 started, EHCI 1.00
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.290476] hub 1-0:1.0: USB hub found
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.290495] hub 1-0:1.0: 2 ports detected
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.290718] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.290729] ohci-pci: OHCI PCI platform driver
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.290750] ohci-platform: OHCI generic platform driver
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.290764] uhci_hcd: USB Universal Host Controller Interface driver
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.290823] usbcore: registered new interface driver usb-storage
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.290869] i8042: PNP: No PS/2 controller found.
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.290875] hv_vmbus: registering driver hyperv_keyboard
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.290922] rtc_cmos 00:03: RTC can wake from S4
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.291256] rtc_cmos 00:03: registered as rtc0
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.291265] rtc_cmos 00:03: alarms up to one month, y3k, 114 bytes nvram, hpet irqs
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.291769] intel_pstate: Intel P-state driver initializing
Thu Sep 28 12:08:37 2023 kern.info kernel: [    0.291967] sdhci: Secure Digital Host Controller Interface driver
[... cut ...]
Thu Sep 28 12:08:56 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Sep 28 12:08:56 2023 user.notice mptcp: Reloading mptcp config due to ifup of wan7 (eth7)
Thu Sep 28 12:08:57 2023 user.notice firewall: Reloading firewall due to ifup of wan8 (eth8)
Thu Sep 28 12:08:57 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Sep 28 12:08:57 2023 user.notice mptcp: Reloading mptcp config due to ifup of wan8 (eth8)

jtiller-sw, 2023-09-29:
KERNEL LOG

[    0.000000] Linux version 5.4.194 (runner@fv-az377-156) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+16594-ce92de8c8c)) #0 SMP Tue May 17 22:11:28 2022
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=5458006b-de35-5090-d950-ab64782a2c02 rootwait console=tty0 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009abff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009ac00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000df14ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000df150000-0x00000000df193fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000df194000-0x00000000df19afff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000df19b000-0x00000000df255fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000df256000-0x00000000df30efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000df30f000-0x00000000df30ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000df310000-0x00000000df352fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000df353000-0x00000000df7fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000021fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: F5 Networks C112/C112, BIOS 4.6.5 07/21/2015
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 1999.796 MHz processor
[    0.000494] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000496] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000502] last_pfn = 0x220000 max_arch_pfn = 0x400000000
[    0.000507] MTRR default type: uncachable
[    0.000508] MTRR fixed ranges enabled:
[    0.000509]   00000-9FFFF write-back
[    0.000510]   A0000-BFFFF uncachable
[    0.000511]   C0000-C3FFF write-protect
[    0.000512]   C4000-E7FFF uncachable
[    0.000512]   E8000-FFFFF write-protect
[    0.000513] MTRR variable ranges enabled:
[    0.000515]   0 base 000000000 mask E00000000 write-back
[    0.000516]   1 base 200000000 mask FE0000000 write-back
[    0.000517]   2 base 0E0000000 mask FE0000000 uncachable
[    0.000518]   3 disabled
[    0.000518]   4 disabled
[    0.000519]   5 disabled
[    0.000519]   6 disabled
[    0.000519]   7 disabled
[    0.000520]   8 disabled
[    0.000520]   9 disabled
[    0.001524] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002131] e820: update [mem 0xe0000000-0xffffffff] usable ==> reserved
[    0.002135] last_pfn = 0xdf800 max_arch_pfn = 0x400000000
[    0.013209] found SMP MP-table at [mem 0x000fd6f0-0x000fd6ff]
[    0.013883] ACPI: Early table checksum verification disabled
[    0.013887] ACPI: RSDP 0x00000000000F0490 000024 (v02 ALASKA)
[    0.013891] ACPI: XSDT 0x00000000DF195070 00005C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.013898] ACPI: FACP 0x00000000DF19A488 0000F4 (v04 ALASKA A M I    01072009 AMI  00010013)
[    0.013904] ACPI: DSDT 0x00000000DF195158 005330 (v02 ALASKA A M I    00000204 INTL 20051117)
[    0.013908] ACPI: FACS 0x00000000DF24DF80 000040
[    0.013911] ACPI: APIC 0x00000000DF19A580 000084 (v03 ALASKA A M I    01072009 AMI  00010013)
[    0.013915] ACPI: FPDT 0x00000000DF19A608 000044 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.013918] ACPI: SSDT 0x00000000DF19A650 000102 (v01 AMICPU PROC     00000001 MSFT 03000001)
[    0.013922] ACPI: MCFG 0x00000000DF19A758 00003C (v01 ALASKA A M I    01072009 MSFT 00000097)
[    0.013925] ACPI: HPET 0x00000000DF19A798 000038 (v01 ALASKA A M I    01072009 AMI. 00000005)
[    0.013929] ACPI: DMAR 0x00000000DF19A7D0 000070 (v01 ALASKA A M I    00000001 INTL 00000001)
[    0.013933] ACPI: Reserving FACP table memory at [mem 0xdf19a488-0xdf19a57b]
[    0.013934] ACPI: Reserving DSDT table memory at [mem 0xdf195158-0xdf19a487]
[    0.013936] ACPI: Reserving FACS table memory at [mem 0xdf24df80-0xdf24dfbf]
[... 787 lines cut ...]
[   10.142716] usbserial: USB Serial support registered for IPWireless converter
[   10.146509] usbcore: registered new interface driver option
[   10.146546] usbserial: USB Serial support registered for GSM modem (1-port)
[   10.147647] usbcore: registered new interface driver qcserial
[   10.147678] usbserial: USB Serial support registered for Qualcomm USB modem
[   10.150555] usbcore: registered new interface driver rt2800usb
[   10.153250] usbcore: registered new interface driver rtl8192cu
[   10.154931] usbcore: registered new interface driver ath9k_htc
[   10.156011] kmodloader: done loading kernel modules from /etc/modules.d/*
[   12.830141] ixgbe 0000:01:00.0: registered PHC device on eth9
[   12.953881] 8021q: adding VLAN 0 to HW filter on device eth9
[... cut ...]
[   13.251238] 8021q: adding VLAN 0 to HW filter on device eth7
[   13.300925] 8021q: adding VLAN 0 to HW filter on device eth8
[   15.651689] ixgbe 0000:01:00.0 eth9: NIC Link is Up 1 Gbps, Flow Control: None
[   15.651905] IPv6: ADDRCONF(NETDEV_CHANGE): eth9: link becomes ready
[   15.687975] tg3 0000:09:00.0 eth0: Link is up at 1000 Mbps, full duplex
[   15.687984] tg3 0000:09:00.0 eth0: Flow control is on for TX and on for RX
[   15.687988] tg3 0000:09:00.0 eth0: EEE is disabled
[   15.688004] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

jtiller-sw, 2023-09-29:
Ping Log

64 bytes from 192.168.100.1: icmp_seq=75 ttl=64 time=0.606 ms
64 bytes from 192.168.100.1: icmp_seq=76 ttl=64 time=0.877 ms
64 bytes from 192.168.100.1: icmp_seq=77 ttl=64 time=1.089 ms
64 bytes from 192.168.100.1: icmp_seq=78 ttl=64 time=0.855 ms
64 bytes from 192.168.100.1: icmp_seq=79 ttl=64 time=1.217 ms
64 bytes from 192.168.100.1: icmp_seq=80 ttl=64 time=1.188 ms
64 bytes from 192.168.100.1: icmp_seq=81 ttl=64 time=1.065 ms
64 bytes from 192.168.100.1: icmp_seq=82 ttl=64 time=1.078 ms
64 bytes from 192.168.100.1: icmp_seq=83 ttl=64 time=1.021 ms
64 bytes from 192.168.100.1: icmp_seq=84 ttl=64 time=0.893 ms
64 bytes from 192.168.100.1: icmp_seq=85 ttl=64 time=0.717 ms
64 bytes from 192.168.100.1: icmp_seq=86 ttl=64 time=0.887 ms
64 bytes from 192.168.100.1: icmp_seq=87 ttl=64 time=0.715 ms
64 bytes from 192.168.100.1: icmp_seq=88 ttl=64 time=0.762 ms
64 bytes from 192.168.100.1: icmp_seq=89 ttl=64 time=0.836 ms
64 bytes from 192.168.100.1: icmp_seq=90 ttl=64 time=0.920 ms
64 bytes from 192.168.100.1: icmp_seq=91 ttl=64 time=0.688 ms
Request timeout for icmp_seq 92
Request timeout for icmp_seq 93
Request timeout for icmp_seq 94
Request timeout for icmp_seq 95
Request timeout for icmp_seq 96
Request timeout for icmp_seq 97
Request timeout for icmp_seq 98
Request timeout for icmp_seq 99
Request timeout for icmp_seq 100
Request timeout for icmp_seq 101
Request timeout for icmp_seq 102
Request timeout for icmp_seq 103
Request timeout for icmp_seq 104
Request timeout for icmp_seq 105
Request timeout for icmp_seq 106
Request timeout for icmp_seq 107
Request timeout for icmp_seq 108
Request timeout for icmp_seq 109
Request timeout for icmp_seq 110
Request timeout for icmp_seq 111
Request timeout for icmp_seq 112
Request timeout for icmp_seq 113
Request timeout for icmp_seq 114
Request timeout for icmp_seq 115
Request timeout for icmp_seq 116
Request timeout for icmp_seq 117
Request timeout for icmp_seq 118
Request timeout for icmp_seq 119
Request timeout for icmp_seq 120
Request timeout for icmp_seq 121
Request timeout for icmp_seq 122
64 bytes from 192.168.100.1: icmp_seq=123 ttl=64 time=1.551 ms
64 bytes from 192.168.100.1: icmp_seq=124 ttl=64 time=0.705 ms
64 bytes from 192.168.100.1: icmp_seq=125 ttl=64 time=0.868 ms
64 bytes from 192.168.100.1: icmp_seq=126 ttl=64 time=0.949 ms
64 bytes from 192.168.100.1: icmp_seq=127 ttl=64 time=0.900 ms
64 bytes from 192.168.100.1: icmp_seq=128 ttl=64 time=0.974 ms
64 bytes from 192.168.100.1: icmp_seq=129 ttl=64 time=0.851 ms
64 bytes from 192.168.100.1: icmp_seq=130 ttl=64 time=0.838 ms
64 bytes from 192.168.100.1: icmp_seq=131 ttl=64 time=1.082 ms
64 bytes from 192.168.100.1: icmp_seq=132 ttl=64 time=0.858 ms
64 bytes from 192.168.100.1: icmp_seq=133 ttl=64 time=0.948 ms
64 bytes from 192.168.100.1: icmp_seq=134 ttl=64 time=0.503 ms
64 bytes from 192.168.100.1: icmp_seq=135 ttl=64 time=0.780 ms
64 bytes from 192.168.100.1: icmp_seq=136 ttl=64 time=0.801 ms
64 bytes from 192.168.100.1: icmp_seq=137 ttl=64 time=0.956 ms
64 bytes from 192.168.100.1: icmp_seq=138 ttl=64 time=0.969 ms
Request timeout for icmp_seq 139
Request timeout for icmp_seq 140
Request timeout for icmp_seq 141
Request timeout for icmp_seq 142
Request timeout for icmp_seq 143
Request timeout for icmp_seq 144
Request timeout for icmp_seq 145
Request timeout for icmp_seq 146
Request timeout for icmp_seq 147
Request timeout for icmp_seq 148
Request timeout for icmp_seq 149
Request timeout for icmp_seq 150
Request timeout for icmp_seq 151
Request timeout for icmp_seq 152
Request timeout for icmp_seq 153
Request timeout for icmp_seq 154
Request timeout for icmp_seq 155
Request timeout for icmp_seq 156
Request timeout for icmp_seq 157
Request timeout for icmp_seq 158
Request timeout for icmp_seq 159
Request timeout for icmp_seq 160
Request timeout for icmp_seq 161
Request timeout for icmp_seq 162
Request timeout for icmp_seq 163
Request timeout for icmp_seq 164
Request timeout for icmp_seq 165
Request timeout for icmp_seq 166
Request timeout for icmp_seq 167
Request timeout for icmp_seq 168
Request timeout for icmp_seq 169
64 bytes from 192.168.100.1: icmp_seq=170 ttl=64 time=0.886 ms
64 bytes from 192.168.100.1: icmp_seq=171 ttl=64 time=0.875 ms
64 bytes from 192.168.100.1: icmp_seq=172 ttl=64 time=0.938 ms
64 bytes from 192.168.100.1: icmp_seq=173 ttl=64 time=0.790 ms
64 bytes from 192.168.100.1: icmp_seq=174 ttl=64 time=1.082 ms
64 bytes from 192.168.100.1: icmp_seq=175 ttl=64 time=0.941 ms
64 bytes from 192.168.100.1: icmp_seq=176 ttl=64 time=1.171 ms
64 bytes from 192.168.100.1: icmp_seq=177 ttl=64 time=0.794 ms
64 bytes from 192.168.100.1: icmp_seq=178 ttl=64 time=0.920 ms
64 bytes from 192.168.100.1: icmp_seq=179 ttl=64 time=0.764 ms
64 bytes from 192.168.100.1: icmp_seq=180 ttl=64 time=0.705 ms
64 bytes from 192.168.100.1: icmp_seq=181 ttl=64 time=0.717 ms
64 bytes from 192.168.100.1: icmp_seq=182 ttl=64 time=0.838 ms
64 bytes from 192.168.100.1: icmp_seq=183 ttl=64 time=0.823 ms
64 bytes from 192.168.100.1: icmp_seq=184 ttl=64 time=0.797 ms
64 bytes from 192.168.100.1: icmp_seq=185 ttl=64 time=0.863 ms
64 bytes from 192.168.100.1: icmp_seq=186 ttl=64 time=0.987 ms
64 bytes from 192.168.100.1: icmp_seq=187 ttl=64 time=0.866 ms
64 bytes from 192.168.100.1: icmp_seq=188 ttl=64 time=0.954 ms

Ysurac, 2023-09-29:
Seems that there is a problem with ethernet ports using igb. Try to set "Force link" on all interfaces in Network->Interfaces.

fareign, 2023-10-01:
USE static IP

closed 2024-01-04 by github-actions[bot]