Ysurac/openmptcprouter #2981: xray and ss-rust can not work

fareign, 2023-10-02:
## NO ss 2022 key
ss rust not work.

## xray not run

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: snapshot 0930
  - OpenMPTCProuter VPS version: 0.1029test the lastest upstream
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:  x86_64 
  - Country: cn

Ysurac, 2023-10-02:
Latest VPS script must be installed from openmptcprouter (not on github repo yet) and a new installation on router must be made or keys force requested via wizard checkbox.

fareign, 2023-10-02:
where is the latest vps script .I want do a test.thanks

Ysurac, 2023-10-02:
Always on https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

fareign, 2023-10-02:
> Always on https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

thanks. I used beta link

fareign, 2023-10-02:
ss-rust can get key can work now,but SS cant get it. I fill it by manaul it stil cant work.

fareign, 2023-10-02:
@Ysurac  multipath seem cant work

Ysurac, 2023-10-02:
working on my side but possible I forgot something...
For both ? What do you have on ```uci show shadowsocks-rust```, ```ps l | grep sslocal``` and ```/var/etc/shadowsocks-rust/ss_redir.hi1.json``` on the router via SSH ?

fareign, 2023-10-02:
### every mode cant work on multipath (even v2ray and xray).

shadowsocks-rust.hi1=ss_redir
shadowsocks-rust.hi1.server='sss0'
shadowsocks-rust.hi1.local_address='::'
shadowsocks-rust.hi1.local_port='1100'
shadowsocks-rust.hi1.mode='tcp_and_udp'
shadowsocks-rust.hi1.timeout='1000'
shadowsocks-rust.hi1.fast_open='0'
shadowsocks-rust.hi1.verbose='0'
shadowsocks-rust.hi1.syslog='1'
shadowsocks-rust.hi1.reuse_port='1'
shadowsocks-rust.hi1.mptcp='1'
shadowsocks-rust.hi1.ipv6_first='1'
shadowsocks-rust.hi1.no_delay='1'
shadowsocks-rust.hi2=ss_redir
shadowsocks-rust.hi2.server='sss0'
shadowsocks-rust.hi2.local_address='0.0.0.0'
shadowsocks-rust.hi2.local_port='1100'
shadowsocks-rust.hi2.mode='tcp_and_udp'
shadowsocks-rust.hi2.timeout='1000'
shadowsocks-rust.hi2.fast_open='1'
shadowsocks-rust.hi2.verbose='0'
shadowsocks-rust.hi2.syslog='1'
shadowsocks-rust.hi2.reuse_port='1'
shadowsocks-rust.hi2.mptcp='1'
shadowsocks-rust.hi2.ipv6_first='1'
shadowsocks-rust.hi2.no_delay='1'
shadowsocks-rust.hi2.disabled='1'
shadowsocks-rust.ss_rules=ss_rules
shadowsocks-rust.ss_rules.disabled='0'
shadowsocks-rust.ss_rules.redir_tcp='hi1'
shadowsocks-rust.ss_rules.src_default='forward'
shadowsocks-rust.ss_rules.dst_default='forward'
shadowsocks-rust.ss_rules.local_default='forward'
shadowsocks-rust.ss_rules.server='sss0'
shadowsocks-rust.sss0=server
shadowsocks-rust.sss0.disabled='1'
shadowsocks-rust.sss0.server='8.210.233.224'
shadowsocks-rust.sss0.server_port='65280'
shadowsocks-rust.sss0.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss0.obfs='0'
shadowsocks-rust.sss0.s_shadowsocks_method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss0.password='R+Gs48aY2TF1tROfNcgHKvQRWOW7KMLYJtOwJUNh9Os=:DOdeSnGiSsbl8gDFz7wnj3QsOhGfAibur1eVEMwu+uM='
shadowsocks-rust.dns=ss_tunnel
shadowsocks-rust.dns.disabled='1'
shadowsocks-rust.dns.mode='tcp_and_udp'
shadowsocks-rust.dns.server='sss0'
shadowsocks-rust.dns.local_port='5353'
shadowsocks-rust.dns.tunnel_address='8.8.8.8:53'
shadowsocks-rust.tracker_sss0=ss_local
shadowsocks-rust.tracker_sss0.server='sss0'
shadowsocks-rust.tracker_sss0.local_address='127.0.0.1'
shadowsocks-rust.tracker_sss0.local_port='1111'
shadowsocks-rust.tracker_sss0.mode='tcp_and_udp'
shadowsocks-rust.tracker_sss0.timeout='600'
shadowsocks-rust.tracker_sss0.fast_open='0'
shadowsocks-rust.tracker_sss0.syslog='0'
shadowsocks-rust.tracker_sss0.reuse_port='1'
shadowsocks-rust.tracker_sss0.mptcp='1'
shadowsocks-rust.tracker_sss0.verbose='0'

S     0 28816 15458  1220   800 pts0  10:17 00:00:00 grep sslocal
S     0 29938     1 17052  5636 0:0   10:12 00:00:00 /usr/bin/sslocal -c /var/etc/shadowsocks-rust/ss_local.tracker_sss0.json
S     0 29939     1 25364  8312 0:0   10:12 00:00:01 /usr/bin/sslocal -c /var/etc/shadowsocks-rust/ss_redir.hi1.json

root@OpenMPTCProuter:~# cat /var/etc/shadowsocks-rust/ss_redir.hi1.json
{
        "method": "2022-blake3-aes-256-gcm",
        "server": "8.210.233.224",
        "server_port": 65280,
        "password": "R+Gs48aY2TF1tROfNcgHKvQRWOW7KMLYJtOwJUNh9Os=:DOdeSnGiSsbl8gDFz7wnj3QsOhGfAibur1eVEMwu+uM=",
        "use_syslog": true,
        "ipv6_first": true,
        "fast_open": false,
        "reuse_port": true,
        "no_delay": true,
        "mptcp": true,
        "protocol": "redir",
        "local_address": "::",
        "local_port": 1100,
        "mode": "tcp_and_udp",
        "timeout": 1000
}

Ysurac, 2023-10-03:
Can you try to install again VPS and router ? This should be fixed now.

closed 2024-01-07 by github-actions[bot]