Ysurac/openmptcprouter #2983: Xray dont work on the VPS

vempire-ghost, 2023-10-02:
## Expected Behavior

## Current Behavior
Xray dont work on the VPS

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)
On the router side the log show this.
Oct  2 21:07:16 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1)

On the vps side when i run this command **xray -version** I get this 
xray: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by xray)
xray: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by xray)

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta1-5.4 r0+16862-170d9 **Snaptshot from 01 october.**
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test **Updated with the teste-server script**
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64
  - Country: Brazil

Ysurac, 2023-10-03:
What is the OS and version on the VPS ?

vempire-ghost, 2023-10-03:
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

Ysurac, 2023-10-03:
It's compiled and tested on Debian 12, I think it's the problem. I will try.

vempire-ghost, 2023-10-03:
Great, its better setup a new vps directly with 12 or can I upgrade my actual 11 to 12?

vempire-ghost, 2023-10-03:
Xray is working great in the Debian 12, but the OMR is not showing the proxy traffic when using xray in the dashboard or in the Network Overview.

Ysurac, 2023-10-04:
Should be fixed now.

vempire-ghost, 2023-10-04:
Just need run the vps test script again?

What is better for 5.4 kernel, Debian 11 or 12? Or didn't matter?

Ysurac, 2023-10-04:
Tests are made on Debian 12 with kernel 6.1.
I will maybe force Debian 12 upgrade for all (kernel 5.4 and 6.1) instead of maintaining several Debian versions...

closed 2024-01-08 by github-actions[bot]