Ysurac/openmptcprouter #2988: ss-rust cant work ,can link from vps but restart sevice frequently

fareign, 2023-10-04:
Oct  4 02:12:26 OpenMPTCProuter user.notice OMR-VPS: Get config from server vps...
Oct  4 02:12:26 OpenMPTCProuter user.notice OMR-VPS: V2ray restart...
Oct  4 02:12:27 OpenMPTCProuter user.notice OMR-VPS: Xray restart...
Oct  4 02:12:27 OpenMPTCProuter user.notice OMR-VPS: Glorytun restart...
Oct  4 02:12:28 OpenMPTCProuter user.notice OMR-VPS: Glorytun UDP restart...
Oct  4 02:12:28 OpenMPTCProuter user.notice OMR-VPS: OpenVPN restart...
Oct  4 02:12:28 OpenMPTCProuter daemon.err openvpn(omr)[25693]: event_wait : Interrupted system call (fd=-1,code=4)
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[25693]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1500 0 10.255.252.10 10.255.252.9 init
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[25693]: net_addr_ptp_v4_del: 10.255.252.10 dev tun0
Oct  4 02:12:28 OpenMPTCProuter daemon.notice ttyd[28049]: [2023/10/04 02:12:28:0833] N: rops_handle_POLLIN_netlink: DELADDR
Oct  4 02:12:28 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Oct  4 02:12:28 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Oct  4 02:12:28 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Oct  4 02:12:28 OpenMPTCProuter daemon.notice ttyd[28049]: [2023/10/04 02:12:28:0840] N: rops_handle_POLLIN_netlink: DELADDR
Oct  4 02:12:28 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[25693]: /usr/libexec/openvpn-hotplug down omr tun0 1500 0 10.255.252.10 10.255.252.9 init
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[25693]: SIGTERM[hard,] received, process exiting
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: OpenVPN 2.6.5 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: library versions: OpenSSL 3.0.11 19 Sep 2023, LZO 2.10
Oct  4 02:12:28 OpenMPTCProuter daemon.warn openvpn(omr)[32251]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Oct  4 02:12:28 OpenMPTCProuter daemon.warn openvpn(omr)[32251]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: TCP/UDP: Preserving recently used remote address: [AF_INET]47.243.110.137:65301
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: Attempting to establish TCP connection with [AF_INET]47.243.110.137:65301
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: TCP connection established with [AF_INET]47.243.110.137:65301
Oct  4 02:12:28 OpenMPTCProuter daemon.warn openvpn(omr)[32251]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: TCPv4_CLIENT link local: (not bound)
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: TCPv4_CLIENT link remote: [AF_INET]47.243.110.137:65301
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: [server] Peer Connection Initiated with [AF_INET]47.243.110.137:65301
Oct  4 02:12:28 OpenMPTCProuter daemon.warn openvpn(omr)[32251]: sitnl_route_save: rtnl: can't get ifname for index 0: No such device or address (errno=6)
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: TUN/TAP device tun0 opened
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: net_iface_mtu_set: mtu 1500 for tun0
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: net_iface_up: set tun0 up
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: net_addr_ptp_v4_add: 10.255.252.10 peer 10.255.252.9 dev tun0
Oct  4 02:12:28 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Oct  4 02:12:28 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Oct  4 02:12:28 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Oct  4 02:12:28 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Oct  4 02:12:28 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: /usr/libexec/openvpn-hotplug up omr tun0 1500 0 10.255.252.10 10.255.252.9 init
Oct  4 02:12:28 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Oct  4 02:12:29 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Oct  4 02:12:29 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Oct  4 02:12:30 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Oct  4 02:12:33 OpenMPTCProuter daemon.warn openvpn(omr)[32251]: sitnl_send: rtnl: generic error (-17): File exists
Oct  4 02:12:33 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: Initialization Sequence Completed
Oct  4 02:12:40 OpenMPTCProuter daemon.err uhttpd[4880]: error: couldn't find the ModemManager process in the bus
Oct  4 02:12:40 OpenMPTCProuter user.notice OMR-VPS: Get config from server vps...
Oct  4 02:12:40 OpenMPTCProuter user.notice OMR-VPS: V2ray restart...
Oct  4 02:12:41 OpenMPTCProuter user.notice OMR-VPS: Xray restart...
Oct  4 02:12:41 OpenMPTCProuter user.notice OMR-VPS: Glorytun restart...
Oct  4 02:12:41 OpenMPTCProuter user.notice OMR-VPS: Glorytun UDP restart...
Oct  4 02:12:41 OpenMPTCProuter user.notice OMR-VPS: OpenVPN restart...
Oct  4 02:12:41 OpenMPTCProuter daemon.err openvpn(omr)[32251]: event_wait : Interrupted system call (fd=-1,code=4)
Oct  4 02:12:41 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1500 0 10.255.252.10 10.255.252.9 init
Oct  4 02:12:41 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: net_addr_ptp_v4_del: 10.255.252.10 dev tun0
Oct  4 02:12:41 OpenMPTCProuter daemon.notice ttyd[28049]: [2023/10/04 02:12:41:8922] N: rops_handle_POLLIN_netlink: DELADDR
Oct  4 02:12:41 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Oct  4 02:12:41 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Oct  4 02:12:41 OpenMPTCProuter daemon.notice ttyd[28049]: [2023/10/04 02:12:41:8929] N: rops_handle_POLLIN_netlink: DELADDR
Oct  4 02:12:41 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Oct  4 02:12:42 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Oct  4 02:12:42 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: /usr/libexec/openvpn-hotplug down omr tun0 1500 0 10.255.252.10 10.255.252.9 init
Oct  4 02:12:42 OpenMPTCProuter daemon.notice openvpn(omr)[32251]: SIGTERM[hard,] received, process exiting
Oct  4 02:12:42 OpenMPTCProuter daemon.notice openvpn(omr)[5742]: OpenVPN 2.6.5 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Oct  4 02:12:42 OpenMPTCProuter daemon.notice openvpn(omr)[5742]: library versions: OpenSSL 3.0.11 19 Sep 2023, LZO 2.10
Oct  4 02:12:42 OpenMPTCProuter daemon.warn openvpn(omr)[5742]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Oct  4 02:12:42 OpenMPTCProuter daemon.warn openvpn(omr)[5742]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Oct  4 02:12:42 OpenMPTCProuter daemon.notice openvpn(omr)[5742]: TCP/UDP: Preserving recently used remote address: [AF_INET]47.243.110.137:65301
Oct  4 02:12:42 OpenMPTCProuter daemon.notice openvpn(omr)[5742]: Attempting to establish TCP connection with [AF_INET]47.243.110.137:65301
Oct  4 02:12:42 OpenMPTCProuter daemon.notice openvpn(omr)[5742]: TCP connection established with [AF_INET]47.243.110.137:65301
Oct  4 02:12:42 OpenMPTCProuter daemon.warn openvpn(omr)[5742]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)
[... 446 lines cut ...]
Oct  4 02:15:02 OpenMPTCProuter daemon.notice openvpn(omr)[579]: [server] Peer Connection Initiated with [AF_INET]47.243.110.137:65301
Oct  4 02:15:02 OpenMPTCProuter daemon.warn openvpn(omr)[579]: sitnl_route_save: rtnl: can't get ifname for index 0: No such device or address (errno=6)
Oct  4 02:15:02 OpenMPTCProuter daemon.notice openvpn(omr)[579]: TUN/TAP device tun0 opened
[... cut ...]
Oct  4 02:15:04 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Oct  4 02:15:08 OpenMPTCProuter daemon.warn openvpn(omr)[579]: sitnl_send: rtnl: generic error (-17): File exists
Oct  4 02:15:08 OpenMPTCProuter daemon.notice openvpn(omr)[579]: Initialization Sequence Completed

## Specifications

  - OpenMPTCProuter version: 0.61 snap
  - OpenMPTCProuter VPS version: 0.1029test upstream 10.03
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country: cn

Ysurac, 2023-10-04:
This seems more a problem on OMR-VPS side. It's maybe fixed with latest changes.

fareign, 2023-10-06:
> This seems more a problem on OMR-VPS side. It's maybe fixed with latest changes.

@Ysurac I test on my rock 3a.find ss-rust cant work on srt.

Ysurac, 2023-10-06:
@Kalimeiro the ip6tables error was another issue, fixed. For Shadowsocks-rust restart this should also be fixed.
I'm still making some tests/fixes on Shadowsocks-Rust/Shadowsocks-go and some changes will also be added to XRay.

Ysurac, 2023-10-14:
https vps script and github VPS script are in sync. You need to use ```http://www.openmptcprouter.com/server-test/debian-x86_64.sh``` for the snapshot (server-test replaced by server-beta for the beta, and server for the current stable release).

I fixed the build of x86_64 release (Github Runner base image was updated and there was not enough free disk space anymore for x86_64 image compilation...)

closed 2024-01-17 by github-actions[bot]