Ysurac/openmptcprouter #2991: Can I use a VPN server provided by a VPN provider instead of a VPS?

j7b3y, 2023-10-07:
No VPS was found that meets your requirements.
Instead, I thought I could use a VPN for traffic aggregation using the router configuration provided by my VPN provider (such as ProtonVPN).
I tried using regular Openwrt, but each Wireguard interface only wants to use one WAN port, so I gave up.
Is this actually possible?

j7b3y, 2023-10-07:
Although many VPSs have high-speed communication, they have problems such as the amount of data transferred per month.
If you could use a VPN server as an alternative, it would be very nice for people who have a VPN contract, including myself.

Ysurac, 2023-10-07:
It's not possible: traffic on each wan need to be aggregated and this is done on VPS/server side. This would be maybe possible to have some fun if VPN providers support MPTCP, but it's not yet the case.
Some VPS have really high amount that is not a real problem or you can use omr-bypass to not use aggregation (so VPS) for some traffic.

j7b3y, 2023-10-07:
Thank you for your detailed reply.
Please continue to do your best in development. I'm rooting for you.

closed 2023-10-07 by j7b3y