Ysurac/openmptcprouter #2992: changing colors on bandwidth graphs

adamb, 2023-10-07:
## Current Behavior

I like looking at this graph http://192.168.100.1/cgi-bin/luci/admin/network/mptcp/bandwidth but I have trouble telling the colors apart for wan1 and wan2.  

Is there a way to change these colors?  I looked in the /www directory and I can't find any place to edit the html.  It looks like the UI might generated by luci code, /www/cgi-bin/luci perhaps?

I looked through the git repo and I can't find this file and I can't find the `luci.sgi.cgi.run()` call either.    I'm guessing this code is pulled in from another git repo so it's not in the OpenMPTCProuter repo maybe?

I'm a coder and would be interested in figuring out how to add an option to select the colors here.  

I was also thinking of trying to put the bandwidth graph on my phone so I can see it anytime. Maybe create an iOS widget that shows it.  The stats would need to be exposed somehow or pushed to a cloud server where they could be loaded.  Maybe an S3 bucket....  

If I could get some pointers on how to modify the UI, I would be happy submit a pull request once I got something working. 

## Specifications

  - OpenMPTCProuter version: [v0.59.1-5.4 r0+16594-ce92de8c8c](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter platform: RPI4

adamb, 2023-10-08:
I found it... `/usr/lib/lua/luci/view/mptc/multipath.htm` was able to edit this file and see different colors in the bandwidth view.

I see it actually comes from the feeds repo: `luci-app-mptcp/luasrc/view/mptcp/multipath.htm`

Still trying to grok how these fit together, but I'm getting there.

ioogithub, 2023-10-16:
@adamb where you able to fix this problem? You identified two files which one should I make the change to?

I found a bug. I swapped eth1 and eth2 cabled. The graph changed from green and blue to blue and dark blue but the legend did not update so it is still showing eth1 = green and eth2 = blue. It is confusing now. I want to fix it.

adamb, 2023-10-17:
I was able to change the colors by editing the file `/usr/lib/lua/luci/view/mptc/multipath.htm`

But you're right. The color setting seems to set it for the first item in the legend.  The second one in the graph is a variation of the color for the first one. Not the color in the legend.  Still trying to figure that one out.  I think it might be in the styling for the graph.  It uses shades of the first color, not the color set in the legend, which is confusing and for me, hard to see.  I'm a little color blind.

Ysurac, 2023-10-17:
There is an opacity defined for the graph, so color change when multiples graph overlaid. Maybe this is what confuse you.

ioogithub, 2023-10-17:
> There is an opacity defined for the graph, so color change when multiples graph overlaid. Maybe this is what confuse you.

I don't understand, on the graph I posted it shows wan2 (starlink) green. On the actual chart it does two different blue colors. There is no green. How can I fix this?

ioogithub, 2023-10-17:
> I was able to change the colors by editing the file `/usr/lib/lua/luci/view/mptc/multipath.htm`

Do you recall where in that file to update? I looked though it but it was overwhelming, I didn't see how to update it. Ideally I would like to update the graph back to green so it matches the legend, the way it was originally.

ioogithub, 2023-10-17:
/usr/lib/lua/luci/view/mptcp/multipath.htm

```
function stringToColour(str) {
		if(str == "total")
			return "OrangeRed";
		if(str.substring(0, 4) == "wan")
			return "FireBrick";
		if(str.substring(0, 4) == "wan1")
			return "DeepSkyBlue";
		if(str.substring(0, 4) == "wan2")
			return "SeaGreen";
		if(str.substring(0, 4) == "wan3")
			return "PaleGreen";
		if(str.substring(0, 4) == "wan4")
			return "PowderBlue";
		if(str.substring(0, 4) == "wan5")
			return "Salmon";
		if(str.substring(0, 4) == "wan6")
			return "LightGreen";
		if(str.substring(0, 4) == "wan7")
			return "PaleTurquoise";
```
This means wan2 should be SeaGreen but my graph is showing wan2 "PowderBlue" after switching swaping wan1 and wan2 cables. PowderBlue is for wan4, I do not have any wan3. How to change it back to seagreen, this is what it should be.

closed 2024-01-20 by github-actions[bot]