Ysurac/openmptcprouter #2995: Unbound stop working when OMR start.

vempire-ghost, 2023-10-10:
## Expected Behavior
Unbound works.

## Current Behavior
service stopped (unbound 1.17.1)

## Context (Environment)
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info: service stopped (unbound 1.17.1).
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info: server stats for thread 0: 169 queries, 153 answers from cache, 16 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info: server stats for thread 0: requestlist max 6 avg 1.375 exceeded 0 jostled 0
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info: average recursion processing time 0.459704 sec
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info: histogram of recursion processing times
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info: [25%]=0.209715 median[50%]=0.393216 [75%]=0.714573
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info: lower(secs) upper(secs) recursions
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info:    0.008192    0.016384 1
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info:    0.131072    0.262144 5
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info:    0.262144    0.524288 4
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info:    0.524288    1.000000 5
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [6360:0] info:    1.000000    2.000000 1
Oct 10 21:51:23 OpenMPTCProuter daemon.notice unbound: [22788:0] notice: init module 0: validator
Oct 10 21:51:23 OpenMPTCProuter daemon.notice unbound: [22788:0] notice: init module 1: iterator
Oct 10 21:51:23 OpenMPTCProuter daemon.info unbound: [22788:0] info: start of service (unbound 1.17.1).
Oct 10 21:51:24 OpenMPTCProuter daemon.info unbound: [22788:0] info: generate keytag query _ta-4f66. NULL IN

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta1-5.4 r0+16862-170d9 (Snapshot from 07/10/2023)
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64
  - Country: Brazil

vempire-ghost, 2023-10-12:
Only on boot, its normally do it?

Ysurac, 2023-10-12:
I was thinking about dnsmasq...
I will check why unbound restart, I should restart it only when down. Do you have more log before the "service stopped" ?

closed 2024-01-16 by github-actions[bot]