Ysurac/openmptcprouter #2996: OMR does not failover when master wan connection is down, loses connection to VPS and goes to direct output mode.

ioogithub, 2023-10-14:
## Expected Behavior
OMR needs to have failover capabilities. On the home page it states: " Always up with connection and server failover ". Failover and redundancy is important.

In a system with wan1 and wan2, wan2 is master:
- Disconnect wan1 - OMR stays in contact with vps, green on everything but wan1
- Disconnect wan2- OMR falls back to direct output mode (after around 5-10 minutes).

## Current Behavior
- Disconnect wan1 - OMR stays in contact with vps, green on everything but wan1
- Disconnect wan2- OMR stays in contact with vps, green on everything but wan2

## Possible Solution
Possible problem with the routing table either post-tracking or omr-tracker isn't setting the routes correctly only for the case where the master wan goes down. The issue is that OMR can't ping the IP address of the VPS if the master wan is disconnected, even though there is one good wan connection available.

## Steps to Reproduce the Problem

  1. Create a 2 wan system in aggregate with wan2 as the master, v2ray is proxy and glorytun as VPN. Status page shows everything green.
  2. logread -f on omr and ping vpsip and ping any website www.ibm.com
  3. Simulate the failure of wan2 by pulling the ethernet cable from the router.
  5. omr reacts, changes routes, but pings to vps are all lost.
  6. after some time omr-tracker omr will switch to direct access mode, bypassing the VPS. This is a problem.

I repeated this test 2x and 1x using dsvpn for proxy (rather than glorytun). Same result, OMR loses connection to VPS after master wan goes down and enters direct connection mode. It never recovers until master wan comes back online.

I repeated this test 3x disconnecting non-master wan, it always works as expected. OMR maintains connection to VPS.

## Context (Environment)
One of the main advantages of using OMR is for port forwarding. If OMR can't maintain a connection to the VPS and falls back to direct output mode, then access to the network can be lost.

If OMR can maintain a connection when the non-master wan fails it should also be able to maintain a connection with the master wan fails. It is an issue with setting up the routing correctly.

## Specifications
    OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
    OpenMPTCProuter VPS version: VPS 0.1028
    OpenMPTCProuter VPS provider:
    OpenMPTCProuter platform: x86_64)

Ysurac, 2023-10-15:
I'm not able to reproduce on latest snapshot.
What is the routing table (```ip r```) before and after ?

vempire-ghost, 2023-10-16:
I also noticed this issue in the recent snapshots. However, OMR doesn't revert to direct output, but in the logs, it keeps showing that it can't establish contact with the VPS. On the status screen, there's a yellow message that it couldn't recover the VPS keys. Nevertheless, it seems like the aggregation continues to work despite these warnings.

Ysurac, 2023-10-16:
Yes, it's then fixed via omr-tracker that remove the faulting route. Even if there is some warning this shouldn't impact anything, only direct communication from router to VPS (for API change, and latency displayed in status page).

vempire-ghost, 2023-10-16:
Would it be possible to temporarily switch the Master connection to another WAN until the one that's offline starts working again, to avoid these warnings?

Ysurac, 2023-10-16:
It's already the case, delay can be defined in omr-tracker.
I can add the detection of the "linkdown" event, but it's only when you unplug ethernet port. Not sure this happen so often in real usage but faster to detect than ping.

vempire-ghost, 2023-10-16:
Here, it always happens that whenever OMR fails to ping the VPS, even if the connection seems to be normal, if OMR can't ping the VPS, it keeps giving these errors until it can ping successfully again.
At least from the errors, it doesn't seem like it switched the master connection, because it only starts reporting successful contact with the VPS again when the original master connection can communicate with the VPS.

vempire-ghost, 2023-10-18:
Today my master connections its having alot of packet losses, the log show this.
Oct 17 23:09:13 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:09:26 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, server API answer on xxxxx.ddns.net)
Oct 17 23:09:40 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Oct 17 23:10:10 OpenMPTCProuter user.notice OMR-VPS: Setting server config error, try again
Oct 17 23:11:34 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, server API answer on xxxxx.ddns.net)
Oct 17 23:12:18 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:12:34 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:12:52 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:13:08 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, server API answer on xxxxx.ddns.net)
Oct 17 23:13:22 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, server API answer on xxxxx.ddns.net)
Oct 17 23:15:34 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:15:51 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:16:05 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, server API answer on xxxxx.ddns.net)
Oct 17 23:17:09 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:18:06 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, server API answer on xxxxx.ddns.net)
Oct 17 23:18:22 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:18:38 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:18:53 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:19:09 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:19:25 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:19:43 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:20:00 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:20:18 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:20:31 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, server API answer on xxxxx.ddns.net)
Oct 17 23:20:48 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:21:00 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, server API answer on xxxxx.ddns.net)
Oct 17 23:21:15 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:21:35 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:21:51 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:22:07 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:22:19 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, server API answer on xxxxx.ddns.net)
Oct 17 23:22:35 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:22:51 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 17 23:23:09 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
[... cut ...]
Oct 18 01:21:27 OpenMPTCProuter user.notice OMR-VPS: Setting server config error, try again
Oct 18 01:21:46 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, no server API answer on xxxxx.ddns.net)
Oct 18 01:22:01 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on xxxxx.ddns.net, server API answer on xxxxx.ddns.net)

They simple try to ping from the master until they can, and never change to other WAN thats is working fine.

vempire-ghost, 2023-10-30:
Did you manage to look into this issue? Whenever I start OMR and the master connection has no internet, I have to go into the settings and manually change the master to another WAN, or else OMR never manages to connect to the VPS and gets stuck in this loop of errors in my log above.

ioogithub, 2023-10-30:
> Did you manage to look into this issue?

For me, no it is an open issue. All of my issues are open issues, I don't think any of them have been resolved. In fact it just happened again now. omr-tracker thinks wan1 is down, I am running a parallel ping and it never misses. Omr kills the interface, the tun0 restarts, the mptcp script removes the routes and it never recovers from here. For the master wan, if I am lucky it falls back to direct output, if not, omr-tracker will fight and bringing the wan lines up and down indefinitely. I went 24 hours with omr-tracker killing the lines every few minutes. It always gets caught in a loop of up and down.

Sadly, the only way I know how to recover from these issue is to reboot omr. The scripts do not seem  stable enough yet to recover from these sort of problems automatically. I can't reproduce what omr-tracker is doing either to try and find out the root cause of the problems.

For example, wan2 is good in my case and omr should be able to failover to it is there is a problem with wan1. It never does.

I am fairly certain this is root cause of one of my problem:

```
Mon Oct 30 18:40:38 2023 user.notice post-tracking-post-tracking: wan1 (eth1) switched off because check error and ping from wan1 error (1.1.1.1,1.0.0.1,1.1.1.1)
Mon Oct 30 18:40:38 2023 user.notice post-tracking-post-tracking: Delete default route to vpsip via wan1 dev eth1
```

Once omr does this, even once the router cannot recover. It needs to be rebooted. I have been trying for 2 years now to get a stable omr router. I've tried two different platforms (armx64 and x86) I have replaced all of the original equipment, tried usb NICs, dedicated Intel NICs, replaced all the cables, tested 4 different vps providers and maybe a dozen installs.

But at the end of the day a router needs to work and I can't seem to get omr working for more than a few hours without a problem that requires manual human intervention. I have really tried.

Ysurac, 2023-10-31:
@vempire-ghost Can you give me the result of ```ifstatus wan1``` (if it's wan1 defined as master) and ```ps l | grep omr-tracker``` ?
It's seems that for some reason the connection is not detected as down, so routes are not removed and you can't reach VPS at start.

vempire-ghost, 2023-11-01:
> @vempire-ghost Can you give me the result of `ifstatus wan1` (if it's wan1 defined as master) and `ps l | grep omr-tracker` ? It's seems that for some reason the connection is not detected as down, so routes are not removed and you can't reach VPS at start.

root@OpenMPTCProuter:~# ifstatus wan1
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 986,
        "l3_device": "eth1",
        "proto": "dhcp",
        "device": "eth1",
        "updated": [
                "addresses",
                "routes",
                "data"
        ],
        "metric": 8,
        "dns_metric": 0,
        "delegation": true,
        "ipv4-address": [
                {
                        "address": "192.168.0.50",
                        "mask": 24
                }
        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [

        ],
        "route": [

        ],
        "dns-server": [

        ],
        "dns-search": [

        ],
        "neighbors": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [
                        {
                                "target": "0.0.0.0",
                                "mask": 0,
                                "nexthop": "192.168.0.1",
                                "source": "192.168.0.50/32"
                        }
                ],
                "dns-server": [
                        "208.67.222.222",
                        "208.67.220.220"
                ],
                "dns-search": [

                ],
                "neighbors": [

                ]
        },
        "data": {
                "dhcpserver": "192.168.0.1",
                "leasetime": 4294967295
        }
}

root@OpenMPTCProuter:~# ps l | grep omr-tracker
S     0   603     1  1512  1008 0:0   22:57 00:00:00 /bin/sh /bin/omr-tracker wan2
S     0  1885     1  1368  1064 0:0   22:57 00:00:00 /bin/sh /bin/omr-tracker wan2
S     0  1886     1  1368  1160 0:0   22:57 00:00:00 /bin/sh /bin/omr-tracker wan3
S     0  1887     1  1368  1076 0:0   22:57 00:00:00 /bin/sh /bin/omr-tracker WAN4
S     0  1888     1  1204   908 0:0   22:57 00:00:00 /bin/sh /bin/omr-tracker-xray
S     0  3297  1885  1512  1008 0:0   22:57 00:00:00 /bin/sh /bin/omr-tracker wan2
S     0  5452  1887  1380   884 0:0   23:07 00:00:00 /bin/sh /bin/omr-tracker WAN4
S     0  5819  1886  1380   928 0:0   23:07 00:00:00 /bin/sh /bin/omr-tracker wan3
R     0  5935 29472  1176   432 pts0  23:07 00:00:00 grep omr-tracker
S     0  9483     1  1512  1060 0:0   22:53 00:00:00 /bin/sh /bin/omr-tracker wan2
S     0 12629     1  1512  1008 0:0   22:57 00:00:00 /bin/sh /bin/omr-tracker wan2
S     0 15284     1  1512  1052 0:0   22:53 00:00:00 /bin/sh /bin/omr-tracker wan2
S     0 25989     1  1512  1052 0:0   22:55 00:00:00 /bin/sh /bin/omr-tracker wan2

Ysurac, 2023-11-02:
So omr-tracker is not launched on wan1, no idea why ...
What is the result of ```uci show network.wan1``` via SSH on the router ?

vempire-ghost, 2023-11-02:
> uci show network.wan1

root@OpenMPTCProuter:~# uci show network.wan1
network.wan1=interface
network.wan1.device='eth1'
network.wan1.ip4table='wan'
network.wan1.multipath='master'
network.wan1.defaultroute='0'
network.wan1.addlatency='0'
network.wan1.peerdns='0'
network.wan1.metric='9'
network.wan1.proto='dhcp'
network.wan1.ipv6='0'

Ysurac, 2023-11-02:
All seems to be ok.
What do you have in ```uci show omr-tracker``` ?

vempire-ghost, 2023-11-02:
root@OpenMPTCProuter:~# uci show omr-tracker
omr-tracker.defaults=defaults
omr-tracker.defaults.enabled='1'
omr-tracker.defaults.hosts='4.2.2.1' '8.8.8.8' '80.67.169.12' '8.8.4.4' '9.9.9.9' '1.0.0.1' '114.114.115.115' '1.2.4.8' '80.67.169.40' '114.114.114.114' '1.1.1.1'
omr-tracker.defaults.hosts6='2606:4700:4700::1111' '2606:4700:4700::1001' '2620:fe::fe' '2620:fe::9' '2001:4860:4860::8888' '2001:4860:4860::8844'
omr-tracker.defaults.timeout='2'
omr-tracker.defaults.tries='3'
omr-tracker.defaults.interval='2'
omr-tracker.defaults.interval_tries='1'
omr-tracker.defaults.wait_test='0'
omr-tracker.defaults.server_http_test='0'
omr-tracker.defaults.restart_down='0'
omr-tracker.defaults.mail_alert='0'
omr-tracker.defaults.type='ping'
omr-tracker.proxy=proxy
omr-tracker.proxy.enabled='1'
omr-tracker.proxy.hosts='1.0.0.1' '212.27.48.10' '198.27.92.1' '151.101.129.164' '77.88.55.77' '1.1.1.1' '74.82.42.42' '198.41.212.162'
omr-tracker.proxy.timeout='10'
omr-tracker.proxy.tries='3'
omr-tracker.proxy.wait_test='0'
omr-tracker.proxy.interval_tries='1'
omr-tracker.proxy.interval='5'

Ysurac, 2023-11-02:
All is ok here too.
What is the result of ```/etc/init.d/omr-tracker restart``` and then a ```ps l | grep omr-tracker``` ? if there is no wan1 entry in the result, what do you have in ```logread``` ?
Maybe something make omr-tracker for wan1 crash... no idea what... Maybe I can add a check and start it again when this happen.

vempire-ghost, 2023-11-02:
root@OpenMPTCProuter:~# ps l | grep omr-tracker
S     0  2902     1  1368  1156 0:0   15:23 00:00:00 /bin/sh /bin/omr-tracker wan1
S     0  2903     1  1368  1156 0:0   15:23 00:00:00 /bin/sh /bin/omr-tracker wan2
S     0  2904     1  1368  1088 0:0   15:23 00:00:00 /bin/sh /bin/omr-tracker wan3
S     0  2905     1  1368  1148 0:0   15:23 00:00:00 /bin/sh /bin/omr-tracker wan4
S     0  2906     1  1204   988 0:0   15:23 00:00:00 /bin/sh /bin/omr-tracker-xray
S     0  6579  2905  1380   924 0:0   15:23 00:00:00 /bin/sh /bin/omr-tracker wan4
S     0  7476  2903  1380   928 0:0   15:23 00:00:00 /bin/sh /bin/omr-tracker wan2
S     0  7911  2902  1380   916 0:0   15:23 00:00:00 /bin/sh /bin/omr-tracker wan1
R     0  8022 10166  1168   496 pts0  15:23 00:00:00 grep omr-tracker

A few days ago, a problem started where when I start OMR, it connects normally, but after a few minutes, it begins to give these errors in the log. Established connections continue to work, but it can't establish any new connections through the proxy until I change the MASTER to another WAN.

Nov  2 13:11:15 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on VPS, no server API answer on VPS)
Nov  2 13:11:29 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 198.41.212.162, 1.0.0.1, 212.27.48.10)
Nov  2 13:11:36 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on VPS, no server API answer on VPS)
Nov  2 13:11:44 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Nov  2 13:11:52 OpenMPTCProuter daemon.err omr-tracker-xray[14864]: ping: VPS: Try again
Nov  2 13:11:52 OpenMPTCProuter daemon.info omr-tracker-xray: Server (VPS) seems down, no answer to ping
Nov  2 13:12:06 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on VPS, no server API answer on VPS)
Nov  2 13:12:08 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 198.41.212.162, 1.0.0.1, 212.27.48.10, 198.27.92.1)

Ysurac, 2023-11-07:
Script to restart omr-tracker when missing on a connection is in the last snapshot. Not yet tested.

vempire-ghost, 2023-11-08:
Using the latest snapshot, OMR enters a looping behavior with this log for a few minutes until it stabilizes and stops restarting the proxy. Is this the expected behavior?

Nov  8 03:25:04 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 1.0.0.1)
Nov  8 03:25:04 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps up (my vps ip), set it back
Nov  8 03:25:05 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Nov  8 03:25:05 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Nov  8 03:25:05 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Nov  8 03:25:05 OpenMPTCProuter user.notice xray: add rules
Nov  8 03:25:05 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov  8 03:25:05 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov  8 03:25:06 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov  8 03:25:06 OpenMPTCProuter user.notice xray: Rules UP
Nov  8 03:25:06 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov  8 03:25:06 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov  8 03:25:07 OpenMPTCProuter user.notice OMR-VPS: Restart XRay...
Nov  8 03:25:07 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Nov  8 03:25:07 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov  8 03:25:07 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Nov  8 03:25:07 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Nov  8 03:25:07 OpenMPTCProuter user.notice xray: add rules
Nov  8 03:25:07 OpenMPTCProuter daemon.info xray[20496]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.1 linux/amd64)
Nov  8 03:25:07 OpenMPTCProuter daemon.info xray[20496]: A unified platform for anti-censorship.
Nov  8 03:25:07 OpenMPTCProuter daemon.info xray[20496]: 2023/11/08 03:25:07 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Nov  8 03:25:07 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov  8 03:25:07 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov  8 03:25:08 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov  8 03:25:08 OpenMPTCProuter user.notice xray: Rules UP
Nov  8 03:25:08 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov  8 03:25:08 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov  8 03:25:09 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov  8 03:25:15 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps up (my vps ip), set it back
Nov  8 03:25:15 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Nov  8 03:25:15 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Nov  8 03:25:15 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Nov  8 03:25:15 OpenMPTCProuter user.notice xray: add rules
Nov  8 03:25:15 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov  8 03:25:15 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov  8 03:25:16 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov  8 03:25:16 OpenMPTCProuter user.notice xray: Rules UP
Nov  8 03:25:16 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov  8 03:25:16 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov  8 03:25:16 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov  8 03:25:16 OpenMPTCProuter daemon.info xray[27571]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.1 linux/amd64)
Nov  8 03:25:16 OpenMPTCProuter daemon.info xray[27571]: A unified platform for anti-censorship.
Nov  8 03:25:16 OpenMPTCProuter daemon.info xray[27571]: 2023/11/08 03:25:16 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Nov  8 03:26:07 OpenMPTCProuter user.notice omr-tracker: Launching...
Nov  8 03:26:18 OpenMPTCProuter user.notice omr-tracker: Launched
Nov  8 03:26:19 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 1.0.0.1)
Nov  8 03:26:21 OpenMPTCProuter user.notice OMR-VPS: Restart XRay...
Nov  8 03:26:21 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Nov  8 03:26:21 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Nov  8 03:26:21 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Nov  8 03:26:21 OpenMPTCProuter user.notice xray: add rules
Nov  8 03:26:21 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov  8 03:26:21 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov  8 03:26:22 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov  8 03:26:22 OpenMPTCProuter user.notice xray: Rules UP
Nov  8 03:26:22 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov  8 03:26:22 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov  8 03:26:22 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov  8 03:26:22 OpenMPTCProuter daemon.info xray[7646]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.1 linux/amd64)
Nov  8 03:26:22 OpenMPTCProuter daemon.info xray[7646]: A unified platform for anti-censorship.
Nov  8 03:26:22 OpenMPTCProuter daemon.info xray[7646]: 2023/11/08 03:26:22 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Nov  8 03:26:26 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps up (my vps ip), set it back
Nov  8 03:26:29 OpenMPTCProuter user.notice omr-tracker: Launching...
Nov  8 03:26:40 OpenMPTCProuter user.notice omr-tracker: Launched
Nov  8 03:26:40 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov  8 03:26:41 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov  8 03:26:41 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 1.0.0.1)

vempire-ghost, 2023-11-08:
Also after I change the VPS IP, the OMR star to show this in the logs
Nov  8 04:00:20 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on muvps.ddns.net)
Nov  8 04:00:29 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164)
Nov  8 04:00:39 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on myvps.ddns.net)
Nov  8 04:00:49 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Nov  8 04:00:59 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (old vps ip)

Than after the dns is updated start to show this in the logs:
Nov  8 04:04:15 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (**new vps ip**)
Nov  8 04:04:23 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Nov  8 04:04:36 OpenMPTCProuter daemon.info omr-tracker-xray: Server (**old vps ip**) seems down, no answer to ping
Nov  8 04:04:40 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on myvps.ddns.net)

For some reason they see the new IP but xray still try to connect to the old one until I restart the OMR.

Ysurac, 2023-11-08:
I found the issue, this will be fixed in next snapshot.

vempire-ghost, 2023-11-10:
Tried the new snapshot from today, the looping bug seens to be fixed, but OMR continue to never recover when I change de VPS IP.
Nov 10 23:44:14 OpenMPTCProuter daemon.info omr-tracker-xray: Server (OLD IP) seems down, no answer to ping
Nov 10 23:44:18 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on myvps.ddns.net)
Nov 10 23:44:29 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162)
Nov 10 23:44:35 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on myvps.ddns.net)
Nov 10 23:44:45 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Nov 10 23:44:49 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (NEW IP)
Nov 10 23:44:58 OpenMPTCProuter daemon.info omr-tracker-xray: Server (OLD IP) seems down, no answer to ping
Nov 10 23:45:02 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on myvps.ddns.net)
Nov 10 23:45:13 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162, 1.0.0.1)
Nov 10 23:45:18 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on myvps.ddns.net)
Nov 10 23:45:29 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Nov 10 23:45:38 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (NEW IP)
Nov 10 23:45:42 OpenMPTCProuter daemon.info omr-tracker-xray: Server (OLD IP) seems down, no answer to ping
Nov 10 23:45:45 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on myvps.ddns.net)
Nov 10 23:45:57 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162, 1.0.0.1, 212.27.48.10)
Nov 10 23:46:02 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on myvps.ddns.net)
Nov 10 23:46:12 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Nov 10 23:46:25 OpenMPTCProuter daemon.info omr-tracker-xray: Server (OLD IP) seems down, no answer to ping
Nov 10 23:46:28 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (NEW IP)
Nov 10 23:46:29 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on myvps.ddns.net)
Nov 10 23:46:40 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162, 1.0.0.1, 212.27.48.10, 198.27.92.1)
Nov 10 23:46:46 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on myvps.ddns.net, no server API answer on myvps.ddns.net)

They find the new IP but cannot connect until I save/apply in the http://192.168.100.1/cgi-bin/luci/admin/system/openmptcprouter page.

Ysurac, 2023-11-11:
Can you give me the result of ```ip r```, ```ping <newvpsip>``` and ```traceroute <newvpsip>``` via SSH from the router ?

vempire-ghost, 2023-11-11:
root@OpenMPTCProuter:~# ip r
default metric 1 
        nexthop via 192.168.0.1 dev eth1 weight 100 
        nexthop via 192.168.0.1 dev eth2 weight 1 
        nexthop via 192.168.2.1 dev eth3 weight 1 
        nexthop via 192.168.0.1 dev eth4 weight 1 
default via 192.168.0.1 dev eth1 metric 9 
default via 192.168.0.1 dev eth2 metric 10 
default via 192.168.2.1 dev eth3 metric 11 
default via 192.168.0.1 dev eth4 metric 12 
35.215.194.152 metric 1 
        nexthop via 192.168.0.1 dev eth1 weight 100 
        nexthop via 192.168.0.1 dev eth2 weight 1 
        nexthop via 192.168.2.1 dev eth3 weight 1 
        nexthop via 192.168.0.1 dev eth4 weight 1 
127.0.0.0/8 dev lo proto static scope link metric 7 
192.168.0.0/24 dev eth1 scope link metric 9 
192.168.0.0/24 dev eth2 scope link metric 10 
192.168.0.0/24 dev eth4 scope link metric 12 
192.168.2.0/24 dev eth3 scope link metric 11 

root@OpenMPTCProuter:~# ping 35.215.194.XXX
PING 35.215.194.XXX (35.215.194.XXX) 56(84) bytes of data.
64 bytes from 35.215.194.XXX: icmp_seq=1 ttl=50 time=15.9 ms
64 bytes from 35.215.194.XXX: icmp_seq=2 ttl=50 time=17.5 ms
64 bytes from 35.215.194.XXX: icmp_seq=3 ttl=50 time=17.4 ms
64 bytes from 35.215.194.XXX: icmp_seq=4 ttl=50 time=16.6 ms
64 bytes from 35.215.194.XXX: icmp_seq=5 ttl=50 time=18.1 ms
64 bytes from 35.215.194.XXX: icmp_seq=6 ttl=50 time=15.1 ms
64 bytes from 35.215.194.XXX: icmp_seq=7 ttl=50 time=15.7 ms
64 bytes from 35.215.194.XXX: icmp_seq=8 ttl=50 time=17.9 ms
64 bytes from 35.215.194.XXX: icmp_seq=9 ttl=50 time=17.6 ms
64 bytes from 35.215.194.XXX: icmp_seq=10 ttl=50 time=22.6 ms
64 bytes from 35.215.194.XXX: icmp_seq=11 ttl=50 time=15.5 ms
64 bytes from 35.215.194.XXX: icmp_seq=12 ttl=50 time=15.0 ms
64 bytes from 35.215.194.XXX: icmp_seq=13 ttl=50 time=15.6 ms
64 bytes from 35.215.194.XXX: icmp_seq=14 ttl=50 time=15.8 ms
64 bytes from 35.215.194.XXX: icmp_seq=15 ttl=50 time=17.0 ms
64 bytes from 35.215.194.XXX: icmp_seq=16 ttl=50 time=15.2 ms
64 bytes from 35.215.194.XXX: icmp_seq=17 ttl=50 time=15.8 ms
64 bytes from 35.215.194.XXX: icmp_seq=18 ttl=50 time=17.0 ms
64 bytes from 35.215.194.XXX: icmp_seq=19 ttl=50 time=16.6 ms
64 bytes from 35.215.194.XXX: icmp_seq=20 ttl=50 time=15.7 ms
64 bytes from 35.215.194.XXX: icmp_seq=21 ttl=50 time=17.0 ms
64 bytes from 35.215.194.XXX: icmp_seq=22 ttl=50 time=15.3 ms
64 bytes from 35.215.194.XXX: icmp_seq=23 ttl=50 time=17.8 ms
64 bytes from 35.215.194.XXX: icmp_seq=24 ttl=50 time=17.3 ms
64 bytes from 35.215.194.XXX: icmp_seq=25 ttl=50 time=15.6 ms
64 bytes from 35.215.194.XXX: icmp_seq=26 ttl=50 time=17.9 ms
64 bytes from 35.215.194.XXX: icmp_seq=27 ttl=50 time=17.3 ms
64 bytes from 35.215.194.XXX: icmp_seq=28 ttl=50 time=18.2 ms
64 bytes from 35.215.194.XXX: icmp_seq=29 ttl=50 time=16.9 ms
64 bytes from 35.215.194.XXX: icmp_seq=30 ttl=50 time=18.5 ms
64 bytes from 35.215.194.XXX: icmp_seq=31 ttl=50 time=22.8 ms

root@OpenMPTCProuter:~# traceroute 35.215.194.XXX
traceroute to 35.215.194.XXX (35.215.194.XXX), 30 hops max, 46 byte packets
 1  192.168.0.1 (192.168.0.1)  0.953 ms  0.615 ms  0.547 ms
 2  192.168.2.1 (192.168.2.1)  1.100 ms  0.872 ms  0.908 ms
 3  45.239.185.1 (45.239.185.1)  2.124 ms  5.175 ms  3.649 ms
 4  172.18.52.5 (172.18.52.5)  2.463 ms  4.667 ms  4.548 ms
 5  *  *  *
 6  *  *  *
 7  *  *  *
 8  *  *  *
 9  *  *  *
10  *  *  *
11  *  *  *
12  *  *  *
13  *  *  *
14  *  *  *
15  *  *  *
16  *  *  72.14.204.24 (72.14.204.24)  14.515 ms
17  *  *  *
18  *  *  *
19  *  *  *
20  *  *  *
21  *  *  *
22  *  *  *
23  *  *  *
24  *  *  *
25  *  *  *
26  *  *  *
27  *  *  *
28  *  *  *
29  *  *  *
30  *  *  *

vempire-ghost, 2023-11-18:
In today's snapshot, the problem has been resolved. Now, when I change the server's IP, OMR can find the new one and update it, returning to normal operation afterward. Thank you very much, Ysurac.

What is the time interval that OMR checks for the new IP? Is this configurable?

closed 2024-02-22 by github-actions[bot]