Ysurac/openmptcprouter #2997: Total bandwidth seems to be limited to around 250Mbps

cmsmith81, 2023-10-15:
Five broadband connections, each with 500Mbps throughput up and down are linked with MPTCP, total throughput should be 2,500Mbps (minus overhead, so reasonably between 1,500Mbps and 2,000Mbps)

## Current Behavior

Bandwidth is always less than 300Mbps, regardless of hardware configuration both on the router side and the VPS side.  I have tested using a VPS with more than 10Gbps actual bandwidth up and down along with a powerful x86 router with six separate 1Gbps WAN links.  I have tried many, many configurations including many, many different hardware configs on both sides.  There seems to always be a bottleneck of the total throughput.   I have tried this with both shadowsocks and V2Ray.  I have tried with many other configurations and optimizations.  I have tried both TCP and UDP.

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

I am trying to increase upload and download bandwidth by aggregating multiple WAN connections.

## Specifications

  - OpenMPTCProuter version: 0.59.1
  - OpenMPTCProuter VPS version: (latest as of Oct 6 2023, sorry can't find the specific version number)
  - OpenMPTCProuter VPS provider: Multiple - Rackspace, Azure, On-metal (self hosted), several other -- all very high perf servers with 10Gbps or greater actual upload and download speeds
  - OpenMPTCProuter platform: Multiple: RPI4, x86_64 High perf hardware
  - Country: United States

Ysurac, 2023-10-15:
VPS version is visible in /etc/motd on the VPS or when you login via SSH on the VPS.
RPI4 can reach, with chacha20 encryption enabled, up to 550Mb/s (sometimes a bit more in good conditions).
There is no limit to the bandwidth by OpenMPTCProuter, you can reach easily some hardware limits (network bandwidth, CPU power when encryption is enabled,...)

Latency between connections and VPS is important, if you have a high latency then you will get a lower aggregated connection.
Also bad WAN connections (when some packets are lost) can also give lower aggregated connections.
Setting BBRv2 as TCP Congestion Control can also help when connections are good.

If this can help, tests are done using Vultr and Digital Ocean (because they are available worldwide).

cmsmith81, 2024-01-16:
Changing from $1,000/mo Rackspace sever and physical self-hosted high performance server to your recommended Digital Ocean VPS service fixed the problem!  I appreciate the input and this solved the issue.  Thank you!

closed 2024-04-21 by github-actions[bot]

dakser, 2025-07-31:
Hi cmsmith81, can you share if you achieved the throughput of 2,500Mbps?