Ysurac/openmptcprouter #2998: Constant stream of mptcp subsocket kernel errors: "mptcp_add_sock failed ret: -1" in the log

ioogithub, 2023-10-16:
## Expected Behavior
There should be no kernel errors.

## Current Behavior
Lots of errors, started a few days ago. There is no other indication in the logs of problems. What are these errors and how can I fix them? If I reboot OMR, the errors come back.

## Possible Solution
No idea, does the stable version has complied kernel mptcp tools to troubleshoot these errors? 

## Steps to Reproduce the Problem

  1. Start OMR
  2. logread -f
  3. observe the errors

## Context (Environment)
I don't know what problems this is causing.

## Log
```
Mon Oct 16 15:55:15 2023 kern.err kernel: [ 6634.141479] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:15 2023 kern.err kernel: [ 6634.141940] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:15 2023 kern.err kernel: [ 6634.142388] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:15 2023 kern.err kernel: [ 6634.142822] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:15 2023 kern.err kernel: [ 6634.143450] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:15 2023 kern.err kernel: [ 6634.143692] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:15 2023 kern.err kernel: [ 6634.143877] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:15 2023 kern.err kernel: [ 6634.144350] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:15 2023 kern.err kernel: [ 6634.144755] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:15 2023 kern.err kernel: [ 6634.145150] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:30 2023 kern.warn kernel: [ 6649.499339] net_ratelimit: 5 callbacks suppressed
Mon Oct 16 15:55:30 2023 kern.err kernel: [ 6649.499341] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:30 2023 kern.err kernel: [ 6649.500079] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 15:55:30 2023 kern.err kernel: [ 6649.500423] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:05:16 2023 kern.err kernel: [ 7235.704584] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:05:16 2023 kern.err kernel: [ 7235.704930] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:12:57 2023 kern.err kernel: [ 7696.496316] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:12:57 2023 kern.err kernel: [ 7696.496645] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:12:57 2023 kern.err kernel: [ 7696.496950] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:12:57 2023 kern.err kernel: [ 7696.497257] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:12:57 2023 kern.err kernel: [ 7696.497546] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:12:57 2023 kern.err kernel: [ 7696.497812] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:12:57 2023 kern.err kernel: [ 7696.498066] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:12:57 2023 kern.err kernel: [ 7696.498318] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:12:57 2023 kern.err kernel: [ 7696.498548] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:12:57 2023 kern.err kernel: [ 7696.499893] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:13:10 2023 kern.warn kernel: [ 7709.300992] net_ratelimit: 24 callbacks suppressed
Mon Oct 16 16:13:10 2023 kern.err kernel: [ 7709.300993] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:13:10 2023 kern.err kernel: [ 7709.301439] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:13:10 2023 kern.err kernel: [ 7709.301627] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:29:20 2023 kern.err kernel: [ 8679.505203] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:29:20 2023 kern.err kernel: [ 8679.505398] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:29:20 2023 kern.err kernel: [ 8679.505588] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:29:20 2023 kern.err kernel: [ 8679.505801] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:29:20 2023 kern.err kernel: [ 8679.505984] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:29:20 2023 kern.err kernel: [ 8679.506165] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:29:20 2023 kern.err kernel: [ 8679.506406] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:29:20 2023 kern.err kernel: [ 8679.506592] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:29:20 2023 kern.err kernel: [ 8679.506779] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:29:20 2023 kern.err kernel: [ 8679.506963] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:38:46 2023 kern.warn kernel: [ 9245.244057] net_ratelimit: 18 callbacks suppressed
Mon Oct 16 16:44:26 2023 kern.err kernel: [ 9585.710471] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:44:26 2023 kern.err kernel: [ 9585.710661] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:44:26 2023 kern.err kernel: [ 9585.710856] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:44:26 2023 kern.err kernel: [ 9585.711040] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:44:26 2023 kern.err kernel: [ 9585.711225] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:44:26 2023 kern.err kernel: [ 9585.711404] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:44:26 2023 kern.err kernel: [ 9585.711588] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:44:39 2023 kern.err kernel: [ 9598.505907] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 16:44:39 2023 kern.err kernel: [ 9598.506096] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
[... 37 lines cut ...]
Mon Oct 16 17:08:27 2023 kern.err kernel: [11026.951901] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:08:27 2023 kern.err kernel: [11026.952047] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:16:08 2023 kern.err kernel: [11487.729882] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:16:08 2023 kern.err kernel: [11487.730051] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:16:08 2023 kern.err kernel: [11487.730668] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:16:08 2023 kern.err kernel: [11487.730831] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:16:08 2023 kern.err kernel: [11487.730983] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:16:08 2023 kern.err kernel: [11487.731170] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:16:08 2023 kern.err kernel: [11487.731314] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:16:08 2023 kern.err kernel: [11487.731453] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:16:08 2023 kern.err kernel: [11487.731576] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:16:08 2023 kern.err kernel: [11487.731700] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:26:38 2023 kern.err kernel: [12117.466211] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:26:38 2023 kern.err kernel: [12117.466530] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:26:38 2023 kern.err kernel: [12117.466689] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:26:38 2023 kern.err kernel: [12117.466855] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
Mon Oct 16 17:27:11 2023 kern.err kernel: [12150.735869] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1
```

## Specifications

- OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
- OpenMPTCProuter VPS version: VPS 0.1028
- OpenMPTCProuter VPS provider:
- OpenMPTCProuter platform: x86_64)

Ysurac, 2023-10-17:
What do you have in status page and what is the result of ```multipath``` command from the router ?

ioogithub, 2023-10-17:
$ multipath

```bond0 is deactivated
erspan0 is deactivated
eth0 is deactivated
eth1 is in default mode
eth2 is in default mode
eth3 is deactivated
gre0 is deactivated
gretap0 is deactivated
ifb4eth1 is deactivated
ifb4eth2 is deactivated
ifb4eth3 is deactivated
ifb4tun0 is deactivated
ip6gre0 is deactivated
ip6tnl0 is deactivated
lo is deactivated
sit0 is deactivated
teql0 is deactivated
tun0 is deactivated
```

Status page shows all green.

Ysurac, 2023-10-17:
What do you have in Network->MPTCP, "Fullmesh subflows for each pair of IP addresses" ?

ioogithub, 2023-10-17:
The value I am currently using is 3.

Ysurac, 2023-10-18:
Set it to "1", you may also need to reboot.

ioogithub, 2023-10-19:
The errors are transitory, I see them sometimes and sometimes I do not see them. Is a value of 2 or 3 here in appropriate? How does it cause this error? Is there a maximum number of sub flows for the whole system?

Ysurac, 2023-10-23:
There is a max number of available subflows to 8 per MPTCP session in current implementation.

closed 2024-01-26 by github-actions[bot]