Ysurac/openmptcprouter #3002: Alots of errors on log from 18/10 snapshot and only direct output with v2ray or xray.

vempire-ghost, 2023-10-19:
## Current Behavior
Today's snapshot has many errors in the log and is no longer connecting to the VPS with Xray or V2Ray. For the VPS, I ran the server-test script today after downloading the snapshot.

## Context (Environment)
Oct 19 01:08:11 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.33.2
Oct 19 01:08:11 OpenMPTCProuter user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Oct 19 01:08:11 OpenMPTCProuter user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Oct 19 01:08:11 OpenMPTCProuter user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: started, version 2.85 cachesize 150
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: DNS service limited to local subnets
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: using only locally-known addresses for domain test
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: using only locally-known addresses for domain onion
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: using only locally-known addresses for domain localhost
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: using only locally-known addresses for domain local
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: using only locally-known addresses for domain invalid
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: using only locally-known addresses for domain bind
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: using only locally-known addresses for domain use-application-dns.net
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: using only locally-known addresses for domain lan
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: using nameserver 127.0.0.1#5353
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: using 1 more local addresses
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: read /etc/hosts - 4 addresses
Oct 19 01:08:11 OpenMPTCProuter daemon.info dnsmasq[4127]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Oct 19 01:08:11 OpenMPTCProuter authpriv.info dropbear[4189]: Not backgrounding
Oct 19 01:08:11 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Oct 19 01:08:11 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Failed to connect to ubus
Oct 19 01:08:11 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Unable to locate ipset utility, disabling ipset support
Oct 19 01:08:11 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_lan' cannot resolve device of network 'lan'
Oct 19 01:08:11 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan1'
Oct 19 01:08:11 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan2'
Oct 19 01:08:11 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan3'
Oct 19 01:08:11 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omrvpn'
Oct 19 01:08:11 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'
Oct 19 01:08:11 OpenMPTCProuter user.notice xray: Rules UP
Oct 19 01:08:14 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Oct 19 01:08:14 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Oct 19 01:08:14 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Failed to connect to ubus
Oct 19 01:08:14 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Unable to locate ipset utility, disabling ipset support
Oct 19 01:08:14 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_lan' cannot resolve device of network 'lan'
Oct 19 01:08:14 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan1'
Oct 19 01:08:14 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan2'
Oct 19 01:08:14 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan3'
Oct 19 01:08:14 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omrvpn'
Oct 19 01:08:14 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'
Oct 19 01:08:14 OpenMPTCProuter daemon.notice wpa_supplicant[4590]: Successfully initialized wpa_supplicant
Oct 19 01:08:14 OpenMPTCProuter user.notice : Added device handler type: tunnel
Oct 19 01:08:14 OpenMPTCProuter user.notice : Added device handler type: Network device
Oct 19 01:08:14 OpenMPTCProuter user.notice : Added device handler type: bridge
Oct 19 01:08:14 OpenMPTCProuter user.notice : Added device handler type: veth
Oct 19 01:08:14 OpenMPTCProuter user.notice : Added device handler type: macvlan
Oct 19 01:08:14 OpenMPTCProuter user.notice : Added device handler type: 8021ad
Oct 19 01:08:14 OpenMPTCProuter user.notice : Added device handler type: 8021q
Oct 19 01:08:14 OpenMPTCProuter user.notice : Added device handler type: bonding
Oct 19 01:08:15 OpenMPTCProuter daemon.warn netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is enabled
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is setting up now
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is now up
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'wan1' is enabled
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is enabled
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is enabled
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is setting up now
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is now up
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Network device 'lo' link is up
Oct 19 01:08:15 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' has link connectivity 
Oct 19 01:08:15 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of lan (eth0)
Oct 19 01:08:15 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Oct 19 01:08:16 OpenMPTCProuter daemon.info dnsmasq[4127]: exiting on receipt of SIGTERM
[... 456 lines cut ...]
Oct 19 01:09:07 OpenMPTCProuter daemon.info xray[21224]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.1 linux/amd64)
Oct 19 01:09:07 OpenMPTCProuter daemon.info xray[21224]: A unified platform for anti-censorship.
Oct 19 01:09:07 OpenMPTCProuter daemon.info xray[21224]: 2023/10/19 01:09:07 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Oct 19 01:09:07 OpenMPTCProuter daemon.info xray[21224]: Failed to start: main: failed to load config files: [/var/etc/xray/xray.main.json] > infra/conf: Failed to build REALITY config. > infra/conf: empty "publicKey"
Oct 19 01:09:12 OpenMPTCProuter daemon.info xray[21619]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.1 linux/amd64)
Oct 19 01:09:12 OpenMPTCProuter daemon.info xray[21619]: A unified platform for anti-censorship.
Oct 19 01:09:12 OpenMPTCProuter daemon.info xray[21619]: Failed to start: main: failed to load config files: [/var/etc/xray/xray.main.json] > infra/conf: Failed to build REALITY config. > infra/conf: empty "publicKey"
[... cut ...]
Oct 19 01:09:22 OpenMPTCProuter daemon.info procd: Instance xray::xray.main s in a crash loop 6 crashes, 0 seconds since last crash
Oct 19 01:09:40 OpenMPTCProuter user.info : luci: accepted login on / for root from 192.168.100.2

Changing from Reality to VLESS seems to resolve the error in the log, but it still only shows "direct output" on the status page.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta1-5.4 r0+16862-170d9
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64
  - Country: Brazil

Ysurac, 2023-10-19:
For the logs errors, ipset was missing: I added it.
For VLESS-REALITY, you need to update VPS (run install script again) and you may need to force request key in wizard.
I will write some doc for VLESS REALITY, this should allow to bypass some country firewall and, as it's on port 443, may allow to bypass other limits providers can add on ports.

vempire-ghost, 2023-10-19:
I ran the script yesterday; do I need to run it again today?

Ysurac, 2023-10-19:
Yes, the script from the website was updated yesterday, so possible you don't have latest version.
If you have still same problem with vless-reality, what do you have in ```/etc/xray/xray-vless-reality.json``` ?

vempire-ghost, 2023-10-19:
Updated the VPS and the OMR with latest snapshot, but reality still dont work.
The problem now is this they cant contact xray on the vps
Oct 19 23:06:30 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1)
Oct 19 23:06:41 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164)
Oct 19 23:06:51 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77)
Oct 19 23:07:01 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1)
Oct 19 23:07:11 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42)
Oct 19 23:07:21 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162)
Oct 19 23:07:31 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162, 1.0.0.1)
Oct 19 23:07:41 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162, 1.0.0.1, 212.27.48.10)
Oct 19 23:07:42 OpenMPTCProuter daemon.err uhttpd[7836]: error: couldn't find the ModemManager process in the bus
Oct 19 23:07:43 OpenMPTCProuter daemon.err uhttpd[7836]: error: couldn't find the ModemManager process in the bus
Oct 19 23:07:52 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77, 1.1.1.1, 74.82.42.42, 198.41.212.162, 1.0.0.1, 212.27.48.10, 198.27.92.1)

This is the content of the xray-vless-reality.json, I suppressed the Id and keys.
{
    "inbounds": [
        {
            "port": 443,
            "tag": "omrin-vless-reality",
            "protocol": "vless",
            "settings": {
                "clients": [
                    {
                        "id": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
                        "flow": "xtls-rprx-vision"
                    }
                ],
                "decryption": "none"
            },
            "streamSettings": {
                "network": "tcp",
                "security": "reality",
                "realitySettings": {
                    "dest": "1.1.1.1:443",
                    "serverNames": [
                        ""
                    ],
                    "privateKey": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
                    "publicKey": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
                    "shortIds": [
                        ""
                    ]
                },
                "sockopt": {
                    "tcpMptcp": true,
                    "mark": 0
                }
            }
        }
    ],
    "routing": {
        "rules": [
            {
                "type": "field",
                "inboundTag": [
                    "omrin-tunnel"
                ]
            }
        ]
   }
}

Ysurac, 2023-10-20:
A configuration part is missing about xray on the router.
What is the result of ```uci show xray``` ? As I updated configuration multiple times, maybe something was not added in your config.

vempire-ghost, 2023-10-20:
xray.main=xray
xray.main.xray_file='/usr/bin/xray'
xray.main.mem_percentage='0'
xray.main.loglevel='error'
xray.main.access_log='/dev/null'
xray.main.error_log='/dev/null'
xray.main.inbounds='omr' 'omrtest'
xray.main.outbounds='omrout' 'omrexit'
xray.main.enabled='1'
xray.main_dns=dns
xray.main_dns.hosts='example.com|127.0.0.1'
xray.main_dns.enabled='0'
xray.main_policy=policy
xray.main_policy.enabled='1'
xray.main_policy.levels='policy_level_0'
xray.policy_level_0=policy_level
xray.policy_level_0.level='0'
xray.policy_level_0.handshake='4'
xray.policy_level_0.conn_idle='1200'
xray.policy_level_0.uplink_only='0'
xray.policy_level_0.downlink_only='0'
xray.policy_level_0.buffer_size='512'
xray.main_transparent_proxy=transparent_proxy
xray.main_transparent_proxy.proxy_mode='default'
xray.main_transparent_proxy.apnic_delegated_mirror='apnic'
xray.main_transparent_proxy.gfwlist_mirror='github'
xray.main_transparent_proxy.redirect_udp='0'
xray.main_transparent_proxy.redirect_port='1897'
xray.omrout=outbound
xray.omrout.tag='omrout_tunnel'
xray.omrout.s_vmess_port='65250'
xray.omrout.s_vmess_user_alter_id='0'
xray.omrout.s_vless_port='65248'
xray.omrout.s_vless_user_encryption='none'
xray.omrout.s_vless_user_alter_id='0'
xray.omrout.s_trojan_port='65249'
xray.omrout.s_trojan_user_encryption='none'
xray.omrout.s_trojan_user_alter_id='0'
xray.omrout.s_socks_port='65251'
xray.omrout.s_socks_user_encryption='none'
xray.omrout.s_socks_user_alter_id='0'
xray.omrout.ss_network='tcp'
xray.omrout.ss_security='tls'
xray.omrout.ss_tls_allow_insecure='1'
xray.omrout.ss_tls_disable_system_root='1'
xray.omrout.ss_tls_cert_usage='verify'
xray.omrout.ss_tls_cert_file='/etc/luci-uploads/client.crt'
xray.omrout.ss_tls_key_file='/etc/luci-uploads/client.key'
xray.omrout.s_shadowsocks_port='65252'
xray.omrout.mux_concurrency='8'
xray.omrout.ss_sockopt_mptcp='1'
xray.omrout.s_vless_reality_port='443'
xray.omrout.s_vless_reality_flow='xtls-rprx-vision'
xray.omrout.s_vless_reality_user_encryption='none'
xray.omrout.s_vless_reality_user_alter_id='0'
xray.omrout.s_shadowsocks_email='openmptcprouter'
xray.omrout.s_vmess_address='xxxxx.ddns.net'
xray.omrout.s_vless_address='xxxxxx.ddns.net'
xray.omrout.s_vless_reality_address='xxxxx.ddns.net'
xray.omrout.s_trojan_address='xxxxx.ddns.net'
xray.omrout.s_socks_address='xxxxx.ddns.net'
xray.omrout.s_shadowsocks_address='xxxxx.ddns.net'
xray.omrout.s_vmess_user_security='aes-128-gcm'
xray.omrout.s_vless_user_security='aes-128-gcm'
xray.omrout.s_vless_reality_user_security='aes-128-gcm'
xray.omrout.s_trojan_user_security='aes-128-gcm'
xray.omrout.s_socks_user_security='aes-128-gcm'
xray.omrout.s_shadowsocks_method='2022-blake3-aes-256-gcm'
xray.omrout.protocol='vless-reality'
xray.omrout.s_vmess_user_id='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
xray.omrout.s_vless_user_id='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
xray.omrout.s_vless_reality_user_id='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
xray.omrout.s_trojan_user_id='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
xray.omrout.s_socks_user_id='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
xray.omrout.s_shadowsocks_password='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
xray.omrout.s_vless_reality_public_key='xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
xray.omr=inbound
xray.omr.tag='omrtunnel'
xray.omr.listen='0.0.0.0'
xray.omr.port='1897'
xray.omr.protocol='dokodemo-door'
xray.omr.s_dokodemo_door_network='tcp' 'udp'
xray.omr.ss_sockopt_tproxy='redirect'
xray.omr.ss_sockopt_tcp_fast_open='1'
xray.omr.ss_sockopt_mptcp='1'
xray.omr.s_dokodemo_door_follow_redirect='1'
xray.omr6=inbound
xray.omr6.tag='omrtunnel6'
xray.omr6.listen='::'
xray.omr6.port='1898'
xray.omr6.protocol='dokodemo-door'
xray.omr6.s_dokodemo_door_network='tcp' 'udp'
xray.omr6.ss_sockopt_tproxy='tproxy'
xray.omr6.ss_sockopt_tcp_fast_open='1'
xray.omr6.s_dokodemo_door_follow_redirect='1'
xray.omrtest=inbound
xray.omrtest.port='1111'
xray.omrtest.protocol='socks'
xray.omrtest.listen='127.0.0.1'
xray.omrtest.s_socks_auth='noauth'
xray.omrtest.s_socks_udp='1'
xray.omrtest.s_socks_ip='127.0.0.1'
xray.omrtest.s_socks_userlevel='0'
xray.main_reverse=reverse
xray.main_reverse.enabled='1'
xray.main_reverse.bridges='omrbridge|omr.lan'
xray.omrexit=outbound
xray.omrexit.protocol='freedom'
xray.omrexit.tag='out'
xray.omrrouting=routing_rule
xray.omrrouting.type='field'
xray.omrrouting.inbound_tag='omrbridge'
xray.omrrouting.outbound_tag='omrout_tunnel'
xray.omrrouting.domain='full:omr.lan'
xray.omrroutingo=routing_rule
xray.omrroutingo.type='field'
xray.omrroutingo.inbound_tag='omrbridge'
xray.omrroutingo.outbound_tag='out'
xray.main_routing=routing
xray.main_routing.enabled='1'
xray.main_routing.rules='omrrouting' 'omrroutingo'

Ysurac, 2023-10-23:
Can you try on the router, via SSH, a ```xray run -config /var/etc/xray/xray.main.json``` and give me the result ?

Ysurac, 2023-10-24:
I'm not able to reproduce the issue for now.
Do you have any firewall on your VPS provider side ?

vempire-ghost, 2023-10-24:
In the VPS, only Shorewall is installed and configured using your script.
Do I need to open any other ports on the GCP firewall, besides the ones listed in your installation guide's wiki?

Ysurac, 2023-10-24:
For Xray VLESS Reality port 443 must be open on GCP.
I will make a special doc for XRay VLESS Reality before it's in release.

vempire-ghost, 2023-10-24:
I have this ports open
[image]

Ysurac, 2023-10-24:
ok, so in fact you have all ports open to the VPS. So nothing block port 443, must be something else.

closed 2024-01-28 by github-actions[bot]