Ysurac/openmptcprouter #3004: omr-vps-admin fails to install on Debian (pip3 install no longer allowed)

Steltek, 2023-10-20:
## Current Behavior

During VPS installation, the post-install script for omr-vps-admin fails with these error messages:

```
Setting up omr-vps-admin (0.5+20231017) ...
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
dpkg: error processing package omr-vps-admin (--configure):
 installed omr-vps-admin package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 omr-vps-admin
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

## Possible Solution
I believe the issue is that Debian will no longer allow installing Python packages into the OS (they must be installed into a venv) and expects globally installed packages to come from its own repositories. A workaround is passing "--break-system-packages", however this may cause issues further down the line.

## Steps to Reproduce the Problem

  1. Download http://www.openmptcprouter.com/server-beta/debian.sh
  2. Run the downloaded script with Upstream=yes set in the environment variables.

## Context (Environment)
VPS is running Debian 12.2

## Specifications

  - OpenMPTCProuter version: [v0.60beta1-6.1 r0+23789-ce6ad123e7](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: Debian 12.2 / http://www.openmptcprouter.com/server-beta/debian.sh
  - OpenMPTCProuter VPS provider: BuyVM
  - OpenMPTCProuter platform: x86_64
  - Country: Luxembourg

Ysurac, 2023-10-20:
There is a problem in the debian file that will be fixed, but it's UPSTREAM6 not UPSTREAM on the command line for the 6.1 kernel

Steltek, 2023-10-20:
Yes, I copied it incorrectly into the report. It IS running (after I did the workaround with --break-system-packages). Just wanted to report the package install problem.

jjtt, 2024-01-06:
@Steltek could you elaborate a bit on how to use the workaround? Where did you add `--break-system-packages`?

Ysurac, 2024-01-06:
@jjtt Snapshot script should not have errors, what is the script you use ?

closed 2024-04-11 by github-actions[bot]