Ysurac/openmptcprouter #3005: The hotplug event for 00-nego triggers ethernet port flapping for the i225 2.5G

khdegraaf, 2023-10-20:
## Expected Behavior
Using DHCP for a WAN connection using the i225 2.5G should be stable and work.  Particularly for the 5.4.238 kernel (which has a working driver for this hardware).  This is specifically using kmod-igc driver.  It is stable on OpenWrt 21.02.07 using the 5.4.238 kernel, and it is stable with a static ip

## Current Behavior
Every 3 to 60 seconds the connection disconnects and then reconnects.  In the system log, netifd is detecting down event, restarting dhcp connection, it comes up, then it goes down again.  It does this repeatedly if using dhcp but is solid and works for a static ip.

## Possible Solution
The code in openmptcprouter/files/etc/hotplug.d/iface/00-nego is too broad.  It matches this hardware (grep 2500), calls 'ethtool -s autoneg on' which triggers the flapping.  Commenting out this code fixes the problem.  I am presuming that this triggers the cycle somehow, but can't say more about that.  Presumably whatever fix you are seeking for certain hardware options needs to be more specific and not trigger this problem.

## Steps to Reproduce the Problem

  1.  Install OMR 60.1beta
  2.  Setup i225 2.5G port with DHCP

## Context (Environment)
We are trying to use the Silicom Cordoba router box with the i225 2.5G ports to connect to multiple StarLink WAN ports and need DHCP to use that reliably.  We can't do so with this bug (without commenting out your code mentioned above).

00-nego is buggy and causes DHCP to flap for i225 2.5G in particular, but possibly other 2.5G ethernet ports.

## Specifications

  - OpenMPTCProuter version: 59.1 and 60.1beta
  - OpenMPTCProuter VPS version: Doesn't matter, tried a lot of versions, and it isn't relevant
  - OpenMPTCProuter VPS provider: Doesn't matter
  - OpenMPTCProuter platform: x86_64
  - Country: USA

Ysurac, 2023-10-21:
Thanks, I added fix in develop branch, this will be in next release.

closed 2023-10-21 by Ysurac

simonemessina92, 2023-11-24:
@khdegraaf  I have that problem on my 2.5gb ports used as WANs, how exactly you modify the  00-nego file, you mean you can delete the file to avoid that problem? the only way of solving that for me was to enable "force link" for each wan, but it's not goot because I can't hot swap the wans external lte modems and I would like to do it instead!

Ysurac, 2023-11-24:
You can delete the file.