Ysurac/openmptcprouter #3007: OMRVPN restarts every 5minutes and 10 seconds.  Why?

maegibbons, 2023-10-22:
## Expected Behavior
glorytun vpn should NOT restart

## Current Behavior
Here is the current log:
```
Sun Oct 22 13:51:10 2023 daemon.info glorytun[18674]: STOPPED tun0
Sun Oct 22 13:51:11 2023 daemon.err glorytun[18674]: 185.XXX.XXX.XXX.65001: connected
Sun Oct 22 13:51:11 2023 daemon.info glorytun[18674]: STARTED tun0
Sun Oct 22 13:56:22 2023 daemon.info glorytun[18674]: STOPPED tun0
Sun Oct 22 13:56:23 2023 daemon.err glorytun[18674]: 185.XXX.XXX.XXX.65001: connected
Sun Oct 22 13:56:23 2023 daemon.info glorytun[18674]: STARTED tun0
Sun Oct 22 14:01:34 2023 daemon.info glorytun[18674]: STOPPED tun0
Sun Oct 22 14:01:35 2023 daemon.err glorytun[18674]: 185.XXX.XXX.XXX.65001: connected
Sun Oct 22 14:01:35 2023 daemon.info glorytun[18674]: STARTED tun0
Sun Oct 22 14:03:51 2023 user.notice post-tracking-post-tracking: Change master interface from wan2 (30 ms) to wan4 (13 ms)
Sun Oct 22 14:06:46 2023 daemon.info glorytun[18674]: STOPPED tun0
Sun Oct 22 14:06:46 2023 daemon.err glorytun[18674]: 185.XXX.XXX.XXX.65001: connected
Sun Oct 22 14:06:47 2023 daemon.info glorytun[18674]: STARTED tun0
Sun Oct 22 14:11:56 2023 daemon.info glorytun[18674]: STOPPED tun0
Sun Oct 22 14:11:57 2023 daemon.err glorytun[18674]: 185.XXX.XXX.XXX.65001: connected
Sun Oct 22 14:11:58 2023 daemon.info glorytun[18674]: STARTED tun0
Sun Oct 22 14:17:07 2023 daemon.err glorytun[18674]: read: Connection reset by peer
Sun Oct 22 14:17:07 2023 daemon.info glorytun[18674]: STOPPED tun0
Sun Oct 22 14:17:08 2023 daemon.err glorytun[18674]: 185.XXX.XXX.XXX.65001: connected
Sun Oct 22 14:17:08 2023 daemon.info glorytun[18674]: STARTED tun0
Sun Oct 22 14:17:08 2023 daemon.info glorytun[18674]: STOPPED tun0
Sun Oct 22 14:17:42 2023 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Sun Oct 22 14:17:43 2023 user.notice post-tracking-post-tracking: Glorytun VPN down, restart it
Sun Oct 22 14:17:43 2023 daemon.info glorytun: starting glorytun vpn instance vpn
Sun Oct 22 14:17:43 2023 daemon.notice netifd: Network device 'tun0' link is down
Sun Oct 22 14:17:43 2023 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sun Oct 22 14:17:43 2023 daemon.notice netifd: Interface 'omrvpn' is now down
Sun Oct 22 14:17:43 2023 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Oct 22 14:17:43 2023 daemon.err glorytun[28209]: AES-256-GCM is not available on your platform
Sun Oct 22 14:17:43 2023 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Oct 22 14:17:43 2023 daemon.notice netifd: Network device 'tun0' link is up
Sun Oct 22 14:17:43 2023 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sun Oct 22 14:17:43 2023 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sun Oct 22 14:17:43 2023 daemon.notice netifd: Interface 'omrvpn' is now up
Sun Oct 22 14:17:43 2023 daemon.info glorytun[28209]: INITIALIZED tun0
Sun Oct 22 14:17:43 2023 daemon.warn [8316]: <warn>  [SQM_IFB_909d7] invalid sysfs path read for net/SQM_IFB_909d7
Sun Oct 22 14:17:43 2023 daemon.warn [8316]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_909d7 not found
Sun Oct 22 14:17:43 2023 daemon.warn [8316]: <warn>  [SQM_IFB_83cb8] invalid sysfs path read for net/SQM_IFB_83cb8
Sun Oct 22 14:17:43 2023 daemon.warn [8316]: <warn>  [base-manager] couldn't handle kernel event: device net/SQM_IFB_83cb8 not found
Sun Oct 22 14:17:43 2023 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sun Oct 22 14:17:43 2023 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sun Oct 22 14:17:43 2023 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Oct 22 14:17:45 2023 user.notice post-tracking-post-tracking: Set firewall on server vps
Sun Oct 22 14:17:46 2023 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.4.1 dev eth4
```

## Possible Solution
Stop it restarting.

## Steps to Reproduce the Problem

  1.  Standard install of OpenMPTCPRouter
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Own
  - OpenMPTCProuter platform: x86_64 Proxmox VM
  - Country: UK

maegibbons, 2023-10-22:
Ok digging a bit further on the VPS Server end getting the following and it appears the SERVER is taking the tunnel down every 5m 10secs:

```
Oct 22 19:09:49 ubuntumpr systemd[1]: glorytun-tcp@tun0.service: Start-post operation timed out. Stopping.
Oct 22 19:09:49 ubuntumpr systemd[1]: glorytun-tcp@tun0.service: Failed with result 'timeout'.
Oct 22 19:09:49 ubuntumpr systemd[1]: Failed to start Glorytun TCP on tun0.
Oct 22 19:09:49 ubuntumpr omr-service[3130736]: Job for glorytun-tcp@tun0.service failed because a timeout was exceeded.
Oct 22 19:09:49 ubuntumpr omr-service[3130736]: See "systemctl status glorytun-tcp@tun0.service" and "journalctl -xe" for details.
Oct 22 19:09:49 ubuntumpr systemd[1]: glorytun-tcp@tun0.service: Scheduled restart job, restart counter is at 1.
Oct 22 19:09:49 ubuntumpr systemd[1]: Stopped Glorytun TCP on tun0.
Oct 22 19:09:49 ubuntumpr systemd[1]: Starting Glorytun TCP on tun0...
Oct 22 19:09:49 ubuntumpr networkd-dispatcher[640]: WARNING:Unknown index 441 seen, reloading interface list
Oct 22 19:09:49 ubuntumpr systemd-udevd[3130946]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Oct 22 19:09:49 ubuntumpr systemd-networkd[408]: gt-tun0: Link UP
Oct 22 19:09:49 ubuntumpr systemd-networkd[408]: gt-tun0: Gained carrier
Oct 22 19:09:49 ubuntumpr systemd-networkd[408]: gt-tun0: Gained IPv6LL
Oct 22 19:09:49 ubuntumpr systemd[1]: Started Glorytun TCP on tun0.
Oct 22 19:09:50 ubuntumpr glorytun-tcp-run[3130939]: ::ffff:185.254.89.211.45365: connected
Oct 22 19:09:50 ubuntumpr glorytun-tcp-run[3130939]: peer wants chacha20
Oct 22 19:09:52 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:09:52 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:09:52 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:09:52 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:09:57 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:09:57 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:09:57 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:09:57 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:10:02 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:10:02 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:10:02 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:10:02 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:10:07 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:10:07 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:10:07 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:10:07 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:10:12 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:10:12 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:10:12 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:10:12 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:10:17 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:10:17 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:10:17 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:10:17 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:10:22 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:10:22 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:10:22 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:10:22 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:10:27 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:10:27 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:10:27 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:10:27 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:10:32 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:10:32 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:10:32 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:10:32 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:10:37 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:10:37 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:10:37 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:10:37 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:10:42 ubuntumpr multipathd[564]: sda: add missing path
Oct 22 19:10:42 ubuntumpr multipathd[564]: sda: failed to get udev uid: Invalid argument
Oct 22 19:10:42 ubuntumpr multipathd[564]: sda: failed to get sysfs uid: Invalid argument
Oct 22 19:10:42 ubuntumpr multipathd[564]: sda: failed to get sgio uid: No such file or directory
Oct 22 19:10:47 ubuntumpr multipathd[564]: sda: add missing path
[... cut ...]
Oct 22 19:14:59 ubuntumpr networkd-dispatcher[640]: WARNING:Unknown index 442 seen, reloading interface list
Oct 22 19:14:59 ubuntumpr systemd-udevd[3139156]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Oct 22 19:14:59 ubuntumpr systemd-networkd[408]: gt-tun0: Link UP
Oct 22 19:14:59 ubuntumpr systemd-networkd[408]: gt-tun0: Gained carrier
Oct 22 19:14:59 ubuntumpr systemd-networkd[408]: gt-tun0: Gained IPv6LL
Oct 22 19:14:59 ubuntumpr systemd[1]: Started Glorytun TCP on tun0.
```

maegibbons, 2023-10-22:
ok more testing....

the omr.service script cannot ping 10.255.255.2 and nor can i from ssh!

But the tunnel and OMR is still working!

ifconfig gives:

```
gt-tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 10.255.255.1  netmask 255.255.255.252  destination 10.255.255.1
        inet6 fe80::5bd2:7fb:7326:3ca1  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 23494  bytes 24358080 (24.3 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 23053  bytes 3738768 (3.7 MB)
        TX errors 0  dropped 117 overruns 0  carrier 0  collisions 0
```

Should destination be "destination 10.255.255.1"

Routing table is:
```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         185.XXX.XX.X    0.0.0.0         UG    0      0        0 ens160
10.255.246.0    0.0.0.0         255.255.255.0   U     0      0        0 client-wg0
10.255.247.0    0.0.0.0         255.255.255.0   U     0      0        0 wg0
10.255.248.0    0.0.0.0         255.255.255.0   U     0      0        0 omr-bonding
10.255.248.2    0.0.0.0         255.255.255.255 UH    0      0        0 omr-bonding
10.255.250.0    10.255.250.2    255.255.255.0   UG    0      0        0 tun1
10.255.250.2    0.0.0.0         255.255.255.255 UH    0      0        0 tun1
10.255.251.2    0.0.0.0         255.255.255.255 UH    0      0        0 dsvpn0
10.255.252.0    10.255.252.2    255.255.255.0   UG    0      0        0 tun0
10.255.252.2    0.0.0.0         255.255.255.255 UH    0      0        0 tun0
10.255.253.0    0.0.0.0         255.255.255.0   U     0      0        0 mlvpn0
10.255.254.0    0.0.0.0         255.255.255.252 U     0      0        0 gt-udp-tun0
10.255.255.0    0.0.0.0         255.255.255.252 U     0      0        0 gt-tun0
185.254.89.0    0.0.0.0         255.255.255.0   U     0      0        0 ens160
192.168.100.0   10.255.255.2    255.255.255.0   UG    0      0        0 gt-tun0
```

192.168.100.1 does not ping either.

Ysurac, 2023-10-23:
The VPS part restart the VPN when it's not working correctly.
Do you have same issue using another VPN ?

maegibbons, 2023-10-23:
Ok so I traced the issue....

On my local OpenMPTCProuter i had the following route:

10.0.0.0        192.168.100.124 255.0.0.0       UG    0      0        0 eth0

192.168.100.124 is a local routing device for inter subnet routing locally.

For some reason this overrides the "local interface" routes for 10.255.255.0/30 on glorytun.
This is unexpected as local interface routes should override a less specific route which 10.0.0.0/8 certainly is!

I am working around this with a more complex routing table for local 10.x.x.x subnets but it is a pain!

Any suggestions?

Krs

Mark

closed 2024-01-26 by github-actions[bot]