Ysurac/openmptcprouter #3011: The possibility of OMR resolving DNS names from the VPS server's IP address with the default gateway disabled.

vempire-ghost, 2023-10-24:
## Expected Behavior
OMR can resolve DNS names even with the default gateway disabled.

## Current Behavior
OMR cannot resolve DNS names with the default gateway disabled.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta1-5.4 r0+16862-170d9
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64

When I use OMR with a web address instead of an IP for the server, and I turn off the default gateway, OMR loses the ability to resolve DNS. As a result, it can't connect to the server and activate the proxy/VPN. Is there a way to resolve the DNS from the server's IP with the default gateway turned off? I prefer to have it off because sometimes OMR connects directly to websites if it hasn't connected to the VPS yet. With the default gateway disabled, I'm certain that all outbound connections are going through my VPS. 
Unfortunately, I need to use DNS because my VPS has a dynamic IP that changes every time I turn it on or off.

vempire-ghost, 2023-10-24:
If I leave the default route enabled but turn off OMR-Tracker, would it have the effect I need? Would it stop directing connections directly until the VPS comes online?

Ysurac, 2023-10-25:
I don't really understand what you want, what is the result of ```ip r``` before you get VPS ?
If it's what I think, You don't want any DNS request to use your WAN or possible to use WAN directly for VPS ?
In first case, I may maybe force a resolve for this IP only else I would need to request VPS API to resolve it for me.
Else yes, maybe possible to disable OMR-Tracker when IP is still not resolved, I need to check that...
Tell me what is the best solution.

vempire-ghost, 2023-10-25:
I apologize for any confusion earlier. What you need is a way to ensure that no connection passing through OMR connects directly to the internet but only accesses the internet through your VPS. If the VPS is offline, all connections should be blocked until it comes back online. I used to achieve this by deactivating the default route, but with it deactivated, OMR can't resolve the domain address that I use as the IP of my VPS.

If there's another way to block direct internet access through OMR and ensure that all connections exit only through the VPS, that would solve my problem.

closed 2024-01-29 by github-actions[bot]