Ysurac/openmptcprouter #3012: VPN tunnel is always down and I can't get a public ip address

venethia, 2023-10-29:
## Expected Behavior
OMR should create a VPN tunnel from the router to the vps.

## Current Behavior
VPN tunnel is down and I can't get public ip address from ShadowSocks. The internet connection works in direct output sometimes with the main connection (master) and sometimes with the 4g connection. I already reinstall/update the vps script but the problem still remain.

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: 0.1028 5.10.0-26-amd64 (Debian 11)
  - OpenMPTCProuter VPS provider: Aruba 
  - OpenMPTCProuter platform: x86_64 

## Log

1) /var/log/daemon.log on VPS

`Oct 29 20:04:14 omtr systemd[1]: Stopping Shadowsocks-Libev Custom Manager Service for manager...
Oct 29 20:04:14 omtr systemd[1]: shadowsocks-libev-manager@manager.service: Succeeded.
Oct 29 20:04:14 omtr systemd[1]: Stopped Shadowsocks-Libev Custom Manager Service for manager.
Oct 29 20:04:14 omtr systemd[1]: Started Shadowsocks-Libev Custom Manager Service for manager.
Oct 29 20:04:14 omtr ss-manager[730580]:  2023-10-29 20:04:14 INFO: using tcp fast open
Oct 29 20:04:14 omtr ss-manager[730580]:  2023-10-29 20:04:14 INFO: using tcp no-delay
Oct 29 20:04:14 omtr ss-manager[730580]:  2023-10-29 20:04:14 INFO: using MPTCP
Oct 29 20:04:14 omtr ss-manager[730580]:  2023-10-29 20:04:14 INFO: running from root user
Oct 29 20:04:14 omtr ss-manager[730580]:  2023-10-29 20:04:14 INFO: working directory points to /root/.shadowsocks
Oct 29 20:04:14 omtr ss-manager[730580]:  2023-10-29 20:04:14 INFO: try to bind interface: 0.0.0.0, port: 65101
Oct 29 20:04:14 omtr ss-manager[730587]:  2023-10-29 20:04:14 INFO: initializing acl...
Oct 29 20:04:14 omtr ss-manager[730587]:  2023-10-29 20:04:14 INFO: enable TCP no-delay
Oct 29 20:04:14 omtr ss-manager[730587]:  2023-10-29 20:04:14 INFO: enable multipath TCP
Oct 29 20:04:14 omtr ss-server[730588]: resolving hostname to IPv6 address first
Oct 29 20:04:14 omtr ss-server[730588]: using tcp fast open
Oct 29 20:04:14 omtr ss-server[730588]: UDP relay enabled
Oct 29 20:04:14 omtr ss-server[730588]: enable TCP no-delay
Oct 29 20:04:14 omtr ss-server[730588]: initializing ciphers... chacha20-ietf-poly1305
Oct 29 20:04:14 omtr ss-server[730588]: tcp server listening at 0.0.0.0:65101
Oct 29 20:04:14 omtr ss-server[730588]: failed to listen on any address
Oct 29 20:04:14 omtr omr-service[730607]: Your device don't support multipath-TCP.
Oct 29 20:04:14 omtr omr-service[730607]: You have to install the pached kernel to use MPTCP.
Oct 29 20:04:14 omtr omr-service[730607]: See http://multipath-tcp.org/ for details
`
2) System log on router

`Sun Oct 29 20:02:45 daemon.err /usr/bin/ss-local[14691]: getpeername: Socket not connected
Sun Oct 29 20:02:45 2023 daemon.info omr-tracker-ss: Shadowsocks sss0 is down (can't contact via ...)
Sun Oct 29 20:02:45 2023 user.notice post-tracking-post-tracking: Set firewall on server vps
Sun Oct 29 20:06:25 2023 daemon.err /usr/bin/ss-local[14691]: getpeername: Socket not connected
Sun Oct 29 20:06:28 2023 daemon.err glorytun[14869]: couldn't set socket option `MPTCP_ENABLED (42)'
Sun Oct 29 20:06:28 2023 daemon.err glorytun[14869]: couldn't set socket option `MPTCP_ENABLED (26)'
Sun Oct 29 20:06:30 2023 daemon.err /usr/bin/ss-local[14691]: getpeername: Socket not connected`

Ysurac, 2023-10-30:
The kernel on the VPS is incorrect: You have a 5.10.0 kernel and it's not supported.The script install a customized 5.4 kernel.
After script installation, did you reboot ?
If yes, boot on custom kernel is supported ? In this case what is the VPS you use ? I will have to try one.

venethia, 2023-10-30:
I use [Aruba Cloud](https://www.cloud.it/vps.aspx). 

VPS O2A4 (Openstack).

I tried to reinstall/update the VPS with this command:

`wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | sh`

Then I made a reboot.

venethia, 2023-10-30:
Now it's work (partially). From the VPS control panel I can use the web console and choose the 5.4 kernel at boot. But there is no bonding. Only the master connection works. LAN1 (vdsl - master) + LAN2 (4g): no bonding, only LAN1 works. If I remove the eth cable from LAN1 I get an error: VPN tunnel DOWN and can't get public ip address from ShadowSocks (can't contact server admin script - no token yet available).

Ysurac, 2023-10-30:
Can you take a screenshot of System->OpenMPTCProuter, Status page ?

venethia, 2023-10-30:
- Shadowsocks and Glorytun TCP

[image]

- Shadowsocks and Glorytun TCP

[image]

Sometimes I got this warning on both connections with Glorytun TCP: "Multipath seems to be blocked on the connection".

With bot configuration (Glorytun TCP or UPD) only the master connection works: no bonding.

Ysurac, 2023-10-31:
To check if MPTCP is not blocked by ISP, you can check in Network->MPTCP, "MPTCP support check" tab.
How do you check aggregation ?

venethia, 2023-11-01:
MPTCP support check is ok both on wan1 and wan3.

If I unplug the ethernet cable of the master connection (wan1) with only wan3 (4g) connected, I often get these errors.

![4](https://github.com/Ysurac/openmptcprouter/assets/95854664/1d4f4f47-97c0-4eeb-99ff-f0634f37ce6d

[image]

[image]

[image]

Ysurac, 2023-11-01:
Did you change any configuration ? Do you have SQM enabled ?
What is the result on the router via SSH of ```omr-test-speed eth1```, ```omr-test-speed eth2``` and ```omr-test-speed``` (Ctrl+C to stop after 1 minute) ?

venethia, 2023-11-01:
SQM is enabled.

Just changed VPN and now bonding seems to work.

GlorytunTCP don't works for me and I always get "Tunnel VPN Down" on router.

GlorytunUDP works but without bonding.

With MLVPN bonfing works.

Very strange because with the same connections and vps one month ago was all ok with GlorytunTCP.

closed 2024-02-04 by github-actions[bot]