Ysurac/openmptcprouter #3014: DHCP not working on physical wan

ozmo-mac, 2023-10-30:
## Expected Behavior

Physical WAN interface should get dhcp address and work thur it, i have a public ip so now I have to have 2 nat network since I can’t control the other one. I solved by using a extra router but this is not ideal.
## Current Behavior

and it gets it but for some reason it disconnects and gets an new address and this loop continues and it’s unable to use it

Yes, shows interface down and up again 
## Specifications

  - OpenMPTCProuter version:  0.59.15.4
  - OpenMPTCProuter VPS version:  I know last one is not version but I can’t login right now to it.. will update soon
  - OpenMPTCProuter VPS provider: GCC
  - OpenMPTCProuter platform:  x86_64

Ysurac, 2023-10-31:
I don't understand what you mean. If there is a DHCP server on any WAN, it will get provided IP if WAN is set as DHCP client.
Do you use an I22x interfaces ?

ozmo-mac, 2023-10-31:
it gets the ip but the interface will go down for some reason. maybe something to do whit ping? it says can't ping server. if I set the interface static it works. I use RT8125 chipset. (TP-LINK TX201 -2.5Gb).

Ysurac, 2023-10-31:
What do you have in Status->System log ?
Did you try to force the interface ? In Network->Interfaces, edit interface, "Advanded settings" tab, "Force link" checkbox.
If still same problem, can you try, via SSH on the router ```rm /etc/hotplug.d/iface/00-nego``` ? (<= I think it's the problem)

ozmo-mac, 2023-10-31:
I will try tp force the link!
Will get you the log later.

Ysurac, 2023-10-31:
In fact ```rm /etc/hotplug.d/iface/00-nego``` should really fix the problem, as you have a 2.5Gb interface.

ozmo-mac, 2023-10-31:
rm /etc/hotplug.d/iface/00-nego 
what does it do? will it still be able to know if the cable gets disconnected whit out losing a tcp connection to server?

Ysurac, 2023-10-31:
This file was a workaround for a problem on a chipset but make problem on all 2.5Gb chipset...

ozmo-mac, 2023-10-31:
So I should just rm it?