Ysurac/openmptcprouter #3020: the snapshot 11-04 modemmanager cant work

fareign, 2023-11-05:
http://192.168.100.1/cgi-bin/luci/admin/status/modemmanager
 nothing

## Specifications

  - OpenMPTCProuter version: 0.60 snapshorts 1104
  - OpenMPTCProuter VPS version:lastest
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country: cn

fareign, 2023-11-05:
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[3628]: hotplug: remove network interface ifb4tun0: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[3628]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/ifb4tun0)
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[3698]: hotplug: remove network interface ifb4usb0: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[3698]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/ifb4usb0)
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[3752]: hotplug: remove network interface ifb4usb1: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[3752]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/ifb4usb1)
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[3908]: hotplug: remove network interface ifb4usb2: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[3908]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/ifb4usb2)
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[4027]: hotplug: remove network interface ifb4usb3: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[4027]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/ifb4usb3)
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[4097]: hotplug: remove network interface ifb4usb4: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[4097]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/ifb4usb4)
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[4207]: hotplug: add network interface SQM_IFB_6c394: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[4207]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_6c394)
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[4252]: hotplug: remove network interface SQM_IFB_6c394: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[4252]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_6c394)
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[4276]: hotplug: add network interface SQM_IFB_4c1c3: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[4276]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_4c1c3)
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[4316]: hotplug: remove network interface SQM_IFB_4c1c3: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[4316]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_4c1c3)
Nov  5 14:49:01 OpenMPTCProuter daemon.info ModemManager[4343]: hotplug: add network interface ifb4usb0: event processed
Nov  5 14:49:01 OpenMPTCProuter daemon.debug ModemManager[4343]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/ifb4usb0)
Nov  5 14:49:02 OpenMPTCProuter daemon.info ModemManager[4490]: hotplug: add network interface SQM_IFB_055ef: event processed
Nov  5 14:49:02 OpenMPTCProuter daemon.debug ModemManager[4490]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_055ef)
Nov  5 14:49:02 OpenMPTCProuter daemon.info ModemManager[4529]: hotplug: remove network interface SQM_IFB_055ef: event processed
Nov  5 14:49:02 OpenMPTCProuter daemon.debug ModemManager[4529]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_055ef)
Nov  5 14:49:02 OpenMPTCProuter daemon.info ModemManager[4571]: hotplug: add network interface SQM_IFB_2d2bf: event processed
Nov  5 14:49:02 OpenMPTCProuter daemon.debug ModemManager[4571]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_2d2bf)
Nov  5 14:49:02 OpenMPTCProuter daemon.info ModemManager[4623]: hotplug: remove network interface SQM_IFB_2d2bf: event processed
Nov  5 14:49:02 OpenMPTCProuter daemon.debug ModemManager[4623]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_2d2bf)
Nov  5 14:49:02 OpenMPTCProuter daemon.info ModemManager[4653]: hotplug: add network interface ifb4usb1: event processed
Nov  5 14:49:02 OpenMPTCProuter daemon.debug ModemManager[4653]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/ifb4usb1)
Nov  5 14:49:02 OpenMPTCProuter daemon.info ModemManager[4812]: hotplug: add network interface SQM_IFB_f031e: event processed
Nov  5 14:49:02 OpenMPTCProuter daemon.debug ModemManager[4812]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_f031e)
Nov  5 14:49:02 OpenMPTCProuter daemon.info ModemManager[4850]: hotplug: remove network interface SQM_IFB_f031e: event processed
Nov  5 14:49:02 OpenMPTCProuter daemon.debug ModemManager[4850]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_f031e)
Nov  5 14:49:02 OpenMPTCProuter daemon.info ModemManager[4869]: hotplug: add network interface SQM_IFB_e687c: event processed
Nov  5 14:49:02 OpenMPTCProuter daemon.debug ModemManager[4869]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_e687c)
Nov  5 14:49:02 OpenMPTCProuter daemon.info ModemManager[4907]: hotplug: remove network interface SQM_IFB_e687c: event processed
Nov  5 14:49:02 OpenMPTCProuter daemon.debug ModemManager[4907]: hotplug: sysfspath is a virtual device (/sys/devices/virtual/net/SQM_IFB_e687c)
Nov  5 14:49:02 OpenMPTCProuter daemon.info ModemManager[4944]: hotplug: add network interface ifb4tun0: event processed
[... cut ...]
Nov  5 22:49:12 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Nov  5 22:49:12 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Nov  5 22:49:12 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Nov  5 14:49:15 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 49.7.00.41)
**Nov  5 14:49:17 OpenMPTCProuter daemon.err omr-tracker-xray[9834]: sh: missing ]
Nov  5 14:49:17 OpenMPTCProuter daemon.err omr-tracker-xray[9834]: sh: missing ]**

fareign, 2023-11-05:
also xray tracker erro

Ysurac, 2023-11-05:
Do you have an interface using ModemManager already set ?
ModemManager is, for now, only launched when an interface use ModemManager protocol.
Do you have any issues (other then this too verbose log) with ModemManager ?

fareign, 2023-11-06:
> Do you have an interface using ModemManager already set ? ModemManager should, for now, only launched when an interface use ModemManager protocol. Do you have any issues (other then this too verbose log) with ModemManager ?

OK,no modemManager protocol use. but the old version can find modem.

Ysurac, 2023-11-17:
Seems to be now fixed since last week. A script was missing.

closed 2024-02-22 by github-actions[bot]