Ysurac/openmptcprouter #3025: interface

Doraemon309, 2023-11-06:
I wonder if kmod-mlx4 or kmod-mlx5 is available in the interface?

Ysurac, 2023-11-07:
It's available in software packages list System->Software.

Doraemon309, 2023-11-08:
> It's available in software packages list System->Software.

Yes I know it!
I mean is the kmod-mlx package already installed or do you have to install it yourself?

Ysurac, 2023-11-08:
Packages are not installed by default.

Doraemon309, 2023-11-08:
> Packages are not installed by default.

Do you have a schedule for the next release? Can you add a network interface?

Ysurac, 2023-11-08:
Why ? It's the only interface available on your platform ? What is the model ?

Doraemon309, 2023-11-08:
> Why ? It's the only interface available on your platform ? What is the model ?

Well, it must be said that nowadays the encapsulation into sfp makes the x86 model compact!
Maybe that makes you wonder about the upcoming versions!

closed 2024-02-11 by github-actions[bot]