Ysurac/openmptcprouter #3027: OMR doesn't connect VPS with "can't contact server asmin script" error

Chatissimus, 2023-11-06:
## Expected Behavior
OMR is connecting to the VPS

## Current Behavior
an error "can't contact server admin script" is displayed
the VPS section is no more displayed on settings wizard 

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: Milkywan
  - OpenMPTCProuter platform: RPI4

[image]
[image]

Ysurac, 2023-11-07:
Strange, it's a personalized image ?
What do you have, via SSH on the router, if you do a ```uci show openmptcprouter``` ?

Chatissimus, 2023-11-08:
Hello, 
No , strait one, it was working for month since last update.
according log of my NAS, 3 unexpected power shutdown happened last week ... (I was off for 2 weeks)
I think SD card is crappy 

```
root@OpenMPTCProuter:~# uci show openmptcprouter
uci: Entry not found
```

Ysurac, 2023-11-09:
Yes, seems to be a dead SD card. You can try a fsck on it but not sure this will solve something.

closed 2024-02-12 by github-actions[bot]