Ysurac/openmptcprouter #3029: Client OpenVPN with rpi4

Esysteme, 2023-11-07:
It is possible to add an OpenVPN client on OpenMPTCProuter, or I need to add this on the VPS ?

Other question : 
How use VPS to access to his LAN ? (add private IP and add rule on iptable ? and configure route on OpenMPTCProuter (rpi 4) ?

Esysteme, 2023-11-07:
my ultimate goal : 

get one box with everything inside : 

- openmptcprouter (of course)
- powered / charge by usb type C
- 4 port ethernet  (1 port usable for entrance for openmptcprouter)
- client openvpn
- dual sim card
-  router wifi
- with battery (ok nice to have)

and of course the smallest as possible

Ysurac, 2023-11-08:
It's possible to use OpenVPN on OpenMPTCProuter, but you need to use an not already used interface (don't use tun0).
OpenVPN is already installed on the VPS and can be used, but some firewall config will be needed (never tested as client).

VPS have access to IP of the LAN, but by default it's blocked on router side, you need to allow on the router incoming traffic from VPN.

For the box I'm searching the same and I think I'm not the only one. For now I have nothing that can be quite cheap and have all these with a good CPU.

Esysteme, 2023-11-08:
Personally i don't care about price, if you have one even expensive. the goal is to keep all my seup in back bag of 20 liters. (laptop / keyboard / mouse / 2* monitor / speaker / headphone and alim for all of that)

but by default it's blocked on router side, you need to allow on the router incoming traffic from VPN
=> where it is located ?

Ysurac, 2023-11-08:
In Network->Firewall, VPN, Input REJECT to ACCEPT:
[image]

Esysteme, 2023-11-08:
Merci beaucoup.

Encore encore une fois un grand bravo pour ton travail :)

closed 2023-11-08 by Esysteme