Ysurac/openmptcprouter #3030: backup vps not connecting?

ozmo-mac, 2023-11-08:
## Expected Behavior

for some reason its not connecting to backup server. I'm not sure if this expected, since main server connection is good.
## Current Behavior

just fills the logs whit this.

Wed Nov  8 09:16:44 2023 daemon.err /usr/bin/ss-redir[19489]: invalid password or cipher
Wed Nov  8 09:16:44 2023 user.notice OMR-VPS: Setting server config error, try again
Wed Nov  8 09:16:48 2023 daemon.err /usr/bin/ss-redir[19489]: invalid password or cipher
Wed Nov  8 09:16:51 2023 user.notice OMR-VPS: Can't get backup token, try later (can ping server backup on 34.89.1xx.xx, server API answer on 34.89.1xx.xx)
Wed Nov  8 09:16:58 2023 daemon.err /usr/bin/ss-redir[19487]: invalid password or cipher
Wed Nov  8 09:16:59 2023 daemon.err /usr/bin/ss-redir[19488]: invalid password or cipher
Wed Nov  8 09:16:59 2023 user.notice OMR-VPS: Can't get backup token, try later
Wed Nov  8 09:16:59 2023 daemon.err /usr/bin/ss-redir[19487]: invalid password or cipher
Wed Nov  8 09:17:02 2023 user.notice OMR-VPS: Setting server config error, try again

## Possible Solution

feels like a config problem, not sure about it. 
key is the same in router and vps. 

## Steps to Reproduce the Problem

  1. add main vps 
  2. after that works add the backup
  3. observer how it fills the logs
  4. be confused

## Context (Environment)

I'm trying to have a backup server in case the main fails.

main works fine. backup not connecting. I can ssh to vps and it looks fine. router can ping it but can't get token?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c 
  - OpenMPTCProuter VPS version:  0.1028
  - OpenMPTCProuter VPS provider: GCC
  - OpenMPTCProuter platform:  X86_64
  - Country:  Finland

Ysurac, 2023-11-08:
Did you also reboot backup server after install ? 
Check again backup server token (not the same as master server).
If router can't contact the backup server, keys can't be synchronized and this explain ss-redir errors.

ozmo-mac, 2023-11-08:
I did, but only via the google could console.. tried via ssh and sudo reboot, it fixed 
`Wed Nov 8 09:16:59 2023 user.notice OMR-VPS: Can't get backup token, try later`

but got a new error 

`Wed Nov  8 09:59:56 2023 kern.info kernel: [ 6989.620952] traps: logread[28436] general protection fault ip:7f219bf074e9 sp:7ffeb57cac80 error:0 in libc.so[7f219bec6000+49000]`
and still stays still..
Wed Nov  8 09:59:56 2023 daemon.err /usr/bin/ss-redir[19490]: invalid password or cipher

would a different time zone server have anything to do whit it?

ozmo-mac, 2023-11-08:
update it didn't fix it, just came up later..

Wed Nov  8 10:09:44 2023 daemon.err /usr/bin/ss-redir[19487]: invalid password or cipher
Wed Nov  8 10:09:45 2023 user.notice OMR-VPS: Can't get backup token, try later (can ping server backup on 34.89.16x.xx, server API answer on 34.89.16x.xx)
Wed Nov  8 10:09:49 2023 daemon.err /usr/bin/ss-redir[19488]: invalid password or cipher
Wed Nov  8 10:09:53 2023 daemon.err /usr/bin/ss-redir[19490]: invalid password or cipher
Wed Nov  8 10:09:53 2023 user.notice OMR-VPS: Can't get backup token, try later
Wed Nov  8 10:09:53 2023 daemon.err /usr/bin/ss-redir[19487]: invalid password or cipher
Wed Nov  8 10:09:55 2023 daemon.err /usr/bin/ss-redir[19487]: invalid password or cipher
Wed Nov  8 10:09:55 2023 user.notice OMR-VPS: Setting server config error, try again

Ysurac, 2023-11-08:
Timezone doesn't change anything, UTC is used internally.
In the Wizard, "Show advanced settings" checkbox, check "Force retrieve settings" and Save.

ozmo-mac, 2023-11-08:
Tried that, no change. 
here's the log 

[https://pastebin.com/A1WU4J1t](url)

ozmo-mac, 2023-11-08:
and here's the router config if that helps anything

[https://pastebin.com/8QrzE0Za](url)

Ysurac, 2023-11-10:
Range 65000-65535 are open on VPS provider firewall if there is one ?

ozmo-mac, 2023-11-10:
yes. and icmp also

Ysurac, 2023-11-17:
Should be fixed in develop branch, so in snapshot image. Will be fixed in next release.

closed 2024-02-22 by github-actions[bot]