Ysurac/openmptcprouter #3031: Bypassing all traffic except for one machine ?

patimages, 2023-11-08:
## Expected Behavior
Bypassing all traffic except for one machine ? 

## Current Behavior
Hello and thank you again Ycarus for this awesome tool!
A little technical question: would it be possible to use OMR-bypass in reverse ;D. Let me explain: I would like all the machines on my network to go through a single interface (bypass) except for one machine which would go through aggregation (for work). Is this feasible?
Why is that ? In fact, I recently aggregated my ADSL with a 5G box and the speed going directly through the 5G box is almost exactly twice as fast as when I go through aggregation (using omr-test-speed: 57M direct by the box, 27M in aggregate - my ADSL being at 0.5M). I have two very good mini computers whose load never exceeds 50% (1 for the server and 1 for the VPS where I have access to a gigabyte connection through work), so no worries at this level. And I use the latest software versions (not beta). I can't really explain the problem except that the 5G box seems to be 100% CPU in aggregation. If I remove the aggregation, it goes to 10-20%... So it seems that OMR is overloading the CPU of the 5G box.
Many thanks for any help!

## Specifications

  - OpenMPTCProuter version: v0.59.1
  - OpenMPTCProuter VPS version: 0.1028
  - OpenMPTCProuter VPS provider: Work
  - OpenMPTCProuter platform: openmptcprouter-v0.58.5-r0+16336-b36068d35d-x86-64-generic-squashfs-combined

Ysurac, 2023-11-08:
Bad 5G box maybe, but you can try to lower MTU on this connection to check if it's the problem.
Else to bypass all but not one is not possible. But, as there is only one, you can set it as static IP and use OMR IP as gateway and disable DHCP on OMR and use another box as DHCP.

patimages, 2023-11-08:
Many thanks for the quick reply ! It actually helped me to find the solution. 
I'm posting it here in case others have the same problem. What killed the 5G box CPU was a setting called "Easy Mesh" (activated by default on TP-link NX510V). Somehow, the box kept on trying to access some MAC address, permanently. Stopping it solved the CPU issue and thereby the aggregation speed.

patimages, 2023-11-08:
Merci !

patimages, 2023-11-08:
Actually, a bonus question : omr-test-speed is great for checking download speed from the routeur. Any similar command to check upload speed (aggregated or on eth1 for instance) ?

Ysurac, 2023-11-08:
Thanks for the feedback.
On future release, omr-test-speed-server will be able to test upload speed, but it doesn't give real results (VPS python part seems to limit the speed...).
```omr-iperf``` can test download or upload speed between server, and you can use iperf3 with some public servers. But this doesn't work like ```omr-test-speed```...

patimages, 2023-11-08:
Thanks a lot ! I ll test all that asap.

patimages, 2023-11-11:
Hello again, 
Just a quick question : as a reminder, I am aggregating a lousy DSL (5Mbps) and a great 5G box (4-500Mbps). All is working smoothly (not cpu excess and lots of bandwidth available for the VPS). The issue I have is that the aggregated download speed is about 30-40MBps while the download speed of the 5G Box alone (when I switch off the DSL in "network>interface") is more around 50-60MBps. Is such unexpected behaviour due to the huge difference between the bandwidth of 2 interfaces (100-fold) ? Anything I can do except killing the DSL interface ?

hairy-dog, 2023-11-11:
I have a vDSL connection and two LTE connections. I just added a FWA 5G connection and was disappointed to find that it didn't help at all.  

I started by checking that the lan was working properly, but that made no difference. 

Reading the comments in the "100+100=100" discussion gave me things to try, and they worked! 

I now get over 85% of the sum of the speeds of all the wan connections. Frankly, the LTE 4G connections are little use for the downlink, but they help the uplink a lot.

patimages, 2023-11-11:
Hi, 
It did not really help but thanks: I just removed SQM and already had BBR2: still a slightly lower speed in aggregation of the 2 interface vs a single one (in the following, left is aggregation of 2, right is after killing the slow interface).
[image]

hairy-dog, 2023-11-12:
I don't think.those are the only two things I altered.

closed 2023-11-25 by patimages