Ysurac/openmptcprouter #3032: proxy and vps status become yellow cant turn to green if not setup wizard

fareign, 2023-11-08:
no token yet avalilable.

if not setup wizard ,it never turn to green. although the net status is good.

is some relate to this err?

Nov  8 18:42:33 OpenMPTCProuter daemon.err omr-tracker-xray[26434]: sh: missing ]
Nov  8 18:42:33 OpenMPTCProuter daemon.err omr-tracker-xray[26434]: sh: missing ]

## Specifications

  - OpenMPTCProuter version:snapshots the latest 6.1kernel
  - OpenMPTCProuter VPS version: 0.1029 test upstream
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country: cn

Ysurac, 2023-11-08:
I don't know what you mean by "no setup wizard" ? It's after an initial configuration and a failure of proxy ? What do you have in system log else ?

Ysurac, 2023-11-08:
And what is the proxy and protocol used ?

fareign, 2023-11-08:
Nov  8 20:26:48 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 49.7.115.41)
Nov  8 20:26:48 OpenMPTCProuter user.notice omr-tracker: Launching...
Nov  8 20:26:55 OpenMPTCProuter user.notice omr-tracker: Launched
Nov  8 20:26:57 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 49.7.115.41)
Nov  8 20:27:21 OpenMPTCProuter daemon.err omr-tracker-xray[16520]: sh: missing ]
Nov  8 20:27:21 OpenMPTCProuter daemon.err omr-tracker-xray[16520]: sh: missing ]

no erro log ,because it work well ,only status yellow.
[image]

fareign, 2023-11-08:
proxy is xray trojan,v2ray has the same problem

fareign, 2023-11-08:
I test  ss is the same.

fareign, 2023-11-08:
if I save and apply openmptcprouter wizard.it become green ,and afew seconds late become yellow . thoes interface is green.it nerver turn to green. you can test with no very good internet.

fareign, 2023-11-09:
if i remove the 4G antenna,status will change. so the status is working.
but I install the antenna back, the 4G working. the status become (NO token yet avaliable) not green 
@Ysurac

fareign, 2023-11-09:
[Fix when VPS contact is a DNS](https://github.com/Ysurac/openmptcprouter-feeds/commit/5b32c33c0530b3fce4951d5807b6bcc62c3379ef)  Maybe is this fix.

fareign, 2023-11-11:
[Restart missing WAN tracker when needed](https://github.com/Ysurac/openmptcprouter-feeds/commit/510c13a8419af0046e992ba6639cc8a6b29cfd57)  is this update

closed 2024-02-14 by github-actions[bot]