Ysurac/openmptcprouter #3035: A long time between interface down and up

fareign, 2023-11-11:
## Expected Behavior

Nov 11 04:31:43 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (usb1) switched off because check error and ping from 192.168.225.58 error (202.96.209.133,202.97.224.69,223.5.5.5)
Nov 11 04:31:43 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Delete default route to 112.74.39.209 via 192.168.225.1 dev usb1
Nov 11 04:33:01 OpenMPTCProuter user.notice post-tracking-001-post-tracking: wan2 (usb1) switched up
Nov 11 04:33:01 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Change in routes, set ip route replace default scope global nexthop via 192.168.225.1 dev usb1 weight 100 (omrvpn_intf: tun0)

during donw and up about 80 seconds.
In the 80 seconds,interface has internet connect by there no internet because interface  is not switched up.

so if dont delete defaut route what will happen?
 

## Steps to Reproduce the Problem

  1.I closed mptcp,only use one 4G interface directout
  2.I'm not sure which reason cause interface down
  3.during down interface can ping delaytime .

## Specifications

  - OpenMPTCProuter version: 0.60beta
  - OpenMPTCProuter VPS version: 0.1029test
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country: cn

Ysurac, 2023-11-11:
You can change delay in Services->OMR-Tracker

fareign, 2023-11-11:
> You can change delay in Services->OMR-Tracker

interface timeout only 2s
retry3
interval 2
so 80 seconds . is the other program use ,or sleep.

Ysurac, 2023-11-12:
Yes, there is a little sleep after an interface detected as down, I will force it to use parameter instead.

closed 2024-02-15 by github-actions[bot]