Ysurac/openmptcprouter #3042: HELP ! Lost access to OMR...

patimages, 2023-11-15:
OpenMPTCProuter version: v0.59.1
OpenMPTCProuter VPS version: 0.1028
OpenMPTCProuter VPS provider: Work
OpenMPTCProuter platform: openmptcprouter-v0.58.5-r0+16336-b36068d35d-x86-64-generic-squashfs-combined

Hello, 
I messed up: I went to MPTCP and placed one of my 2 links in "enabled" and the other in "backup" (none in master). I lost all accesses to OMR : any way to solve this without a complete reinstall ? 
Many thanks !

Ysurac, 2023-11-15:
You mean local access or remote access via VPS ?
This shouldn't change anything for local access.

patimages, 2023-11-15:
Ah ok : I am not onsite. I ll test tonight. Many thanks : I thought I had to reinstall everything. Ycarus, you created such a great tool that when it breaks, we are panicking ;-D !

closed 2023-11-15 by patimages