Ysurac/openmptcprouter #3046: Master WAN Service Degradation - OMR connectivity becomes unstable

ccmks, 2023-11-23:
## Expected Behavior
When OMR detected service degradation on master WAN, such as packet loss over 15-20%, it will no longer be used as active connection and the master role is given to another WAN that is stable

## Current Behavior
From time to time, we are experiencing WAN degradation service on master WAN, where over 15% packet loss which makes the connectivity from OMR to VPS become very unstable, until we manually unplug the cable from master WAN modem, then the connection become stable.

I was hoping this is something where we can tweak on the parameter to have OMR shutdown the master WAN, if they can detect service degradation. Also when detecting high ping also considered WAN degradation.

## Specifications

  - OpenMPTCProuter version: 0.59.1-5.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Melbicom SG
  - OpenMPTCProuter platform: x86

Ysurac, 2023-11-23:
It's configured via Services->OMR-Tracker. You can put some settings more "aggressive".

ccmks, 2023-11-24:
Thank you for point that out, what is the best recommendation parameter based on my case earlier?

Thank you

ccmks, 2023-11-29:
How to setup parameter, let's say if the packet loss is over 10%, then the WAN will be considered down?

Thank you

Ysurac, 2023-11-29:
You can set on a % packet loss. But you can lower Tries or Retry Interval.
Packet loss need over a period of time, I will check what I can do to detect bad connections.

ccmks, 2023-11-30:
In OMR-Tracker, there is no WAN1 interface, should I add it manually then
configure the tracker against that interface?

Thank you

ccmks, 2023-11-30:
I would suggest to build SLA feature in OMR, it will detect based on % packet loss and ping time

If either ping or packet loss threshold met, then OMR will remove the WAN as available link

closed 2024-03-05 by github-actions[bot]