Ysurac/openmptcprouter #3048: vps install stops on second run

adamb, 2023-11-27:
## Expected Behavior

I'm trying to install the VPS server software on a minipc running Debian11.

For some reason, running this failed.  

`sh debian-x86_64.sh`

So I ran it again and it seemed to finish ok.   But when I use the server key I see errors in /var/log/daemon.log 

I see a lot of these kinds of messages.

```Nov 26 18:15:31 debian glorytun-tcp-run[339264]: ::ffff:xxx.xx.x.xxx.48575: connected
Nov 26 18:15:31 debian glorytun-tcp-run[339264]: peer sends a bad hash!
Nov 26 18:15:31 debian glorytun-tcp-run[339264]: ::ffff:xxx.xx.x.xxx.48575: key exchange failed
Nov 26 18:15:31 debian ss-server[547]: failed to handshake with xxx.xx.x.xxx: authentication error 
```

When I run the setup script again, it ends like this:

```dpkg-deb: building package 'omr-shadowsocks-libev' in '../omr-shadowsocks-libev_3.3.5-3_amd64.deb'.
dpkg-deb: building package 'omr-libshadowsocks-libev2' in '../omr-libshadowsocks-libev2_3.3.5-3_amd64.deb'.
dpkg-deb: building package 'omr-libshadowsocks-libev2-dbgsym' in '../omr-libshadowsocks-libev2-dbgsym_3.3.5-3_amd64.deb'.
dpkg-deb: building package 'omr-shadowsocks-libev-dbgsym' in '../omr-shadowsocks-libev-dbgsym_3.3.5-3_amd64.deb'.
dpkg-deb: building package 'omr-libshadowsocks-libev-dev' in '../omr-libshadowsocks-libev-dev_3.3.5-3_amd64.deb'.
 dpkg-genbuildinfo --build=binary
 dpkg-genchanges --build=binary >../omr-shadowsocks-libev_3.3.5-3_amd64.changes
dpkg-genchanges: info: binary-only upload (no source code included)
 dpkg-source --after-build .
dpkg-buildpackage: info: binary-only upload (no source included)
```

and it doesn't update the `openmptcprouter_config.txt` file.

## Current Behavior

Is there some way to get the correct key from somewhere on the VPS server?  The key in `/etc/openmptcprouter-vps-admin/omr-admin-config.json` matches the one in `openmptcprouter_config.txt`.  

Is there a way to re-run the debian-x86_64.sh script without having to re-install Debian?  It took me a long time to get Debian 11 installed so I would rather not have to reinstall it.

`uname -a` returns `Linux debian 5.4.207-mptcp #1 SMP Sun Jul 24 14:39:44 UTC 2022 x86_64 GNU/Linux` which seems correct.

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: VPS 0.1028
  - OpenMPTCProuter VPS provider: myself
  - OpenMPTCProuter platform:  RPI4

adamb, 2023-11-27:
BTW, this server is running at a personal residence that his fiber.  I'm trying to get better performance by moving the VPS to the same area.  I'm in Puerto Rico and the local VPS provider I tried has really bad ping times.  The closest is Miami.  I would prefer running the VPS from a personal residence since it avoids needing to do any bypass rules.

I have this working at another location, but it's on the other side of the country. I would rather have one running locally if possible.

Ysurac, 2023-11-27:
You can check keys in ```/etc/shadowsocks-libev/manager.json``` and ```/etc/glorytun-tcp/tun0.key```. Keyx should be the same in Settings->OpenMPTCProuter, "Wizard" and "Advanced settings" checkbox. You may have to check "force retrieve keys" if it's a router that was configured with another keys.

adamb, 2023-11-27:
Thanks, that got me further.  The keys on the VPS and the OMR side were definitely different.  In the UI there isn't a way to set the keys on a per server basis.  I have three different VPS servers I swap around between.  I'm surprised that each one would have the same keys.   Where does the wizard get these keys from?  

Anyway, coping over the shadow and glory keys from VPS to OMR on the Pi got me further.

Now I'm seeing `Can't connect Server Admin Script` and it cycles between `No result` or `No token yet available`.  I think it keeps restarting omr-admin-service on the vps.

Now the daemon.log shows

```Nov 27 06:41:09 debian glorytun-tcp-run[444073]: ::ffff:172.56.2.231.24783: connected
Nov 27 06:41:15 debian glorytun-tcp-run[444073]: ::ffff:172.56.2.231.38535: connected
Nov 27 06:41:49 debian systemd[1]: Stopping OMR-Admin...
Nov 27 06:41:49 debian systemd[1]: omr-admin.service: Succeeded.
Nov 27 06:41:49 debian systemd[1]: Stopped OMR-Admin.
Nov 27 06:41:49 debian systemd[1]: omr-admin.service: Consumed 1.104s CPU time.
Nov 27 06:41:49 debian systemd[1]: Started OMR-Admin.
Nov 27 06:43:03 debian systemd[1]: Stopping OMR-Admin...
Nov 27 06:43:03 debian systemd[1]: omr-admin.service: Succeeded.
Nov 27 06:43:03 debian systemd[1]: Stopped OMR-Admin.
Nov 27 06:43:03 debian systemd[1]: omr-admin.service: Consumed 1.303s CPU time.
Nov 27 06:43:03 debian systemd[1]: Started OMR-Admin.```

Ysurac, 2023-11-27:
Some keys are synchronized between all servers, some are set on server change using Server admin key that is set for each servers. There is know issue on server fail over in current release that are fixed in snapshot (so will be fixed in next release).
Can you try to do a ```journalctl -u omr-admin``` onn the VPS ? maybe more log to check why omr-admin is restarting.

adamb, 2023-11-27:
journalctl isn't very verbose:

```root@debian:~# journalctl -u omr-admin | tail -10
Nov 27 07:14:53 debian systemd[1]: Stopping OMR-Admin...
Nov 27 07:14:53 debian systemd[1]: omr-admin.service: Succeeded.
Nov 27 07:14:53 debian systemd[1]: Stopped OMR-Admin.
Nov 27 07:14:53 debian systemd[1]: omr-admin.service: Consumed 1.119s CPU time.
Nov 27 07:14:53 debian systemd[1]: Started OMR-Admin.
Nov 27 07:15:38 debian systemd[1]: Stopping OMR-Admin...
Nov 27 07:15:38 debian systemd[1]: omr-admin.service: Succeeded.
Nov 27 07:15:38 debian systemd[1]: Stopped OMR-Admin.
Nov 27 07:15:38 debian systemd[1]: omr-admin.service: Consumed 1.145s CPU time.
Nov 27 07:15:38 debian systemd[1]: Started OMR-Admin.
```

daemon.log shows this.  It seems like shadowsocks gets reset right after the admin comes up.  I checked and the ss keys match on vps and omr.

```Nov 27 07:13:24 debian systemd[1]: Started OMR-Admin.
Nov 27 07:13:47 debian ss-server[535]: remote recv: Connection reset by peer
Nov 27 07:13:49 debian ss-server[535]: remote recv: Connection reset by peer
Nov 27 07:14:08 debian systemd[1]: Stopping OMR-Admin...
Nov 27 07:14:08 debian systemd[1]: omr-admin.service: Succeeded.
Nov 27 07:14:08 debian systemd[1]: Stopped OMR-Admin.
Nov 27 07:14:08 debian systemd[1]: omr-admin.service: Consumed 1.129s CPU time.
Nov 27 07:14:08 debian systemd[1]: Started OMR-Admin.
Nov 27 07:14:53 debian systemd[1]: Stopping OMR-Admin...
Nov 27 07:14:53 debian systemd[1]: omr-admin.service: Succeeded.
Nov 27 07:14:53 debian systemd[1]: Stopped OMR-Admin.
Nov 27 07:14:53 debian systemd[1]: omr-admin.service: Consumed 1.119s CPU time.
Nov 27 07:14:53 debian systemd[1]: Started OMR-Admin.
Nov 27 07:14:54 debian ss-server[535]: remote recv: Connection reset by peer
```

adamb, 2023-11-28:
I noticed that the omr-admin.service is hanging on `wget -4 -qO- -T 1 http://hostname.openmptcprouter.com` for some reason.  

From my server, it doesn't seem to be able to resolve this.  

I tried the wget by hand without the -q and it tried 20 times with this error:

```
2023-11-27 20:23:17--  (try:20)  http://hostname.openmptcprouter.com/
Connecting to hostname.openmptcprouter.com (hostname.openmptcprouter.com)|94.23.252.192|:80... connected.
HTTP request sent, awaiting response... Read error (Connection timed out) in headers.
Giving up.
```

It seems that `/usr/local/bin/omr-admin.py` is getting tripped up because of this.  

Something unique about my hostname must be tripping up the request.  

``` 
root@debian:~# systemctl status omr-admin.service
● omr-admin.service - OMR-Admin
     Loaded: loaded (/lib/systemd/system/omr-admin.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-11-27 20:23:43 EST; 56s ago
   Main PID: 998594 (python3)
      Tasks: 8 (limit: 4484)
     Memory: 44.0M
        CPU: 982ms
     CGroup: /system.slice/omr-admin.service
             ├─998594 python3 /usr/local/bin/omr-admin.py
             ├─999046 /bin/sh -c wget -4 -qO- -T 1 http://hostname.openmptcprouter.com
             └─999047 wget -4 -qO- -T 1 http://hostname.openmptcprouter.com
```

adamb, 2023-11-28:
BTW, I'm using duckdns.org to set the domain name for this vps server.  Maybe there is a place where I'm supposed to set the hostname on the VPS so omr-admin will be happy?

Ysurac, 2023-11-28:
As a workaround, you can add a "hostname" section at the end of ```/etc/openmptcprouter-vps-admin/omr-admin-config.json``` after debug part (don't forget the comma after debug part) like that:
```
    "debug": false,
    "hostname": "myserver.duckdns.org"
}
```

adamb, 2023-11-28:
That did it!  Thanks.  

I'm curious why `wget -4 -qO- -T 1 http://hostname.openmptcprouter.com` hangs from my machine.  Is it because there is no reverse DNS entry for it?

Ysurac, 2023-11-28:
I modified the script for next release so this will not hang (in fact it retries too much time and another script restart omr-admin if it take too much time to answer).
No idea why you get no answer on this website, no reverse shouldn't change anything. Maybe IP banned somewhere...
I will move this website to another server or check if I can use another way to do that.

adamb, 2023-11-28:
Thanks for the info.  I'm happy to test something for you if you want.  I'm excited to have a server nearby.

closed 2023-12-01 by adamb