Ysurac/openmptcprouter #3050: Modem Manager needs to be restarted on powerON

yannick2801, 2023-11-28:
when I start the router, the interfaces don't start properly automatically, I have to restart each interface until the modemManager catches the network

## Specifications

  - OpenMPTCProuter version: Linux OpenMPTCProuter 5.4.238 v0.60beta1
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: ROCK4 (RPI4)

[image]

After restart, we catch the IPs and the network works properly : 

[image]

how can it restart automaticaly si the network is not catched up ?

Ysurac, 2023-11-28:
Changes are made on ModemManager in latest snapshot.

yannick2801, 2023-11-29:
Do you meen i should add a few seconds of latency ? (is it in seconds ?)
[image]

Ysurac, 2023-11-29:
No, I was thinking there is a wait time field, but it's not for this protocol.
You can try to update to latest snapshot and if you still have the issue I will try to fix it.

yannick2801, 2023-12-05:
same probleme with an image in Kernel 6.1 : 
`root@OpenMPTCProuter:~# uname -a`
`Linux OpenMPTCProuter 6.1.55 #0 SMP PREEMPT Mon Sep 25 21:21:09 2023 aarch64 GNU/Linux`

need to restart ModemModule to get the network.
can it be because the ModemManager are not detected at the begginning, maybe...

yannick2801, 2023-12-05:
When i Start the router i have this : 

[image]

root@OpenMPTCProuter:~# **mmcli -L**
    /org/freedesktop/ModemManager1/Modem/1 [Quectel] RM520N-GL
    /org/freedesktop/ModemManager1/Modem/0 [Quectel] RM520N-GL
    /org/freedesktop/ModemManager1/Modem/2 [Quectel] RM520N-GL
root@OpenMPTCProuter:~# **mmcli -m 0**
  -----------------------------------
  General  |                    path: /org/freedesktop/ModemManager1/Modem/0
           |               device id: f15df47a47e7ad1f753edd0db4b4b0eb268cfdd8
  -----------------------------------
  Hardware |            manufacturer: Quectel
           |                   model: RM520N-GL
           |       firmware revision: RM520NGLAAR01A06M4G
           |          carrier config: France-Commercial-Orange
           | carrier config revision: 0A010B21
           |            h/w revision: 20000
           |               supported: gsm-umts, lte, 5gnr
           |                 current: gsm-umts, lte, 5gnr
           |            equipment id: 868371050194725
  -----------------------------------
  System   |                  device: /sys/devices/platform/fe3c0000.usb/usb5/5-1
           |                 physdev: /sys/devices/platform/fe3c0000.usb/usb5/5-1
           |                 drivers: option1, qmi_wwan
           |                  plugin: quectel
           |            primary port: cdc-wdm0
           |                   ports: cdc-wdm0 (qmi), ttyUSB2 (at), ttyUSB3 (at), wwan0 (net)
  -----------------------------------
  Status   |                    lock: sim-pin2
           |          unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                   state: disabled
           |             power state: on
  -----------------------------------
  Modes    |               supported: allowed: 3g; preferred: none
           |                          allowed: 4g; preferred: none
           |                          allowed: 3g, 4g; preferred: 4g
           |                          allowed: 3g, 4g; preferred: 3g
           |                          allowed: 5g; preferred: none
           |                          allowed: 4g, 5g; preferred: 5g
           |                          allowed: 4g, 5g; preferred: 4g
           |                          allowed: 3g, 5g; preferred: 5g
           |                          allowed: 3g, 5g; preferred: 3g
           |                          allowed: 3g, 4g, 5g; preferred: 5g
           |                          allowed: 3g, 4g, 5g; preferred: 4g
           |                          allowed: 3g, 4g, 5g; preferred: 3g
           |                 current: allowed: 3g, 4g, 5g; preferred: 5g
  -----------------------------------
  Bands    |               supported: utran-1, utran-4, utran-6, utran-5, utran-8, utran-2, 
           |                          eutran-1, eutran-2, eutran-3, eutran-4, eutran-5, eutran-7, eutran-8, 
           |                          eutran-12, eutran-13, eutran-14, eutran-17, eutran-18, eutran-19, 
           |                          eutran-20, eutran-25, eutran-26, eutran-28, eutran-29, eutran-30, 
           |                          eutran-32, eutran-34, eutran-38, eutran-39, eutran-40, eutran-41, 
           |                          eutran-42, eutran-43, eutran-46, eutran-48, eutran-66, eutran-71, 
           |                          utran-19, ngran-1, ngran-2, ngran-3, ngran-5, ngran-7, ngran-8, 
           |                          ngran-12, ngran-13, ngran-14, ngran-18, ngran-20, ngran-25, ngran-26, 
           |                          ngran-28, ngran-29, ngran-30, ngran-38, ngran-40, ngran-41, ngran-48, 
           |                          ngran-66, ngran-70, ngran-71, ngran-75, ngran-76, ngran-77, ngran-78, 
           |                          ngran-79
           |                 current: utran-1, utran-4, utran-5, utran-8, utran-2, eutran-1, 
           |                          eutran-2, eutran-3, eutran-4, eutran-5, eutran-7, eutran-8, 
           |                          eutran-12, eutran-13, eutran-14, eutran-17, eutran-18, eutran-19, 
           |                          eutran-20, eutran-25, eutran-26, eutran-28, eutran-29, eutran-30, 
           |                          eutran-32, eutran-34, eutran-38, eutran-39, eutran-40, eutran-41, 
           |                          eutran-42, eutran-43, eutran-46, eutran-48, eutran-66, eutran-71, 
           |                          utran-19, ngran-1, ngran-2, ngran-3, ngran-5, ngran-7, ngran-8, 
           |                          ngran-12, ngran-13, ngran-14, ngran-18, ngran-20, ngran-25, ngran-26, 
           |                          ngran-28, ngran-29, ngran-30, ngran-38, ngran-40, ngran-41, ngran-48, 
           |                          ngran-66, ngran-70, ngran-71, ngran-75, ngran-76, ngran-77, ngran-78, 
           |                          ngran-79
  -----------------------------------
  IP       |               supported: ipv4, ipv6, ipv4v6
  -----------------------------------
  3GPP     |                    imei: 868371050194725
           |           enabled locks: sim, fixed-dialing
  -----------------------------------
  3GPP EPS |    ue mode of operation: csps-2
           |  initial bearer ip type: ipv4v6
  -----------------------------------
  SIM      |        primary sim path: /org/freedesktop/ModemManager1/SIM/0
           |          sim slot paths: slot 1: /org/freedesktop/ModemManager1/SIM/0 (active)
           |                          slot 2: none
root@OpenMPTCProuter:~# **ifstatus QUECTEL1**
{
        "up": false,
        "pending": false,
        "available": false,
        "autostart": true,
        "dynamic": false,
        "proto": "modemmanager",
        "data": {

        },
        "errors": [
                {
                        "subsystem": "interface",
                        "code": "NO_DEVICE"
                }
        ]
}
root@OpenMPTCProuter:~#

Ysurac, 2023-12-05:
What is the result of ```uci show network``` ?

yannick2801, 2023-12-05:
root@OpenMPTCProuter:~#   **uci show network**

network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='5'
network.globals=globals
network.globals.ula_prefix='fdbd:a475:f759::/48'
network.globals.packet_steering='1'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='cubic'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='2'
network.globals.mptcp_subflows='3'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.proto='static'
network.lan.ipaddr='192.168.100.1'
network.lan.netmask='255.255.255.0'
network.lan.device='eth0'
network.lan.ifname='eth0'
network.lan.metric='6'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.txqueuelen='2000'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1_dev=device
network.wan1_dev.name='wan1'
network.wan1_dev.txqueuelen='1000'
network.wan1_dev.type='macvlan'
network.wan1_dev.ifname='eth1'
network.wan1_dev.mode='vepa'
network.wan1_dev.macaddr='de:8d:e8:b3:89:34'
network.wan2_dev=device
network.wan2_dev.name='wan2'
network.wan2_dev.txqueuelen='1000'
network.wan2_dev.type='macvlan'
network.wan2_dev.ifname='eth2'
network.wan2_dev.mode='vepa'
network.wan2_dev.macaddr='60:be:22:0a:fb:8c'
network.omrvpn=interface
network.omrvpn.device='tun0'
network.omrvpn.proto='none'
network.omrvpn.ip4table='vpn'
network.omrvpn.multipath='off'
network.omrvpn.leasetime='12h'
network.omrvpn.type='tunnel'
network.omrvpn.txqueuelen='100'
network.omrvpn.metric='1500'
network.omrvpn.mtu='1440'
network.omr6in4=interface
network.omr6in4.proto='6in4'
network.omr6in4.ip4table='vpn'
network.omr6in4.multipath='off'
network.omr6in4.ipaddr='10.255.251.2'
network.omr6in4.peeraddr='10.255.251.1'
network.omr6in4.gateway='fe80::a00:1/126'
network.omr6in4.ip6addr='fe80::a00:2/126'
network.omr6in4.auto='0'
network.omr6in4.metric='1201'
network.tun0=device
network.tun0.name='tun0'
network.tun0.mtu='1500'
network.QUECTEL1=interface
network.QUECTEL1.proto='modemmanager'
network.QUECTEL1.pincode='0000'
network.QUECTEL1.iptype='ipv4v6'
network.QUECTEL1.multipath='on'
network.QUECTEL1.addlatency='0'
network.QUECTEL1.ipv6='0'
network.QUECTEL1.metric='7'
network.QUECTEL1.defaultroute='0'
network.QUECTEL1.peerdns='0'
network.QUECTEL1.device='/sys/devices/platform/fe3c0000.usb/usb6/6-1'
network.QUECTEL1.delegate='0'
network.QUECTEL2=interface
network.QUECTEL2.proto='modemmanager'
network.QUECTEL2.pincode='0000'
network.QUECTEL2.iptype='ipv4v6'
network.QUECTEL2.multipath='master'
network.QUECTEL2.addlatency='0'
network.QUECTEL2.device='/sys/devices/platform/usb@fe900000/fe900000.usb/xhci-hcd.1.auto/usb4/4-1'
network.QUECTEL2.metric='10'
network.QUECTEL2.defaultroute='0'
network.QUECTEL2.peerdns='0'
network.QUECTEL2.ipv6='0'
network.QUECTEL2.delegate='0'
network.QUECTEL3=interface
network.QUECTEL3.proto='modemmanager'
network.QUECTEL3.device='/sys/devices/platform/fe380000.usb/usb8/8-1'
network.QUECTEL3.pincode='0000'
network.QUECTEL3.iptype='ipv4v6'
network.QUECTEL3.multipath='on'
network.QUECTEL3.addlatency='0'
network.QUECTEL3.metric='11'
network.QUECTEL3.defaultroute='0'
network.QUECTEL3.peerdns='0'
network.QUECTEL3.ipv6='0'
network.QUECTEL3.delegate='0'
network.QUECTEL3_dev=device
network.QUECTEL2_dev=device
network.QUECTEL1_dev=device
network.lan_dev=device
network.lan_dev.name='eth0'
root@OpenMPTCProuter:~#

Ysurac, 2023-12-05:
From Modemamanger I see: ```/sys/devices/platform/fe3c0000.usb/usb5/5-1``` 
In the config for Quectel1: ```network.QUECTEL1.device='/sys/devices/platform/fe3c0000.usb/usb6/6-1```
And I can't see anything in usb5 in the config, there is something strange here.

yannick2801, 2023-12-06:
i have restarted all the router 
then i have restarted all the ModemManagers

root@OpenMPTCProuter:~# mmcli -L
    /org/freedesktop/ModemManager1/Modem/3 [Quectel] RM520N-GL
    /org/freedesktop/ModemManager1/Modem/2 [Quectel] RM520N-GL
    /org/freedesktop/ModemManager1/Modem/4 [Quectel] RM520N-GL

root@OpenMPTCProuter:~# mmcli -m 2
  ----------------------------------
  General  |                   path: /org/freedesktop/ModemManager1/Modem/2
           |              device id: 561e18e6c545793b10d90cbacaf059bda1279e9a
  ----------------------------------
  Hardware |           manufacturer: Quectel
           |                  model: RM520N-GL
           |      firmware revision: RM520NGLAAR01A06M4G
           |         carrier config: default
           |           h/w revision: 20000
           |              supported: gsm-umts, lte, 5gnr
           |                current: gsm-umts, lte, 5gnr
           |           equipment id: 868371050194659
  ----------------------------------
  System   |                 device: /sys/devices/platform/usb@fe900000/fe900000.usb/xhci-hcd.1.auto/usb4/4-1
           |                physdev: /sys/devices/platform/usb@fe900000/fe900000.usb/xhci-hcd.1.auto/usb4/4-1
           |                drivers: qmi_wwan, option1
           |                 plugin: quectel
           |           primary port: cdc-wdm2
           |                  ports: cdc-wdm2 (qmi), ttyUSB10 (at), ttyUSB11 (at), wwan2 (net)
  ----------------------------------
  Status   |                   lock: sim-pin2
           |         unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                  state: connected
           |            power state: on
           |            access tech: lte, 5gnr
           |         signal quality: 80% (recent)
  ----------------------------------
  Modes    |              supported: allowed: 3g; preferred: none
           |                         allowed: 4g; preferred: none
           |                         allowed: 3g, 4g; preferred: 4g
           |                         allowed: 3g, 4g; preferred: 3g
           |                         allowed: 5g; preferred: none
           |                         allowed: 4g, 5g; preferred: 5g
           |                         allowed: 4g, 5g; preferred: 4g
           |                         allowed: 3g, 5g; preferred: 5g
           |                         allowed: 3g, 5g; preferred: 3g
           |                         allowed: 3g, 4g, 5g; preferred: 5g
           |                         allowed: 3g, 4g, 5g; preferred: 4g
           |                         allowed: 3g, 4g, 5g; preferred: 3g
           |                current: allowed: 3g, 4g, 5g; preferred: 5g
  ----------------------------------
  Bands    |              supported: utran-1, utran-4, utran-6, utran-5, utran-8, utran-2, 
           |                         eutran-1, eutran-2, eutran-3, eutran-4, eutran-5, eutran-7, eutran-8, 
           |                         eutran-12, eutran-13, eutran-14, eutran-17, eutran-18, eutran-19, 
           |                         eutran-20, eutran-25, eutran-26, eutran-28, eutran-29, eutran-30, 
           |                         eutran-32, eutran-34, eutran-38, eutran-39, eutran-40, eutran-41, 
           |                         eutran-42, eutran-43, eutran-46, eutran-48, eutran-66, eutran-71, 
           |                         utran-19, ngran-1, ngran-2, ngran-3, ngran-5, ngran-7, ngran-8, 
           |                         ngran-12, ngran-13, ngran-14, ngran-18, ngran-20, ngran-25, ngran-26, 
           |                         ngran-28, ngran-29, ngran-30, ngran-38, ngran-40, ngran-41, ngran-48, 
           |                         ngran-66, ngran-70, ngran-71, ngran-75, ngran-76, ngran-77, ngran-78, 
           |                         ngran-79
           |                current: utran-1, utran-4, utran-5, utran-8, utran-2, eutran-1, 
           |                         eutran-2, eutran-3, eutran-4, eutran-5, eutran-7, eutran-8, 
           |                         eutran-12, eutran-13, eutran-14, eutran-17, eutran-18, eutran-19, 
           |                         eutran-20, eutran-25, eutran-26, eutran-28, eutran-29, eutran-30, 
           |                         eutran-32, eutran-34, eutran-38, eutran-39, eutran-40, eutran-41, 
           |                         eutran-42, eutran-43, eutran-46, eutran-48, eutran-66, eutran-71, 
           |                         utran-19, ngran-1, ngran-2, ngran-3, ngran-5, ngran-7, ngran-8, 
           |                         ngran-12, ngran-13, ngran-14, ngran-18, ngran-20, ngran-25, ngran-26, 
           |                         ngran-28, ngran-29, ngran-30, ngran-38, ngran-40, ngran-41, ngran-48, 
[... 385 lines cut ...]
network.QUECTEL2.addlatency='0'
network.QUECTEL2.device='/sys/devices/platform/usb@fe900000/fe900000.usb/xhci-hcd.1.auto/usb4/4-1'
network.QUECTEL2.metric='10'
network.QUECTEL2.defaultroute='0'
network.QUECTEL2.peerdns='0'
network.QUECTEL2.ipv6='0'
network.QUECTEL2.delegate='0'
network.QUECTEL3=interface
network.QUECTEL3.proto='modemmanager'
network.QUECTEL3.device='/sys/devices/platform/fe380000.usb/usb8/8-1'
network.QUECTEL3.pincode='0000'
network.QUECTEL3.iptype='ipv4v6'
network.QUECTEL3.multipath='on'
network.QUECTEL3.addlatency='0'
network.QUECTEL3.metric='11'
network.QUECTEL3.defaultroute='0'
network.QUECTEL3.peerdns='0'
network.QUECTEL3.ipv6='0'
network.QUECTEL3.delegate='0'
network.QUECTEL3_dev=device
network.QUECTEL2_dev=device
network.QUECTEL1_dev=device
network.lan_dev=device
network.lan_dev.name='eth0'
root@OpenMPTCProuter:~#

Ysurac, 2023-12-06:
If you do a ```ifup QUECTEL3``` then it's working ? if not, a ```ifup QUECTEL3 && ifdown QUECTEL3``` ?

yannick2801, 2023-12-06:
it worked when it's the MM_MODEM_DISABLED error : 

[image]

Ysurac, 2023-12-06:
OK, I will check what I can do for both cases...

yannick2801, 2023-12-12:
Do you think i can copie this file into a router in 5.4 because in a 6.1 router my VPS doesn't works properly  ([Issue 3055](https://github.com/Ysurac/openmptcprouter/issues/3055))

BR
Yannick