Ysurac/openmptcprouter #3051: Unable to Install OMR Script on VPS

ccmks, 2023-11-28:
I have been trying to install the following script:

wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | sh

Whenever, I am trying to run the script, always getting the following error

Install mptcp kernel and shadowsocks...
Get:1 https://repo.openmptcprouter.com buster InRelease [2128 B]
Hit:2 http://security.debian.org/debian-security buster/updates InRelease
Hit:3 http://deb.debian.org/debian buster-backports InRelease
Fetched 2128 B in 1s (1716 B/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package pkg-config is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package rename is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'rename' has no installation candidate
E: Package 'pkg-config' has no installation candidate

This happen when installing on Debian 10 and Debian 11

Please let me know how to fix this

Thank you

Ysurac, 2023-11-28:
Please use template when you open an issue.
Package rename ( https://packages.debian.org/search?keywords=rename ) and package pkg-config ( https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=pkg-config ) are still package available on Debian.
I think you have an issue in your package source configuration, do you have ```deb http://deb.debian.org bullseye main``` somewhere in /etc/apt/sources.list.d ?
What is your VPS provider ?

ccmks, 2023-11-28:
It is local VPS provider in Indonesia

ccmks, 2023-11-28:
I looked on here /etc/apt/sources.list.d and it is an empty directory

Ysurac, 2023-11-28:
Maybe ```/etc/apt/sources.list```.
All I can say is that it seems you have a problem in your Debian installation.

closed 2023-11-29 by ccmks