Ysurac/openmptcprouter #3052: Is it possible to copy keys from one VPS to another?

vempire-ghost, 2023-11-28:
Using two VPS that I'd like to have both use the same keys for OMR, proxy, and VPN, so I wouldn't need to change anything in OMR when switching between them. I can copy the keys by copying the configuration files from omr-admin and the shadowsocks, v2ray, and xray proxies. However, copying the files from the VPNs as I did with the proxies didn't work, and when I copied the files from OPENVPN, practically everything stopped working, and I had to do a clean installation again.

Is there a way to transfer or copy-paste the keys from one VPS to another more effectively?

Ysurac, 2023-11-28:
OMR support multiples servers. It's working correctly in snapshot release and it retrieve missing keys when needed using VPS API.

vempire-ghost, 2023-11-28:
> Rsync might work, which basically complete clone from one VPS to another VPS. I used to test it several years ago and it was working OK, even cloning from OMR VPS to freshly created VPS that had never OMR installed.
> 
> But again, It was several years ago, so I am not sure how effective it would be now, but it is worth to try
I will give a try, thanks.

> OMR support multiples servers. It's working correctly in snapshot release and it retrieve missing keys when needed using VPS API.

I tried to configure a second server, but OMR didn't connect to it even when I turned off the main one. In the log, it only said that OMR could ping, but it couldn't connect to the API of the second server. When I configured the second server as the main one, OMR could connect normally. I don't know if this could be caused by the fact that I use DNS names instead of IPs to connect to the servers.
I tested this in the snapshot from last week.

Ysurac, 2023-11-29:
This should work on latest snapshot with latest VPS script.
What is the proxy used ?
I would need system log and, via SSH on the router, the result of ```uci show openmptcprouter``` (you can obfuscate IP or/and keys).

vempire-ghost, 2023-11-29:
I configured a new VPS, left the master off, configured the second VPS on OMR. Here's the log.
Nov 29 21:57:33 OpenMPTCProuter user.notice omr-tracker: Launching...
Nov 29 21:57:44 OpenMPTCProuter user.notice omr-tracker: Launched
Nov 29 21:57:44 OpenMPTCProuter daemon.info vnstatd[8463]: Info: SIGTERM received, exiting.
Nov 29 21:57:44 OpenMPTCProuter daemon.info vnstatd[9899]: Info: vnStat daemon 2.9 started. (pid:9899 uid:0 gid:0 64-bit)
Nov 29 21:57:44 OpenMPTCProuter daemon.info vnstatd[9899]: Info: Monitoring (4): eth4 (1000 Mbit) eth3 (1000 Mbit) eth2 (1000 Mbit) eth1 (1000 Mbit) 
Nov 29 21:57:44 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Nov 29 21:57:44 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Nov 29 21:57:44 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Nov 29 21:57:44 OpenMPTCProuter user.notice xray: add rules
Nov 29 21:57:44 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov 29 21:57:44 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov 29 21:57:44 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov 29 21:57:45 OpenMPTCProuter daemon.info xray[11291]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.4 linux/amd64)
Nov 29 21:57:45 OpenMPTCProuter daemon.info xray[11291]: A unified platform for anti-censorship.
Nov 29 21:57:45 OpenMPTCProuter daemon.info xray[11291]: 2023/11/29 21:57:45 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Nov 29 21:57:45 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:57:45 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:57:45 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:57:45 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:57:45 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:57:45 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:57:45 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:57:45 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:57:45 OpenMPTCProuter user.notice SQM-autorate: Set config for omrvpn
Nov 29 21:57:46 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on server1.ddns.net, no server API answer on server1.ddns.net)
Nov 29 21:58:02 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later (can't ping server vps_us on server2.ddns.net, no server API answer on server2.ddns.net)
Nov 29 21:58:02 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Nov 29 21:58:02 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Nov 29 21:58:02 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Nov 29 21:58:02 OpenMPTCProuter user.notice xray: add rules
Nov 29 21:58:02 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov 29 21:58:02 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov 29 21:58:02 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov 29 21:58:03 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov 29 21:58:03 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov 29 21:58:05 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Nov 29 21:58:06 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:58:06 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:58:06 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:58:06 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:58:06 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:58:06 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:58:06 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:58:06 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Nov 29 21:58:06 OpenMPTCProuter user.notice SQM-autorate: Set config for omrvpn
Nov 29 21:58:07 OpenMPTCProuter user.notice omr-tracker: Launching...
Nov 29 21:58:18 OpenMPTCProuter user.notice omr-tracker: Launched
Nov 29 21:58:34 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on server1.ddns.net, no server API answer on server1.ddns.net)
Nov 29 21:58:50 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later (can't ping server vps_us on server2.ddns.net, no server API answer on server2.ddns.net)
Nov 29 21:58:50 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Nov 29 21:58:50 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Nov 29 21:58:50 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Nov 29 21:58:50 OpenMPTCProuter user.notice xray: add rules
Nov 29 21:58:50 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Nov 29 21:58:50 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov 29 21:58:51 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov 29 21:58:51 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov 29 21:58:51 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
[... cut ...]
openmptcprouter.latest_versions.lc='1701295036'
openmptcprouter.vps_us=server
openmptcprouter.vps_us.username='openmptcprouter'
openmptcprouter.vps_us.get_config='0'
openmptcprouter.vps_us.master='0'
openmptcprouter.vps_us.backup='1'
openmptcprouter.vps_us.password=''
openmptcprouter.vps_us.ip='server2.ddns.net'
openmptcprouter.vps_us.port='65500'
openmptcprouter.vps_us.set_firewall='1'
openmptcprouter.vps_us.admin_error='1'
openmptcprouter.vps_us.current='1'

I am using xray as a proxy and none as a vpn.

vempire-ghost, 2023-11-29:
Configuring the backup as the master makes it work. Then, I turned off VPS2 and turned on VPS1, which is now configured as the backup, and OMR can't connect to it. 
Here's the log.
Nov 29 22:10:36 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps1.ddns.net, no server API answer on vps1.ddns.net)
Nov 29 22:10:37 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps_us down (xxx.xxx.xxx.xxx)
Nov 29 22:10:45 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1)
Nov 29 22:10:52 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later (can't ping server vps_us on vps2.ddns.net, no server API answer on vps2.ddns.net)
Nov 29 22:11:02 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Nov 29 22:11:12 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later
Nov 29 22:11:12 OpenMPTCProuter user.notice xray: Rules DOWN
Nov 29 22:11:23 OpenMPTCProuter daemon.info omr-tracker-xray: Server (xxx.xxx.xxx.xxx) seems down, no answer to ping
Nov 29 22:11:29 OpenMPTCProuter daemon.notice netifd: Network device 'eth3' link is down
Nov 29 22:11:29 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity loss
Nov 29 22:11:33 OpenMPTCProuter daemon.notice netifd: Network device 'eth3' link is up
Nov 29 22:11:33 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity 
Nov 29 22:12:10 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps_us down (xxx.xxx.xxx.xxx)
Nov 29 22:12:11 OpenMPTCProuter user.notice OMR-Tracker-Server: Use backup server vps1.ddns.net (xxx.xxx.xxx.xxx)
Nov 29 22:12:51 OpenMPTCProuter daemon.err uhttpd[8269]: error: couldn't find the ModemManager process in the bus
Nov 29 22:13:00 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps_us down (xxx.xxx.xxx.xxx)
Nov 29 22:13:02 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Nov 29 22:13:03 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Nov 29 22:13:03 OpenMPTCProuter user.notice omr-tracker: Launching...
Nov 29 22:13:14 OpenMPTCProuter user.notice omr-tracker: Launched
Nov 29 22:13:41 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later (can't ping server vps_us on vps2.ddns.net, no server API answer on vps2.ddns.net)
Nov 29 22:13:57 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later (can't ping server vps_us on vps2.ddns.net, no server API answer on vps2.ddns.net)
Nov 29 22:13:58 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps_us down (xxx.xxx.xxx.xxx)
Nov 29 22:14:14 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later (can't ping server vps_us on vps2.ddns.net, no server API answer on vps2.ddns.net)

vempire-ghost, 2023-12-02:
I tested today's snapshot. To make the backup work, I had to first configure it as the master, connect, then change it to backup and wait for about 3 minutes until OMR connects to it. However, the status in the Network overview shows as if there were an error.

[image]

Ysurac, 2023-12-02:
The API is still working on the server ? What is the result of ```ps aux | grep omr``` and ```journalctl -u omr-admin``` ?

vempire-ghost, 2023-12-04:
vempire@instance-5:~$ ps aux | grep omr
root         446  0.6  8.4 191904 85624 ?        Ssl  21:09   0:02 python3 /usr/local/bin/omr-admin.py --host=::
root        1199  0.2  0.3   4596  3532 ?        Ss   21:09   0:00 /bin/bash /usr/local/bin/omr-service
root        1250  0.0  0.0      0     0 ?        I<   21:09   0:00 [omr-bonding]
root        1765  0.6  8.5 268940 86652 ?        Ssl  21:09   0:02 python3 /usr/local/bin/omr-admin.py
vempire    15243  0.0  0.9  33136  9656 pts/0    T    21:16   0:00 journalctl -u omr-admin
vempire    16660  0.0  0.1   3736  1996 pts/0    S+   21:16   0:00 grep omr

vempire@instance-5:~$ journalctl -u omr-admin
Nov 29 16:57:25 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Started omr-admin.service - OMR-Admin.
Nov 29 16:58:58 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Stopping omr-admin.service - OMR-Admin...
Nov 29 16:58:58 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: omr-admin.service: Deactivated successfully.
Nov 29 16:58:58 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Stopped omr-admin.service - OMR-Admin.
Nov 29 16:58:58 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: omr-admin.service: Consumed 2.718s CPU time.
-- Boot b05844d7d33a4b36a3ec7eac10a7fe50 --
Nov 29 21:56:22 localhost systemd[1]: Started omr-admin.service - OMR-Admin.
Nov 29 22:10:00 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Stopping omr-admin.service - OMR-Admin...
Nov 29 22:10:01 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: omr-admin.service: Deactivated successfully.
Nov 29 22:10:01 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Stopped omr-admin.service - OMR-Admin.
Nov 29 22:10:01 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: omr-admin.service: Consumed 5.988s CPU time.
-- Boot d0c12e868e344c1bb5b827858c330511 --
Dec 02 01:46:00 localhost systemd[1]: Started omr-admin.service - OMR-Admin.
Dec 02 01:46:15 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Stopping omr-admin.service - OMR-Admin...
Dec 02 01:46:15 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: omr-admin.service: Deactivated successfully.
Dec 02 01:46:15 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Stopped omr-admin.service - OMR-Admin.
Dec 02 01:46:15 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Started omr-admin.service - OMR-Admin.
Dec 02 02:01:25 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Stopping omr-admin.service - OMR-Admin...
Dec 02 02:01:27 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: omr-admin.service: Deactivated successfully.
Dec 02 02:01:27 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Stopped omr-admin.service - OMR-Admin.
Dec 02 02:01:27 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: omr-admin.service: Consumed 6.377s CPU time.
-- Boot 08fdfe7dc73448299bbe04154f5736a4 --
Dec 04 21:09:15 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Started omr-admin.service - OMR-Admin.
Dec 04 21:09:25 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Stopping omr-admin.service - OMR-Admin...
Dec 04 21:09:25 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: omr-admin.service: Deactivated successfully.
Dec 04 21:09:25 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Stopped omr-admin.service - OMR-Admin.
Dec 04 21:09:25 instance-5.us-east1-b.c.helical-fin-392321.internal systemd[1]: Started omr-admin.service - OMR-Admin.

He is the log form router
Dec  4 21:09:19 OpenMPTCProuter user.notice omr-tracker: Launched
Dec  4 21:09:19 OpenMPTCProuter daemon.info vnstatd[8229]: Info: SIGTERM received, exiting.
Dec  4 21:09:19 OpenMPTCProuter daemon.info vnstatd[2218]: Info: vnStat daemon 2.9 started. (pid:2218 uid:0 gid:0 64-bit)
Dec  4 21:09:19 OpenMPTCProuter daemon.info vnstatd[2218]: Info: Monitoring (4): eth4 (1000 Mbit) eth3 (1000 Mbit) eth2 (1000 Mbit) eth1 (1000 Mbit) 
Dec  4 21:09:19 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Dec  4 21:09:19 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Dec  4 21:09:19 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Dec  4 21:09:19 OpenMPTCProuter user.notice xray: add rules
Dec  4 21:09:19 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Dec  4 21:09:19 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Dec  4 21:09:20 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Dec  4 21:09:20 OpenMPTCProuter daemon.info xray[3626]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.4 linux/amd64)
Dec  4 21:09:20 OpenMPTCProuter daemon.info xray[3626]: A unified platform for anti-censorship.
Dec  4 21:09:20 OpenMPTCProuter daemon.info xray[3626]: 2023/12/04 21:09:20 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Dec  4 21:09:20 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:09:20 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:09:20 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:09:20 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:09:21 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
[... cut ...]
Dec  4 21:13:50 OpenMPTCProuter daemon.info xray[11639]: 2023/12/04 21:13:50 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Dec  4 21:13:53 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164)
Dec  4 21:13:53 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Dec  4 21:13:56 OpenMPTCProuter user.notice xray: Rules DOWN
Dec  4 21:14:29 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)

vempire-ghost, 2023-12-04:
They only connect if I set it as a master.
Dec  4 21:17:52 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)
Dec  4 21:18:43 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)
Dec  4 21:19:08 OpenMPTCProuter daemon.err uhttpd[8038]: error: couldn't find the ModemManager process in the bus
Dec  4 21:19:15 OpenMPTCProuter user.notice omr-tracker: Launching...
Dec  4 21:19:16 OpenMPTCProuter user.notice SQM-autorate: Set config for omrvpn
Dec  4 21:19:16 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:16 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:16 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:16 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:16 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:16 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:16 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:16 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:26 OpenMPTCProuter user.notice omr-tracker: Launched
Dec  4 21:19:26 OpenMPTCProuter user.notice omr-tracker: Launching...
Dec  4 21:19:26 OpenMPTCProuter daemon.info vnstatd[2218]: Info: SIGTERM received, exiting.
Dec  4 21:19:26 OpenMPTCProuter daemon.info vnstatd[29295]: Info: vnStat daemon 2.9 started. (pid:29295 uid:0 gid:0 64-bit)
Dec  4 21:19:26 OpenMPTCProuter daemon.info vnstatd[29295]: Info: Monitoring (4): eth4 (1000 Mbit) eth3 (1000 Mbit) eth2 (1000 Mbit) eth1 (1000 Mbit) 
Dec  4 21:19:26 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Dec  4 21:19:27 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Dec  4 21:19:27 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Dec  4 21:19:27 OpenMPTCProuter user.notice xray: add rules
Dec  4 21:19:27 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Dec  4 21:19:27 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Dec  4 21:19:27 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Dec  4 21:19:27 OpenMPTCProuter daemon.info xray[30691]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.4 linux/amd64)
Dec  4 21:19:27 OpenMPTCProuter daemon.info xray[30691]: A unified platform for anti-censorship.
Dec  4 21:19:27 OpenMPTCProuter daemon.info xray[30691]: 2023/12/04 21:19:27 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Dec  4 21:19:27 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:27 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:27 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:27 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:27 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:27 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:27 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:27 OpenMPTCProuter daemon.err modprobe: failed to find a module named act_ipt
Dec  4 21:19:27 OpenMPTCProuter user.notice SQM-autorate: Set config for omrvpn
Dec  4 21:19:38 OpenMPTCProuter user.notice omr-tracker: Launched
Dec  4 21:19:38 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Dec  4 21:19:38 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Dec  4 21:19:38 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Dec  4 21:19:38 OpenMPTCProuter user.notice xray: add rules
Dec  4 21:19:38 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Dec  4 21:19:38 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Dec  4 21:19:38 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Dec  4 21:19:39 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps_us up (X.X.X.X), set it back
Dec  4 21:19:42 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddns.net, no server API answer on vps.ddns.net)
Dec  4 21:19:44 OpenMPTCProuter user.notice OMR-VPS: Get config from server vps_us...
Dec  4 21:19:44 OpenMPTCProuter user.notice OMR-VPS: V2ray restart...
Dec  4 21:19:44 OpenMPTCProuter user.notice OMR-VPS: Xray restart...
Dec  4 21:19:44 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Dec  4 21:19:44 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Dec  4 21:19:44 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Dec  4 21:19:44 OpenMPTCProuter user.notice xray: add rules
Dec  4 21:19:44 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Dec  4 21:19:44 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Dec  4 21:19:45 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Dec  4 21:19:45 OpenMPTCProuter daemon.info xray[6931]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.4 linux/amd64)
Dec  4 21:19:45 OpenMPTCProuter daemon.info xray[6931]: A unified platform for anti-censorship.
Dec  4 21:19:45 OpenMPTCProuter daemon.info xray[6931]: 2023/12/04 21:19:45 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Dec  4 21:19:45 OpenMPTCProuter user.notice OMR-VPS: Glorytun restart...
Dec  4 21:19:45 OpenMPTCProuter user.notice OMR-VPS: Glorytun UDP restart...
Dec  4 21:19:45 OpenMPTCProuter user.notice OMR-VPS: OpenVPN restart...
Dec  4 21:19:45 OpenMPTCProuter user.notice OMR-VPS: MLVPN restart...
Dec  4 21:19:45 OpenMPTCProuter user.notice OMR-VPS: DSVPN restart...
Dec  4 21:19:49 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 198.27.92.1)

vempire-ghost, 2023-12-14:
How do I add a backup VPS using uci commands?
I tried this commands but didn't worked.
uci set openmptcprouter.vps_us.ip=
uci set openmptcprouter.vps_us.password=

Ysurac, 2023-12-14:
```
uci set openmptcprouter.vps_us=server
uci set openmptcprouter.vps_us.ip=
uci set openmptcprouter.vps_us.password=
uci set openmptcprouter.vps_us.backup=1
uci set openmptcprouter.vps_us.master=0
uci set openmptcprouter.vps_us.current=0
uci set openmptcprouter.vps_us.port="65500"
uci set openmptcprouter.vps_us.get_config=0
uci commit openmptcprouter
/etc/init.d/openmtpcprouter-vps restart
```

vempire-ghost, 2023-12-28:
When connecting using the backupt vps with the master vps offline, the omr work fine but has this bug in the Status Page.
[image]
I think the omr try to take this info only in the master vps and because its down, they show this.

Ysurac, 2023-12-28:
Should not be the case.
What is the result of ```uci show openmptcprouter```  (remove password and IPs) ?

vempire-ghost, 2023-12-28:
root@OpenMPTCProuter:~# uci show openmptcprouter
openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.sipalg='1'
openmptcprouter.settings.master='balancing'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.check_ipv4_website='http://ip.openmptcprouter.com'
openmptcprouter.settings.check_ipv6_website='http://ipv6.openmptcprouter.com'
openmptcprouter.settings.status_vps_timeout='1'
openmptcprouter.settings.status_getip_timeout='1'
openmptcprouter.settings.enable_nodelay='1'
openmptcprouter.settings.scaling_governor='performance'
openmptcprouter.settings.menu='OpenMPTCProuter'
openmptcprouter.settings.disableintfrename='1'
openmptcprouter.settings.tracebox='0'
openmptcprouter.settings.version='0.60beta1-5.4'
openmptcprouter.settings.serial='0'
openmptcprouter.settings.allmptcpovervpn='0'
openmptcprouter.settings.disable_fastopen='0'
openmptcprouter.settings.external_check='1'
openmptcprouter.settings.debug='0'
openmptcprouter.settings.disablegwping='0'
openmptcprouter.settings.disableloopdetection='0'
openmptcprouter.settings.disableserverhttptest='0'
openmptcprouter.settings.defaultgw='1'
openmptcprouter.settings.modemmanager='1'
openmptcprouter.settings.disableserverping='0'
openmptcprouter.settings.shadowsocksudp='0'
openmptcprouter.settings.ndpi='1'
openmptcprouter.settings.sfe_enabled='0'
openmptcprouter.settings.sfe_bridge='0'
openmptcprouter.settings.vnstat_backup='0'
openmptcprouter.settings.mptcpovervpn='wireguard'
openmptcprouter.settings.dns64='0'
openmptcprouter.settings.proxy='xray-trojan'
openmptcprouter.settings.ha='1'
openmptcprouter.settings.encryption='aes-256-gcm'
openmptcprouter.settings.vpn='none'
openmptcprouter.settings.firstboot='0'
openmptcprouter.settings.apilc='1703735948'
openmptcprouter.vps=server
openmptcprouter.vps.username='openmptcprouter'
openmptcprouter.vps.master='1'
openmptcprouter.vps.backup='0'
openmptcprouter.vps.port='65500'
openmptcprouter.vps.password=''
openmptcprouter.vps.redirect_ports='0'
openmptcprouter.vps.nofwredirect='0'
openmptcprouter.vps.wgkey=''
openmptcprouter.vps.available_vpn='glorytun_tcp' 'glorytun_udp' 'dsvpn' 'openvpn' 'openvpn_bonding' 'mlvpn'
openmptcprouter.vps.available_proxy='shadowsocks' 'shadowsocks-go' 'v2ray' 'v2ray-vmess' 'v2ray-socks' 'v2ray-trojan' 'xray' 'xray-vless-reality' 'xray-vmess' 'xray-socks' 'xray-trojan' 'xray-shadowsocks'
openmptcprouter.vps.pihole='0'
openmptcprouter.vps.kernel='5.4.207-mptcp'
openmptcprouter.vps.machine='x86_64'
openmptcprouter.vps.omr_version='0.1029-test'
openmptcprouter.vps.lastchange='1703735950'
openmptcprouter.vps.set_firewall='1'
openmptcprouter.vps.admin_error='1'
openmptcprouter.vps.token_error='1'
openmptcprouter.vps.current='1'
openmptcprouter.vps.ip=''
openmptcprouter.vps.get_config='1'
openmptcprouter.omr=router
openmptcprouter.omr.detected_public_ipv4=''
openmptcprouter.omr.xray='up'
openmptcprouter.loopback=interface
openmptcprouter.loopback.multipath='off'
openmptcprouter.loopback.metric='7'
openmptcprouter.lan=interface
openmptcprouter.lan.multipath='off'
openmptcprouter.lan.metric='8'
openmptcprouter.lan.multipathvpn='0'
openmptcprouter.lan.testspeed='0'
openmptcprouter.wan1=interface
openmptcprouter.wan1.multipath='master'
openmptcprouter.wan1.metric='9'
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.multipathvpn='0'
openmptcprouter.wan1.testspeed='0'
openmptcprouter.wan1.publicip=''
openmptcprouter.wan1.mptcp_status='MPTCP enabled'
openmptcprouter.wan1.manufacturer='huawei'
openmptcprouter.wan1.lc='1703777053'
openmptcprouter.wan2=interface
openmptcprouter.wan2.multipath='on'
openmptcprouter.wan2.metric='10'
openmptcprouter.wan2.state='up'
openmptcprouter.wan2.publicip=''
openmptcprouter.wan2.mptcp_status='MPTCP enabled'
openmptcprouter.wan2.manufacturer='huawei'
openmptcprouter.wan2.multipathvpn='0'
openmptcprouter.wan2.testspeed='0'
openmptcprouter.wan3=interface
openmptcprouter.wan3.multipath='on'
openmptcprouter.wan3.metric='11'
openmptcprouter.wan3.state='up'
openmptcprouter.wan3.multipathvpn='0'
openmptcprouter.wan3.testspeed='0'
openmptcprouter.wan3.publicip='4'
openmptcprouter.wan3.mtu='1480'
openmptcprouter.wan3.mptcp_status='MPTCP enabled'
openmptcprouter.wan3.manufacturer='huawei'
openmptcprouter.wan3.lc='1703780556'
openmptcprouter.wan4=interface
openmptcprouter.wan4.multipath='on'
openmptcprouter.wan4.metric='12'
openmptcprouter.wan4.state='up'
openmptcprouter.wan4.mptcp_status='MPTCP enabled'
openmptcprouter.wan4.manufacturer='huawei'
openmptcprouter.wan4.multipathvpn='0'
openmptcprouter.wan4.testspeed='0'
openmptcprouter.wan4.publicip=''
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.multipath='off'
openmptcprouter.omrvpn.metric='1500'
openmptcprouter.omr6in4=interface
openmptcprouter.omr6in4.multipath='off'
openmptcprouter.omr6in4.metric='1201'
openmptcprouter.vps_us=server
openmptcprouter.vps_us.password=''
openmptcprouter.vps_us.backup='1'
openmptcprouter.vps_us.master='0'
openmptcprouter.vps_us.port='65500'
openmptcprouter.vps_us.username='openmptcprouter'
openmptcprouter.vps_us.redirect_ports='0'
openmptcprouter.vps_us.nofwredirect='0'
openmptcprouter.vps_us.get_config='0'
openmptcprouter.vps_us.token_error='0'
openmptcprouter.vps_us.pihole='0'
openmptcprouter.vps_us.kernel='5.4.207-mptcp'
openmptcprouter.vps_us.machine='x86_64'
openmptcprouter.vps_us.omr_version='0.1029-test'
openmptcprouter.vps_us.ip=''
openmptcprouter.vps_us.current='1'
openmptcprouter.vps_us.set_firewall='0'
openmptcprouter.vps_us.admin_error='0'
openmptcprouter.vps_us.token=''
openmptcprouter.vps_us.lastchange='1703780568'
openmptcprouter.latest_versions=latest_versions
openmptcprouter.latest_versions.omr='0.59.1-5.4'
openmptcprouter.latest_versions.vps='0.1028'
openmptcprouter.latest_versions.lc='1703777644'