Ysurac/openmptcprouter #3053: Wireguard key is not retrieved from server part

klucsik, 2023-11-29:
## Expected Behavior
In Settings wizard, I check the MPTCP over VPN checkbox, leave the advanced settings on default (wireguard as VPN)
The WG interface is created at both sides, and populated correctly. After a few seconds, the interface up and running (green in the status page)

## Current Behavior
The WG interface peer public key in router part is not populated from the server. If I copy it manually from the server, then the interface start up and everything is working like a charm.

## Steps to Reproduce the Problem

  1. Setup an WAN interface with MPTCP over VPN checked in
  2. (I also setted up a regular interface, so the rtouter and admin part could communicate)

## Context (Environment)
One of the ISP blocks mptcp, I want to change to mptcp over vpn

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta1-6.1 r0+23789-ce6ad123e7 , openmptcprouter-v0.59.1-5.4-r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version:   0.1029-test (also present on stablem I dont have version number for that, sorry)
  - OpenMPTCProuter VPS provider: pq.hosting,https://alpenhost.at/
  - OpenMPTCProuter platform: RPI4
  - Country: Austria

Ysurac, 2023-11-30:
I'm not able to reproduce the issue.
Can you give me a copy of System->OpenMPTCProuter, "status" page and of Status->System log ?

klucsik, 2023-12-05:
Yes, sorry for late reply
[image]

https://pastebin.com/J528Mjky

I have issues on my other network too, maybe that can contribute to the problem

klucsik, 2023-12-05:
I switched both of my network to use VPN and after a few minutes, it seems to be working... probably this issue is at my side

klucsik, 2023-12-05:
now I removed VPN on both connection, and it works fine https://pastebin.com/9hG4T0u0
I dont know for how long though :/

klucsik, 2023-12-07:
Cant reproduce, probably was network error

closed 2023-12-07 by Ysurac