Ysurac/openmptcprouter #3055: update VPS 5.4 to 6.1

yannick2801, 2023-11-30:
Hi

This command below does not seems to work. (séame with sudo before the command)

Debian11
VPS : Linux vps-3ed9bf51 5.4.207-mptcp #1 SMP Sun Jul 24 14:39:44 UTC 2022 x86_64 GNU/Linux

For kernel 6.1 You need to also use router part with kernel 6.1, it's not the default image:
`wget -O - http://www.openmptcprouter.com/server-test/debian-x86_64.sh | UPSTREAM6="yes" sh`

BR
Yannick

Ysurac, 2023-11-30:
What is the error ?

yannick2801, 2023-12-01:
if i launch all the command : 

[image]

if i only launch the first part : `sudo wget -O - http://www.openmptcprouter.com/server-test/debian-x86_64.sh`

[image]

it does not download any files.

BR
Yannick

Ysurac, 2023-12-01:
You need to run it as root, do a ```sudo -s``` or ```su``` before running the command.

yannick2801, 2023-12-01:
Worked perfect with `sudo -s`

Thanks
BR

yannick2801, 2023-12-01:
The VPS server does not seems to work in the version : Linux vps-3ed9bf51 6.1.0-13-cloud-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.55-1 (2023-09-29) x86_64 GNU/Linux

[image]

BR
Yannick

Ysurac, 2023-12-01:
It's the good kernel for 6.1.
Check Status->System log for why it can't contact API on server.

Ysurac, 2023-12-01:
So it's not able to ping VPS, it's not good. VPS IP is pingable from another connection ? 
If yes, what is the result of ```ip r``` via SSH on the router ? Can you also make a full screenshot of Status page ?

yannick2801, 2023-12-01:
i can connect on my OVH serveur but i can't ping it

this is the result of `ip r`
[image]

Ysurac, 2023-12-01:
You tested ping from another connection (or via ```ping -B -I wwan1 <vpsip>```) ? You have a custom firewall on it or in front of it ?

yannick2801, 2023-12-01:
here is the log systeme : 

`Nov 30 16:07:34 OpenMPTCProuter daemon.info vnstatd[3980]: Info: Data retention: 48 5MinuteHours, 4 HourlyDays, 62 DailyDays, 25 MonthlyMonths, -1 YearlyYears, 20 TopDayEntries
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/network
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/network
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/glorytun-udp reload dependency on /etc/config/network
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/dsvpn reload dependency on /etc/config/network
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/mptcpovervpn reload dependency on /etc/config/network
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/omr6in4 reload dependency on /etc/config/network
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/omr-quota reload dependency on /etc/config/network
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/omr-tracker reload dependency on /etc/config/network
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/openmptcprouter reload dependency on /etc/config/network
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/wireless
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/luci-splash reload dependency on /etc/config/firewall
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/qos reload dependency on /etc/config/firewall
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/miniupnpd reload dependency on /etc/config/firewall
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/sqm reload dependency on /etc/config/firewall
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/sqm-autorate reload dependency on /etc/config/firewall
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/odhcpd reload dependency on /etc/config/dhcp
Nov 30 16:07:35 OpenMPTCProuter user.notice unbound: default protocol configuration
Nov 30 16:07:35 OpenMPTCProuter user.notice unbound: default memory configuration
Nov 30 16:07:35 OpenMPTCProuter user.notice unbound: default recursion configuration
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up non-init /etc/config/fstab reload handler: /sbin/block mount
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/system reload trigger for non-procd /etc/init.d/led
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/luci_statistics reload dependency on /etc/config/system
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/system
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/glorytun-udp reload dependency on /etc/config/glorytun
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/openmptcprouter-vps reload dependency on /etc/config/glorytun
Nov 30 16:07:35 OpenMPTCProuter daemon.notice netifd: Network device 'eth0' link is up
Nov 30 16:07:35 OpenMPTCProuter daemon.notice netifd: Interface 'lan' has link connectivity 
Nov 30 16:07:35 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/openmptcprouter-vps reload dependency on /etc/config/glorytun-udp
Nov 30 16:07:36 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/omr-tracker reload dependency on /etc/config/shadowsocks-libev
Nov 30 16:07:36 OpenMPTCProuter user.notice ucitrack: Setting up /etc/config/mlvpn reload trigger for non-procd /etc/init.d/mlvpn
Nov 30 16:07:37 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of loopback (lo)
Nov 30 16:07:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Nov 30 16:07:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Nov 30 16:07:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Nov 30 16:07:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Nov 30 16:07:37 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S90omr-dscp: iptables: No chain/target/match by that name.
Nov 30 16:07:41 OpenMPTCProuter user.notice omr-tracker: Launching...
Nov 30 16:07:41 OpenMPTCProuter daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Nov 30 16:07:41 OpenMPTCProuter daemon.err procd: Got unexpected signal 1
Nov 30 16:07:41 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Nov 30 16:07:41 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Nov 30 16:07:41 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Nov 30 16:07:41 OpenMPTCProuter daemon.warn dnsmasq[1]: warning: no upstream servers configured
Nov 30 16:07:41 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.248, lease time 12h
[... cut ...]
Nov 30 16:14:16 OpenMPTCProuter daemon.info /usr/bin/ss-redir[19999]: UDP relay enabled
Nov 30 16:14:16 OpenMPTCProuter daemon.info /usr/bin/ss-redir[19999]: udp port reuse enabled
Nov 30 16:14:16 OpenMPTCProuter daemon.info /usr/bin/ss-redir[19999]: running from root user
Nov 30 16:14:18 OpenMPTCProuter user.notice post-tracking-020-status: New public ip detected for QUECTEL2 (wwan2): 93.23.14.215 (previous: 93.23.13.73)
Nov 30 16:14:18 OpenMPTCProuter user.notice post-tracking-020-status: Reload MPTCP for wwan2`

BR

Ysurac, 2023-12-01:
I still need full screenshot of System->OpenMPTCProuter, "status" page.
What is the platform used as router ?

Ysurac, 2023-12-01:
I pushed a new commit that may solve your issue (at least a part of it), snapshot is building now: https://github.com/Ysurac/openmptcprouter-feeds/actions/runs/7065594894

yannick2801, 2023-12-04:
Hello,

i use a Radxa Rock pi 4 C+

[image]

BR
Yannick

yannick2801, 2023-12-04:
i have reinstalled on an other SD_Card :

yannick2801, 2023-12-04:
i have tried all of them but no changes : 

[image]

Ysurac, 2023-12-04:
As it's an image on an unsupported device, what is the result of ```uname -a``` ?
Can you take also a screenshot of Network->Interfaces page ?

yannick2801, 2023-12-05:
yes I know, I compiled an image from the develop branch in Kernel 6.1 because I had problems with ModemManager at startup.

`root@OpenMPTCProuter:~# uname -a
Linux OpenMPTCProuter 6.1.55 #0 SMP PREEMPT Mon Sep 25 21:21:09 2023 aarch64 GNU/Linux`

[image]

Ysurac, 2023-12-05:
Configuration seems to be ok.
Did you try with only one modem ?
Can you also try a ```ping -B -I wwan0 <server_ip>``` ? If this doesn't work, can you try a ```ping -B -I wwan0 1.1.1.1``` ? If this also doesn't work, than ping can be blocked.
Can you also try a ```curl -k -4 --interface wwan0 https://<server_ip>:65500/``` ?

yannick2801, 2023-12-05:
this worked : 

[image]

yannick2801, 2023-12-05:
this workes fine : 

[image]

Ysurac, 2023-12-05:
As requested, did you try with only one modem enabled ? (You can disable them by disabling MPTCP on the interfaces in Network->MPTCP)

Ysurac, 2023-12-06:
Strange that it can't contact DNS servers and the API...
What is the result of ```uci show openmptcprouter``` on the router via SSH ?

yannick2801, 2023-12-06:
root@OpenMPTCProuter:~# uci show openmptcprouter

openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.sipalg='1'
openmptcprouter.settings.master='balancing'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.check_ipv4_website='http://ip.openmptcprouter.com'
openmptcprouter.settings.check_ipv6_website='http://ipv6.openmptcprouter.com'
openmptcprouter.settings.status_vps_timeout='1'
openmptcprouter.settings.status_getip_timeout='1'
openmptcprouter.settings.enable_nodelay='1'
openmptcprouter.settings.scaling_governor='performance'
openmptcprouter.settings.menu='OpenMPTCProuter'
openmptcprouter.settings.disableintfrename='1'
openmptcprouter.settings.tracebox='0'
openmptcprouter.settings.version='0.60beta1-6.1'
openmptcprouter.settings.allmptcpovervpn='0'
openmptcprouter.settings.vpn='openvpn'
openmptcprouter.settings.mptcpovervpn='wireguard'
openmptcprouter.settings.dns64='0'
openmptcprouter.settings.proxy='shadowsocks'
openmptcprouter.settings.ha='0'
openmptcprouter.settings.encryption='aes-256-gcm'
openmptcprouter.settings.firstboot='0'
openmptcprouter.settings.apilc='1701501543'
openmptcprouter.settings.country='world'
openmptcprouter.vps=server
openmptcprouter.vps.username='openmptcprouter'
openmptcprouter.vps.master='1'
openmptcprouter.vps.backup='0'
openmptcprouter.vps.port='65500'
openmptcprouter.vps.set_firewall='0'
openmptcprouter.vps.get_config='0'
openmptcprouter.vps.current='1'
openmptcprouter.vps.password='DE90A867177437388433769133D63862216F1ADB1C4B7456447FDEDACC7CF50F'
openmptcprouter.vps.admin_error='0'
openmptcprouter.vps.redirect_ports='0'
openmptcprouter.vps.wgkey='N+WOO6jvYTUg/u3EX4CLmVo8G3dQ5H+svBruBUxku2s='
openmptcprouter.vps.ipv6='2001:41d0:401:3200::3266'
openmptcprouter.vps.pihole='0'
openmptcprouter.vps.kernel='6.1.0-13-cloud-amd64'
openmptcprouter.vps.machine='x86_64'
openmptcprouter.vps.omr_version='0.1029-test'
openmptcprouter.vps.token_error='1'
openmptcprouter.vps.lastchange='1701501551'
openmptcprouter.vps.ip='217.182.129.213'
openmptcprouter.vps.available_vpn='glorytun_tcp' 'glorytun_udp' 'dsvpn' 'openvpn' 'openvpn_bonding' 'mlvpn'
openmptcprouter.vps.token='eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJzdWIiOiJvcGVubXB0Y3Byb3V0ZXIiLCJleHAiOjE3MDE5Mzk3MDZ9.dq5Ws0iELFa5yka9CFqqjAoiWLH0fGvCsLc1v87wRw0'
openmptcprouter.omr=router
openmptcprouter.omr.ss_sss0='up'
openmptcprouter.loopback=interface
openmptcprouter.loopback.multipath='off'
openmptcprouter.loopback.metric='5'
openmptcprouter.lan=interface
openmptcprouter.lan.multipath='off'
openmptcprouter.lan.metric='6'
openmptcprouter.lan.multipathvpn='0'
openmptcprouter.lan.testspeed='0'
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.multipath='off'
openmptcprouter.omrvpn.metric='1500'
openmptcprouter.omrvpn.publicip='10.255.252.6'
openmptcprouter.omrvpn.mptcp_status='MPTCP enabled'
openmptcprouter.omrvpn.state='up'
openmptcprouter.omrvpn.lc='1701501541'
openmptcprouter.omr6in4=interface
openmptcprouter.omr6in4.multipath='off'
openmptcprouter.omr6in4.metric='1201'
openmptcprouter.QUECTEL1=interface
openmptcprouter.QUECTEL1.multipath='on'
openmptcprouter.QUECTEL1.metric='7'
openmptcprouter.QUECTEL1.multipathvpn='0'
openmptcprouter.QUECTEL1.testspeed='0'
openmptcprouter.QUECTEL1.state='up'
openmptcprouter.QUECTEL1.publicip='92.184.119.122'
openmptcprouter.QUECTEL1.latency='124'
openmptcprouter.QUECTEL1.mptcp_status='MPTCP enabled'
openmptcprouter.QUECTEL1.lc='1701499390'
openmptcprouter.QUECTEL1.latency_previous='38'
openmptcprouter.QUECTEL2=interface
openmptcprouter.QUECTEL2.multipath='master'
openmptcprouter.QUECTEL2.metric='10'
openmptcprouter.QUECTEL2.multipathvpn='0'
openmptcprouter.QUECTEL2.testspeed='0'
openmptcprouter.QUECTEL2.state='up'
openmptcprouter.QUECTEL2.publicip='93.22.150.154'
openmptcprouter.QUECTEL2.latency='56'
openmptcprouter.QUECTEL2.mptcp_status='MPTCP enabled'
openmptcprouter.QUECTEL2.latency_previous='54'
openmptcprouter.QUECTEL2.lc='1701498895'
openmptcprouter.QUECTEL3=interface
openmptcprouter.QUECTEL3.multipath='on'
openmptcprouter.QUECTEL3.metric='11'
openmptcprouter.QUECTEL3.multipathvpn='0'
openmptcprouter.QUECTEL3.testspeed='0'
openmptcprouter.QUECTEL3.state='up'
openmptcprouter.QUECTEL3.publicip='176.179.156.42'
openmptcprouter.QUECTEL3.latency='109'
openmptcprouter.QUECTEL3.mptcp_status='MPTCP enabled'
openmptcprouter.QUECTEL3.latency_previous='134'
openmptcprouter.QUECTEL3.lc='1701499388'
root@OpenMPTCProuter:~#

ccmks, 2023-12-09:
Is your OMR appliance already upgraded to 6.1 kernel?

yannick2801, 2023-12-11:
yes, 

The router and the VPS are in 6.1 Version.

BR
Yannick

yannick2801, 2023-12-12:
the VPS works with a router in 5.4 but in 6.1, the router cannot launch the script

router 5.4
[image]

router 6.1
[image]

closed 2024-03-17 by github-actions[bot]