Ysurac/openmptcprouter #3059: Firewall and Dashboard Problems

SonicFM, 2023-12-01:
## Expected Behavior
Firewall works as expected.
And OMR does show the status with the Dashboard.

## Current Behavior
Setup OMR as VM in Proxmox and initialy tested with only one Interface.
Later i added 4 more Interfaces via PCIe Passthrough in Proxmox.
Nearly everything works as expected, exept the Status Dashboard and, kind of, the Firewall.

Dashboard Problem:
After adding 4 more Interfaces and renaming all 5, the Dashboard just shows nothing about the status of OMR.
However OMR aggregates everything as expected. It just seems to be a GUI bug?!
See Screenshots.

Firewall Problem:
Added VLAN to OMR via DSA. (Created a bridge and enabled VLAN-Filtering)
So far this works.
However somehow the Firewall dont seem to understand that i want to block my IoT-VLAN to have access to WAN.
Each VLAN does have a Zone. IoT isn't allowed to forward to WAN or VPN or any other Zone. Therefore it should not have Internet access, but strangly it does. I don't understand why.

I suspect that maybe Firewall rules form OMR are interfering. In original OpenWRT 21.02 this case does work.
I would appreciate if anyone could point my error out.
[image]
[image]
[image]

## Specifications

OpenMPTCProuter version: openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c
OpenMPTCProuter VPS version: 0.1028
OpenMPTCProuter VPS OS/Kernel: Ubuntu 20.04.6 LTS (5.4.207-mptcp x86_64)
OpenMPTCProuter VPS provider: PlusServer GmbH
OpenMPTCProuter platform: x86_64 (VM on Proxmox)

Ysurac, 2023-12-04:
Can you take a screenshot of System->OpenMPTCProuter, "Status" page ?
For IoT, I think traffic go directly to the proxy, if you disable the proxy in status page it's working correctly ?
In this case I will need the result of ```iptables-save``` via SSH on the router.

SonicFM, 2023-12-05:
Hi,

the first problem was that I didn't even saw the dashboard, as shown in the pictures in my first post (picture 2 is System -> OpenMPTCP -> Status).
But I managed to figure it out. Actually, the problem seemed to be that if you don't have an interface with IP 192.168.100.1, the dashboard doesn't work.
So you need an interface (even if it is unused and has no zone assigned) with the default IP of OMR.
I created one and now the dashboard works again. Strange behavior.

Regarding the proxy:
I don't see an option to disable a proxy on the status page?!
Do you mean Services -> OMR tracker?

Translated with DeepL.com (free version)
[image]

Ysurac, 2023-12-05:
Dashboard work when LAN IP is modified, can you take a screenshot of Network->Interfaces ?
To disable the proxy, go to System->OpenMPTCProuter, "Advanced settings" checkbox and set proxy to "None".

SonicFM, 2023-12-06:
Hello,

Thank you.
I switched from V2Ray VLESS to none and to Shadowsocks.
Both "none" and "Shadowsocks" allow me to use the firewall rule for IoT as configured.
So far, your assumption that the traffic goes directly to the proxy seems to be correct. Why is this the case?

See attached screenshots of the interfaces.
I am not using the 192.168.100.1 range. I am using 192.168.1.1 for LAN via a DSA switch that I have configured in OpenWRT to use VLANs, of which IoT is one.
But as soon as I delete the interface with the static IP 192.168.100.1, the dashboard looks like my first screenshots.

Thanks for these hints.
I hope that soon it will be possible to use V2Ray with a correctly working firewall.

Please see the ZIP-File, there are two iptables-save Logs. One with Shadowsocks and one with V2Ray VLESS.

Thank you very much for your great work!
[image]
[image]
[iptables-save.zip](https://github.com/Ysurac/openmptcprouter/files/13589667/iptables-save.zip)

SonicFM, 2023-12-06:
Pastebin-Links:
Shadowsocks: https://pastebin.com/ZGxXC36Y
V2Ray: https://pastebin.com/pixF7UKA
PW: KMRiG1taQF

Thank you for clarifying the problem with the dashboard.

SonicFM, 2023-12-09:
Hi,

i double-checked my firewall-config.
It seems i was missing the cruzial "Block IoT Zone to Input (This Device)" to disable access to the Proxy.
As stated by you in https://github.com/Ysurac/openmptcprouter/wiki/Block-client-Internet-access.

I only tried to Block Access to WAN Zone as i would in "normal" OpenWRT.

So i guess this issue goes on me. Now i can also use V2Ray again.
Sorry for the headache!
But issue #3068 sadly remains.

Thnaks for your help!

SonicFM, 2023-12-11:
Closing as it seemed to my fault.

closed 2023-12-11 by SonicFM