Ysurac/openmptcprouter #3065: Not possible to enable support for my network card

GermanAizek, 2023-12-05:
I'm trying to build openmptcprouter with support for my pci-e network card HP NC375T using tutorial on reddit page https://www.reddit.com/r/openwrt/comments/vpqtg2/get_netxen_nic_running/, but I can't find my network device in list. I maybe found only 2 devices from tutorial:

My steps:

[image]

[image]

Tutorial steps:

[image]

[image]

[image]

[image]

closed 2023-12-06 by Ysurac

GermanAizek, 2023-12-06:
@Ysurac thank you very much for hotfix, I found original post tutorial on openwrt forum, only my network pci-e card has name in .config file (highlighted in the screenshot)
[image]

GermanAizek, 2023-12-06:
QLA3XXX I didn't see

Ysurac, 2023-12-06:
If I understand correctly, should be supported by netxen-nic driver.

GermanAizek, 2023-12-06:
I'm going to try to rebuild and test it now.

GermanAizek, 2023-12-24:
> If I understand correctly, should be supported by netxen-nic driver.
@Ysurac, yes indeed this driver, the best support maintainer, but there is a problem when I use two network cards. Only one adds new interfaces, I looked through ifconfig, and other gives driver error `netxen_nic: failed card response code:0x14`

This print function in driver source code: https://github.com/torvalds/linux/blob/3f82f1c3a03694800a4104ca6b6d3282bd4e213d/drivers/net/ethernet/qlogic/netxen/netxen_nic_ctx.c#L62-L66
This code error in define: https://github.com/torvalds/linux/blob/3f82f1c3a03694800a4104ca6b6d3282bd4e213d/drivers/net/ethernet/qlogic/netxen/netxen_nic.h#L755

it's about a bad signature, but I don't understand.

there is no mention this problem on internet, I have different pci-express x16 ports, there should be no problems.

GermanAizek, 2023-12-26:
@Ysurac I checked same behavior on corporate server, took HDD with openmptcp x86_64 image and two my network cards. When connecting one to PCIE on server, everything is fine and eth0-eth7 is detected. I have inner 4 lan ports on server and 4 external from PCIE network card, when connecting an additional PCIE network card for using 12 lan ports, ifconfig shows only eth0-eth3 and when starting openmptcp, he is shows `netxen_nic: failed card response code:0x14` again. Very unclear behavior, whether BIOS settings are to blame or something is missing from linux kernel.

GermanAizek, 2023-12-26:
Users on forums write about `firmware_nextxen` package, I understand correctly that firmware network card needs to be updated only using package, not linux kernel driver, drivers do not know how to do this.

GermanAizek, 2023-12-26:
[image]

It has an NX3031 chipset, here user was prompted not only to enable the drivers, but also enable bnx2x firmware using make menuconfig

https://forum.openwrt.org/t/finding-the-right-kmod-module-for-netxen-nx3031/131263/1

[image]

GermanAizek, 2024-02-17:
@Ysurac, there were no updates to driver code in linux 6.x kernel. Behavior driver is very strange.

Ysurac, 2024-02-17:
Did you finally try on Debian ?

GermanAizek, 2024-02-17:
> Did you finally try on Debian ?

on any linux, OMR image, this problem exists, but on windows, this driver works correctly with multiple network cards. there is no information netxen_nic driver error code. I tried google it, most likely I will have to ask on forums about errors driver code, I do not understand why "Bad Signature 0x14" error is being thrown. Advise me forum linux devices, I will write to them about problem. and it is desirable for us to create separate issue, this issue with addition driver has already been solved.

GermanAizek, 2024-02-19:
> Did you finally try on Debian ?
This problem on any linux and OMR, but older version linux not tested.

GermanAizek, 2024-02-21:
@Ysurac,
> If I understand correctly, should be supported by netxen-nic driver.

I decided to build 6.1 kernel with my drivers on network card and did not find them in make menuconfig. Did u patch and add only in 6.6?

Ysurac, 2024-02-21:
It's directly in "Kernel modules" submenu for all kernels.

GermanAizek, 2024-02-21:
> It's directly in "Kernel modules" submenu for all kernels.

and in "Ethernet Devices" submenu?

Ysurac, 2024-02-21:
no, directly here, at the end.

GermanAizek, 2024-02-21:
> no, directly here, at the end.

indeed, why did you decide to make it so hidden, would you put it in 'Network Devices' as all others.

GermanAizek, 2024-03-09:
> Did you finally try on Debian ?

@Ysurac, I double-checked several network cards connected to pci-e and you won't believe it, they all create interfaces without problems on Debian 11 and Debian 12. here I attach my system logs and detailed information, including the kernel config. i.e. the problem exists in OMR, several network cards refuse to work with netxen_nic firmware. I don't have another pair network cards to make sure that problem is only with this firmware.

[hw_info_debian_11_5.10.txt](https://github.com/Ysurac/openmptcprouter/files/14544352/hw_info_debian_11_5.10.txt)

[hw_info_debian_12_6.5.txt](https://github.com/Ysurac/openmptcprouter/files/14544359/hw_info_debian_12_6.5.txt)

Ysurac, 2024-03-09:
I would need a dmesg from debian 12 if possible to check what it is doing at boot.

Ysurac, 2024-03-09:
Not needed. The firmware is missing: https://packages.debian.org/fr/sid/firmware-netxen
I will add this.

GermanAizek, 2024-03-09:
> Not needed. The firmware is missing: https://packages.debian.org/fr/sid/firmware-netxen I will add this.

Do I need to rebuild with two checkboxes now? Network device and firmware?