Ysurac/openmptcprouter #3066: Network/Interfaces do not working on last snapshot.

vempire-ghost, 2023-12-07:
## Expected Behavior
The Interface settings work when I open them.

## Current Behavior
The page doesn't load and only shows this error.
[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta1-5.4 r0+16862-170d9 snapshot from 06/12
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64
  - Country: Brazil

closed 2023-12-07 by Ysurac

psfabio, 2023-12-12:
Hi Ysurac, this problem still persist on the latest snapshot (12/11/2023)
openmptcprouter v0.60beta1-5.4 r0+16862-170d9e447d
Thanks

Ysurac, 2023-12-12:
I'm not able to reproduce the issue using same snapshot. Can you try to remove cookies and cache for OpenMPTCProuter IP in your browser ?