Ysurac/openmptcprouter #3069: OMR disconnecting to reactivate a VPS that was already connected and functional.

vempire-ghost, 2023-12-10:
The proxy connection between OMR and VPS remains uninterrupted.

## Current Behavior
All connections are interrupted when OMR sets the master VPS.
After being connected and working normally for a few hours, the OMR tracker detected the master VPS as active and reactivated it, killing all connections. I don't know why the OMR tracker reconfigured the master VPS without any changes to it. I currently only have the master VPS configured.

Would it be possible to configure the OMR script so that when it reactivates the VPS, it doesn't disconnect all active connections at the moment? Or make it not reactivate a VPS that is already online and functional?

Here the log when happened.
Dec 10 05:36:59 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 198.27.92.1)
Dec 10 05:36:59 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Dec 10 05:36:59 OpenMPTCProuter daemon.info omr-tracker-xray: Reload xray rules
Dec 10 05:36:59 OpenMPTCProuter user.notice xray: Rules UP
Dec 10 05:36:59 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Dec 10 05:36:59 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Dec 10 05:36:59 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Dec 10 05:37:02 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on vps.ddns.net, server API answer on vps.ddns.net)
Dec 10 05:39:40 OpenMPTCProuter user.info : luci: accepted login on / for root from 192.168.100.2
Dec 10 05:40:02 OpenMPTCProuter daemon.err uhttpd[8231]: error: couldn't find the ModemManager process in the bus
Dec 10 05:43:39 OpenMPTCProuter daemon.info unbound: [21170:0] info: generate keytag query _ta-4f66. NULL IN
Dec 10 05:58:46 OpenMPTCProuter user.notice post-tracking-020-status: New public ip detected for wan1 (eth1): x.x.x.x (previous: )
Dec 10 05:58:46 OpenMPTCProuter user.notice post-tracking-020-status: Reload MPTCP for eth1
Dec 10 13:49:44 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Dec 10 13:49:44 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Dec 10 14:49:53 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Dec 10 14:49:53 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Dec 10 15:13:10 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps up (vps.ddns.net), set it back
Dec 10 15:13:11 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Dec 10 15:13:11 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Dec 10 15:13:11 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Dec 10 15:13:11 OpenMPTCProuter user.notice xray: add rules
Dec 10 15:13:11 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Dec 10 15:13:11 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Dec 10 15:13:11 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Dec 10 15:13:11 OpenMPTCProuter daemon.info xray[32663]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.4 linux/amd64)
Dec 10 15:13:11 OpenMPTCProuter daemon.info xray[32663]: A unified platform for anti-censorship.
Dec 10 15:13:11 OpenMPTCProuter daemon.info xray[32663]: 2023/12/10 15:13:11 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Dec 10 15:13:18 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps up (), set it back
Dec 10 15:13:18 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Dec 10 15:13:18 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Dec 10 15:13:18 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Dec 10 15:13:18 OpenMPTCProuter user.notice xray: add rules
Dec 10 15:13:18 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Dec 10 15:13:18 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Dec 10 15:13:19 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Dec 10 15:13:19 OpenMPTCProuter daemon.info xray[5431]: Xray 1.8.4 (Xray, Penetrates Everything.) Custom (go1.21.4 linux/amd64)
Dec 10 15:13:19 OpenMPTCProuter daemon.info xray[5431]: A unified platform for anti-censorship.
Dec 10 15:13:19 OpenMPTCProuter daemon.info xray[5431]: 2023/12/10 15:13:19 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.60beta1-5.4 r0+16862-170d9 08/12 snapshot
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1029-test
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64

Ysurac, 2023-12-10:
When resolving domain doesn't work (slow DNS or any problem like that), domains was not resolved and original master was unknown. Now, when resolving domains doesn't work, it's ignored.

closed 2023-12-10 by Ysurac

vempire-ghost, 2024-01-04:
Today I had a similar issue again; the OMR was connected and functional. When the master WAN lost internet for a few minutes, everything remained normal until the internet returned. At that moment, the OMR reconnected to the VPS and reset all established connections, dropping everything that was connected. Is there a way to configure the OMR not to force this disconnection when it returns from a lack of connection from the master WAN?
Here the log:
Jan  4 18:09:42 OpenMPTCProuter daemon.info omr-tracker-xray: Server (X.X.X.X) seems down, no answer to ping
Jan  4 18:10:22 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)
Jan  4 18:10:23 OpenMPTCProuter user.notice OMR-Tracker-Server: Use backup server vps.ddnsfree.com (X.X.X.X)
Jan  4 18:10:23 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Jan  4 18:10:23 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Jan  4 18:10:23 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Jan  4 18:10:23 OpenMPTCProuter user.notice xray: add rules
Jan  4 18:10:23 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Jan  4 18:10:23 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Jan  4 18:10:24 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Jan  4 18:10:24 OpenMPTCProuter daemon.info xray[27078]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.21.4 linux/amd64)
Jan  4 18:10:24 OpenMPTCProuter daemon.info xray[27078]: A unified platform for anti-censorship.
Jan  4 18:10:24 OpenMPTCProuter daemon.info xray[27078]: 2024/01/04 18:10:24 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Jan  4 18:10:25 OpenMPTCProuter daemon.notice netifd: Network device 'eth3' link is down
Jan  4 18:10:25 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity loss
Jan  4 18:10:27 OpenMPTCProuter daemon.notice netifd: Network device 'eth3' link is up
Jan  4 18:10:27 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity 
Jan  4 18:10:32 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 74.82.42.42)
Jan  4 18:11:13 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Jan  4 18:11:13 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Jan  4 18:28:49 OpenMPTCProuter user.notice OMR-Tracker-Server: Master server vps down (X.X.X.X)
Jan  4 18:29:00 OpenMPTCProuter daemon.info omr-tracker-xray: xray is down (can't contact via http 77.88.55.77, 1.1.1.1, 74.82.42.42)
Jan  4 18:29:08 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later
Jan  4 18:29:18 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later
Jan  4 18:29:18 OpenMPTCProuter user.notice xray: Rules DOWN
Jan  4 18:29:30 OpenMPTCProuter daemon.info omr-tracker-xray: Server (X.X.X.X) seems down, no answer to ping
Jan  4 18:29:40 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 18:29:56 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later (can't ping server vps_us on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 18:30:18 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 18:30:34 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later (can't ping server vps_us on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 18:31:01 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 18:31:13 OpenMPTCProuter user.notice OMR-Tracker-Server: Use backup server vps.ddnsfree.com (vps.ddnsfree.com)
Jan  4 18:31:13 OpenMPTCProuter user.notice OMR-VPS: Can't get vps_us token, try later (can ping server vps_us on vps.ddnsfree.com, server API answer on vps.ddnsfree.com)
Jan  4 18:31:13 OpenMPTCProuter daemon.info xray: DNS disabled: main_dns
Jan  4 18:31:13 OpenMPTCProuter daemon.info xray: Setting transparent proxy on port: 1897
Jan  4 18:31:13 OpenMPTCProuter daemon.info xray: Transparent proxy mode: default
Jan  4 18:31:13 OpenMPTCProuter user.notice xray: add rules
Jan  4 18:31:13 OpenMPTCProuter user.notice xray: Reload omr-bypass rules
Jan  4 18:31:13 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Jan  4 18:31:13 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Jan  4 18:31:13 OpenMPTCProuter daemon.info xray[25151]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.21.4 linux/amd64)
Jan  4 18:31:13 OpenMPTCProuter daemon.info xray[25151]: A unified platform for anti-censorship.
Jan  4 18:31:13 OpenMPTCProuter daemon.info xray[25151]: 2024/01/04 18:31:13 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json
Jan  4 18:31:20 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 1.1.1.1)
Jan  4 18:31:26 OpenMPTCProuter user.notice OMR-Tracker-Server: Use backup server vps.ddnsfree.com (X.X.X.X)
Jan  4 18:31:39 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on vps.ddnsfree.com, no server API answer on vps.ddnsfree.com)
Jan  4 18:31:43 OpenMPTCProuter user.info : luci: accepted login on / for root from 192.168.100.2
[... cut ...]
Jan  4 18:31:47 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Jan  4 18:31:47 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Jan  4 18:31:47 OpenMPTCProuter daemon.info xray[31753]: Xray 1.8.6 (Xray, Penetrates Everything.) Custom (go1.21.4 linux/amd64)
Jan  4 18:31:47 OpenMPTCProuter daemon.info xray[31753]: A unified platform for anti-censorship.
Jan  4 18:31:47 OpenMPTCProuter daemon.info xray[31753]: 2024/01/04 18:31:47 [Info] infra/conf/serial: Reading config: /var/etc/xray/xray.main.json