Ysurac/openmptcprouter #3072: Port redirection : Keep original IP ?

simicar29, 2023-12-12:
## Expected Behavior
When doing port redirection to access applications hosted internally, keep track of the original IP making the request to allow monitoring and filtering rules

## Current Behavior
All connexions "originate" from the router IP. Is it a matter of configuration of the port redirection ?

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.60beta1-6.1-r0+24041-74e7f8ebbd-x86-64-generic-ext4-combined-efi.img
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1029-test
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: x86_64

Ysurac, 2023-12-12:
By default, masquerade is used. You can change this setting in router firewall configuration. I never tested to disable masquerade, maybe it's also needed to disable it on shorewall side on the VPS.

simicar29, 2023-12-13:
Thank you for you reply. Stupid of me, I had a bad gateway definition in the end server, requiring SNAT to get replies from it. After correction, all is fine, no need to remove masquerading.
Kindest regards.

closed 2023-12-13 by simicar29