Ysurac/openmptcprouter #3073: Does OMR have Self Healing WAN?

ccmks, 2023-12-12:
## Expected Behavior
I run simulation where we are intentionally simulate the connection go down on the WAN alternatively, then the real-time traffic will keep going on without any issue.

## Current Behavior
This is not the case, we are running call test on VoIP phone that is running RTP using UDP, when trying to simulate the WAN down between multiple WAN alternatively, the phone call keep dropping, not be able to continue the call when the WAN is switching over to another WAN.

I thought the idea of OMR is having self-healing WAN, otherwise, this is same as regular failover router.

## Specifications

  - OpenMPTCProuter version: 0.60beta-1-6.1 (snapshot)
  - OpenMPTCProuter VPS version: 0.1029-test 6.1.0-13-amd64
  - OpenMPTCProuter VPS provider: local VPS in Indonesia
  - OpenMPTCProuter platform: x86_64 Hyper-V platform

ccmks, 2023-12-12:
I am using X2RAY Trojan, whether using it as UDP, or OpenVPN TCP as UDP (I can't do other VPN if using X2RAY, as the other VPN won't connect)

Ysurac, 2023-12-12:
The main idea is still to aggregate bandwidth, but yes it also support switching between WAN without disconnection.
UDP is still a bit special, because it use VPN (or some proxy).
Some Proxy/VPN are working better in some case. OpenVPN TCP should work correctly with failover.
What do you have in Status->System log when a wan disconnect ? This WAN is set as master or not ?

ccmks, 2023-12-12:
I am using shadowsocks now with OpenVPN TCP, if I am using other than proxy, the UDP won't work at all

I have 4 WAN, but I just started to disconnect the master WAN

Dec 13 01:51:36 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) 00:15:5d:02:96:07 
Dec 13 01:51:36 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.100.144 00:15:5d:02:96:07 
Dec 13 01:51:36 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) 00:15:5d:02:96:07 
Dec 13 01:51:36 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.100.144 00:15:5d:02:96:07 
Dec 13 01:51:36 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth0) 00:15:5d:02:96:07 
Dec 13 01:51:36 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth0) 192.168.100.144 00:15:5d:02:96:07 
Dec 13 01:51:36 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth0) 192.168.100.144 00:15:5d:02:96:07 
Dec 13 01:51:36 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth0) 192.168.100.144 00:15:5d:02:96:07 Border-Rou
Dec 12 17:51:39 OpenMPTCProuter user.notice omr-tracker: Launching...
Dec 12 17:51:51 OpenMPTCProuter user.notice omr-tracker: Launched
Dec 12 17:51:52 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Tunnel up : Replace default route by 10.255.252.9 dev tun0
Dec 12 17:51:52 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 1.0.0.1)
Dec 12 17:51:54 OpenMPTCProuter daemon.err openvpn(omr)[20101]: Connection reset, restarting [0]
Dec 12 17:51:54 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1400 0 10.255.252.10 10.255.252.9 init
Dec 12 17:51:54 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: net_addr_ptp_v4_del: 10.255.252.10 dev tun0
Dec 12 17:51:54 OpenMPTCProuter daemon.notice ttyd[4964]: [2023/12/13 01:51:54:4179] N: rops_handle_POLLIN_netlink: DELADDR
Dec 12 17:51:54 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Dec 12 17:51:54 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Dec 12 17:51:54 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Dec 12 17:51:54 OpenMPTCProuter daemon.notice ttyd[4964]: [2023/12/13 01:51:54:4249] N: rops_handle_POLLIN_netlink: DELADDR
Dec 12 17:51:54 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: /usr/libexec/openvpn-hotplug down omr tun0 1400 0 10.255.252.10 10.255.252.9 init
Dec 12 17:51:54 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: SIGUSR1[soft,connection-reset] received, process restarting
Dec 12 17:51:54 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Dec 12 17:51:55 OpenMPTCProuter daemon.warn openvpn(omr)[20101]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Dec 12 17:51:55 OpenMPTCProuter daemon.warn openvpn(omr)[20101]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Dec 12 17:51:55 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: TCP/UDP: Preserving recently used remote address: [AF_INET]103.161.185.4:65301
Dec 12 17:51:55 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: Attempting to establish TCP connection with [AF_INET]103.161.185.4:65301
Dec 12 17:51:55 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: TCP connection established with [AF_INET]103.161.185.4:65301
Dec 12 17:51:55 OpenMPTCProuter daemon.warn openvpn(omr)[20101]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)
Dec 12 17:51:55 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: TCPv4_CLIENT link local: (not bound)
Dec 12 17:51:55 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: TCPv4_CLIENT link remote: [AF_INET]103.161.185.4:65301
Dec 12 17:51:55 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: [server] Peer Connection Initiated with [AF_INET]103.161.185.4:65301
Dec 12 17:51:56 OpenMPTCProuter daemon.warn openvpn(omr)[20101]: sitnl_route_save: rtnl: can't get ifname for index 0: No such device or address (errno=6)
Dec 12 17:51:56 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: TUN/TAP device tun0 opened
Dec 12 17:51:56 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: net_iface_mtu_set: mtu 1400 for tun0
Dec 12 17:51:56 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: net_iface_up: set tun0 up
Dec 12 17:51:56 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: net_addr_ptp_v4_add: 10.255.252.10 peer 10.255.252.9 dev tun0
Dec 12 17:51:56 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Dec 12 17:51:56 OpenMPTCProuter daemon.notice openvpn(omr)[20101]: /usr/libexec/openvpn-hotplug up omr tun0 1400 0 10.255.252.10 10.255.252.9 init
Dec 12 17:51:56 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Dec 12 17:51:56 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Dec 12 17:51:56 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Dec 12 17:51:56 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Dec 12 17:51:57 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Dec 12 17:51:57 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Dec 12 17:51:58 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
[... cut ...]
Dec 12 17:57:20 OpenMPTCProuter daemon.warn openvpn(omr)[5680]: sitnl_send: rtnl: generic error (-17): File exists
Dec 12 17:57:20 OpenMPTCProuter daemon.notice openvpn(omr)[5680]: Initialization Sequence Completed
Dec 12 17:57:23 OpenMPTCProuter user.notice post-tracking-010-services: Set firewall on server vps
Dec 12 17:57:25 OpenMPTCProuter user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Dec 12 17:57:27 OpenMPTCProuter user.notice post-tracking-001-post-tracking: omrvpn down because gateway down
Dec 12 17:57:34 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later

ccmks, 2023-12-13:
In my scenario, I was able to successfully get the seamless failover done without much noticeable packet drop, by using Shadowsocks + Glorytun UDP. It seemed to be working well in my case, I was able to transition between 4 different WAN alternatively and the call still keep going on. Hopefully, this will still be the case for next several days.

I am using the 6.1 kernel, but I agree with you that @Ysurac  needs to add more scheduler option to have redundant, and more congestion control option.

closed 2024-03-17 by github-actions[bot]