Ysurac/openmptcprouter #3074: Adding the modified scheduler to the list of schedulers #1

rezapoorzare1, 2023-12-13:
Hello there,

I hope you are doing well.

I have two questions:

    How can I see the available MPTCP scheduler in Linux?

    How can a newly created scheduler be added to the schedulers lists and set by using:
    sudo sysctl net.mptcp.mptcp_scheduler

Is there a procedure like:
cc=$(< /proc/sys/net/ipv4/tcp_allowed_congestion_control)
cc="$mptcp_wvegas"
sudo sysctl net.ipv4.tcp_allowed_congestion_control=$cc
Which is for congestion control?

Thanks a lot

Ysurac, 2023-12-13:
For MPTCP out-of-tree version (kernel < 5.10): http://multipath-tcp.org/pmwiki.php/Users/ConfigureMPTCP
For MPTCP in tree kernel version, you have only default scheduler available for now. Some are also available as BPF in https://github.com/multipath-tcp/mptcp_net-next

rezapoorzare1, 2023-12-13:
Thank you so much for the response. 

I modified a scheduler and have a new one. I make the module and add it to the kernel. However, I cannot set it as a scheduler by using the command:

sysctl net.mptcp.mptcp_scheduler

as mentioned in:

http://multipath-tcp.org/pmwiki.php/Users/ConfigureMPTCP
I think I should add the new one to the list of available schedulers, but I do not know how. Is it possible for you to help me on this?

Thanks

closed 2024-03-17 by github-actions[bot]