Ysurac/openmptcprouter #3075: [0.60 BETA 1] WAN port of the NanoPI R4S not beeing detected / working

tompointexe, 2023-12-13:
## Expected Behavior
When plugin an ethernet cable to the wan port of the NanoPI R4S, it should light up, and be assignable to a wan interface, or as a macvlal

## Current Behavior
The interface is not detected, wich makes the lan interface as the default lan, and no lan port is beeing affected

## Possible Solution
For now, i'm using an USB Adapter to look around for a possible fix

## Steps to Reproduce the Problem

  1. Install the latest 0.60 beta image
  2. wait for it to boot
  3. be unable to access the configuration interface unless a USB Ethernet adapter is plugged in

## Context (Environment)
Simple NanoPI R4S, Realtek USB 2.5gig used for testing and accessing the lan part
8 gig SD Card, build downloaded on 13/12/2023 off the download site

## Specifications

  - OpenMPTCProuter version:  0.60 Beta 1
  - OpenMPTCProuter VPS version: Not Related
  - OpenMPTCProuter VPS provider: Own (selfhosted behing Orange)
  - OpenMPTCProuter platform: NanoPI R4S
  - Country: France

N3EOP, 2023-12-25:
I had the same problem.  Plugging into the LAN port lit the WAN indicator on the top of the R4S, not the LAN indicator.  The WAN port simply did nothing.  I was totally unable to log into the system.

0.60b1-5.4 works as expected.

tompointexe, 2024-01-05:
Fix confirmed !
Thanks

closed 2024-01-05 by tompointexe