Ysurac/openmptcprouter #3077: Image build error

longxi0412, 2023-12-14:
Image compile system: Ubuntu 20.04 LST

#sudo apt update
#sudo apt upgrade
#sudo apt-get install busybox curl rsync build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler wget tar
#git clone https://github.com/Ysurac/openmptcprouter.git
#cd openmptcprouter
#git checkout master

## Expected Behavior

## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2023-12-14:
Try in "x86_64/<kernel_number>/source" subdirectory to do a ```make -j1 V=s package/feeds/openmptcprouter/glorytun-udp/compile``` to see the problem.
For information, as you can see in https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/.github/workflows/main.yml all builds are done under Ubuntu (22.04 I think).

longxi0412, 2023-12-14:
Thanks for your reply @Ysurac  I will check it follow your guidance

longxi0412, 2023-12-14:
An error still occurs,Do I need to start over again? I feel like it is caused by a bad network.Due to character limit.I only post part of the text(beginning and the end.)

longxi@longxi:~/openmptcprouter$ ls
5.15  build.sh           CODE_OF_CONDUCT.md  config-bpi-r1   config-espressobin  config-r2s  config-r7800  config-rpi4      config-wrt3200acm  config-x86_64    deploy_rsa.enc  patches    x86_64
5.4   CLA-entity.md      common              config-bpi-r2   config-p2w_r619ac   config-r4s  config-rpi2   config-rutx      config-wrt32x      CONTRIBUTING.md  feeds           README.md
6.1   CLA-individual.md  config              config-bpi-r64  config-qnap-301w    config-r5s  config-rpi3   config-ubnt-erx  config-x86         contributors     LICENSE         sign.sh
longxi@longxi:~/openmptcprouter$ cd x86_64
longxi@longxi:~/openmptcprouter/x86_64$ ls
5.4
longxi@longxi:~/openmptcprouter/x86_64$ cd 5.4
longxi@longxi:~/openmptcprouter/x86_64/5.4$ ls
source
longxi@longxi:~/openmptcprouter/x86_64/5.4$ cd source
longxi@longxi:~/openmptcprouter/x86_64/5.4/source$ ls
BSDmakefile  config     COPYING  feeds       feeds.conf.default  key-build      key-build.ucert         LICENSES  Makefile  README.md  rules.mk.orig  staging_dir  tmp        tools
build_dir    Config.in  dl       feeds.conf  include             key-build.pub  key-build.ucert.revoke  logs      package   rules.mk   scripts        target       toolchain
longxi@longxi:~/openmptcprouter/x86_64/5.4/source$ make -j1 V=s package/feeds/openmptcprouter/glorytun-udp/compile
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-cdrom', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-ledtrig-audio', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'dnsmasq_full_ipset', which does not exist
WARNING: Makefile 'package/feeds/openmptcprouter/openmptcprouter-full/Makefile' has a dependency on 'kmod-crypto-lib-blake2s', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/wmt/Makefile' has a dependency on 'kmod-mt6625l-wlan-gen2', which does not exist
make[1]: Entering directory '/home/longxi/openmptcprouter/x86_64/5.4/source'
make[2]: Entering directory '/home/longxi/openmptcprouter/x86_64/5.4/source/package/libs/toolchain'
touch /home/longxi/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/toolchain/.prepared_b19ebd9bcfc8344d63844ed7b7ac0bd3_18f1e190c5d53547fed41a3eaa76e9e9_check
mkdir -p /home/longxi/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/toolchain
touch /home/longxi/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/toolchain/.prepared_b19ebd9bcfc8344d63844ed7b7ac0bd3_18f1e190c5d53547fed41a3eaa76e9e9
rm -f /home/longxi/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/toolchain/.configured_*
rm -f /home/longxi/openmptcprouter/x86_64/5.4/source/staging_dir/target-x86_64_musl/stamp/.toolchain_installed

........................................................

mkdir -p /home/longxi/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel
# nothing to do

touch /home/longxi/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel.installed
mkdir -p /home/longxi/openmptcprouter/x86_64/5.4/source/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/longxi/openmptcprouter/x86_64/5.4/source/tmp/.root-copy.flock -c 'cp -fpR /home/longxi/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/longxi/openmptcprouter/x86_64/5.4/source/staging_dir/target-x86_64_musl/root-x86/'
touch /home/longxi/openmptcprouter/x86_64/5.4/source/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
sed: can't read /home/longxi/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/modules.builtin: No such file or directory
Packaged contents of /home/longxi/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kernel into /home/longxi/openmptcprouter/x86_64/5.4/source/bin/targets/x86/64/packages/kernel_5.4.238-1-66bdfc6d6758b62c21ed83aed473eeb2_x86_64.ipk
grep: /home/longxi/openmptcprouter/x86_64/5.4/source/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/modules.builtin: No such file or directory
[... cut ...]
make[2]: *** [modules/block.mk:22: /home/longxi/openmptcprouter/x86_64/5.4/source/bin/targets/x86/64/packages/kmod-aoe_5.4.238-1_x86_64.ipk] Error 1
make[2]: Leaving directory '/home/longxi/openmptcprouter/x86_64/5.4/source/package/kernel/linux'
time: package/kernel/linux/compile#1.82#0.09#2.01
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/home/longxi/openmptcprouter/x86_64/5.4/source'
make: *** [/home/longxi/openmptcprouter/x86_64/5.4/source/include/toplevel.mk:230: package/feeds/openmptcprouter/glorytun-udp/compile] Error 2

Ysurac, 2023-12-14:
It's seems that the kernel is not compiled here.
Try a ```make -j1 V=s``` in same directory. And check available disk space.

closed 2024-03-18 by github-actions[bot]