Ysurac/openmptcprouter #3078: getpeername: Transport endpoint is not connected

hehh2001, 2023-12-15:
VPS----------SERVER

OS: Ubuntu 20.04.6 LTS x86_64 
Host: VMware Virtual Platform None
Kernel: 5.4.207-mptcp 
Uptime: 1 day 
Packages: 829 (dpkg), 4 (snap) 
 Shell: bash 5.0.17
CPU: Intel Xeon E7- 4830 (1) @ 2.128GHz
Memory: 337MiB / 1995MiB 

##############################
router

Model: Raspberry Pi Compute Module 4 Rev 1.0

Architecture: ARMv8 Processor rev 3

Firmware version: openmptcprouter v0.59.1-5.4 r0+16594-ce92de8c8c

##########################
Dec 15 15:23:42 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:23:44 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:23:46 zzb ss-server[914]: remote recv: Connection reset by peer
Dec 15 15:23:51 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:23:51 zzb ss-server[914]: remote recv: Connection reset by peer
Dec 15 15:23:53 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:23:57 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:23:59 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:01 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:07 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:10 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:10 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:11 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:11 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:11 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:12 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:13 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:15 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:15 zzb ss-server[914]: getpeername: Transport endpoint is not connected
Dec 15 15:24:15 zzb ss-server[914]: getpeername: Transport endpoint is not connected

hehh2001, 2023-12-15:
The aggregation router is working normally, the network is normal, and the vps reports an error! ! ! !
Thanks for your great project, I’d like to buy you a coffee

Ysurac, 2023-12-15:
This mean a website can't be reached. Maybe an IP used in OMR-Tracker to check proxy connectivity is not answering anymore.

xzjq, 2023-12-26:
Getting quite a few of these too, even after `1.0.0.1` was removed from OMR-Tracker.

What is this looking for? `198.41.212.162` has a `curl` exit code of `0`, for example, but the cloudflare page exhibits an error 1003 (Direct IP access not allowed).

Most of the others on the list give "unsatisfactory" results such as HTTP 301 (moved permanently) type hypertext, though `curl` still has an exit code of `0`. I guess it all depends on what the tracker is looking for... any result at all that doesn't explicitly result in an process error code from the process on fetch?

Ysurac, 2023-12-28:
@xzjq it's not directly related as the question is on 0.59.1 release using kernel 5.4
Test are done with ```curl -I -w %{http_code} <ip>```, if the answer is not "000" then it's ok.

closed 2024-04-01 by github-actions[bot]