Ysurac/openmptcprouter #3079: OOM error on VPS

xzjq, 2023-12-16:
## Expected Behavior
VPS with 3 GB of RAM, dedicated to OMR (doing nothing else) should not run of out of memory.

## Current Behavior
Just lost internet connectivity, putatively related to OOM-kill, running XRay Shadowsocks 2022.

`Dec 15 16:21:04 vps kernel: oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/system.slice/xray.service,task=xray,pid=1047436,uid=0`
`Dec 15 16:21:04 vps kernel: Out of memory: Killed process 1047436 (xray) total-vm:3945912kB, anon-rss:2219436kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:5436kB oom_score_adj:0`

Checking `htop` reveals that `/usr/bin/xray` is consuming over 2,000 MB of RAM.

OMR + VPS is being used for home internet use. Browsing & client VPNs; no P2P or "odd" usage.

## Specifications

  - OpenMPTCProuter version: Snapshot, `[v0.60beta1-6.1 r0+24041-74e7f8ebbd]`
  - OpenMPTCProuter VPS version: Snapshot, `0.1029-test 6.1.0-15-amd64`
  - OpenMPTCProuter platform:  Debian 12 / x86_64

Ysurac, 2023-12-16:
I will update xray, the shadowsocks part, based on SagerNet sing, was updated recently and this may fix the issue...

Ysurac, 2023-12-18:
XRay is now updated on VPS side, you need to run VPS script again to update it.

closed 2024-03-22 by github-actions[bot]