Ysurac/openmptcprouter #3082: OMR interface repeatedly cycling multipath

xzjq, 2023-12-18:
## Expected Behavior
Interfaces will stay up.

## Current Behavior
After trying multiple versions of the kernel/6.1-based OMR & multiple VPSs (beta & snapshot), each exhibits an issue where one or more WAN interfaces will be repeatedly cycled, with MPTCP reloaded. 

_This issue is not present using the same local OMR hardware/network infrastructure/WANs/VPS providers on 5.4-based OMR._

Below is an example of it cycling all three WANs (thousands of these messages over hours):

Dec 18 21:49:04 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.300 switched to on (from off)
Dec 18 21:49:11 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.400
Dec 18 21:49:11 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.500
Dec 18 21:49:11 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.400 switched to on (from off)
Dec 18 21:49:12 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.300
Dec 18 21:49:12 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.500 switched to on (from off)
Dec 18 21:49:13 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.300 switched to on (from off)
Dec 18 21:49:19 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.400
Dec 18 21:49:19 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.400 switched to on (from off)
Dec 18 21:49:20 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.500
Dec 18 21:49:20 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.500 switched to on (from off)
Dec 18 21:49:20 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.300
Dec 18 21:49:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.300 switched to on (from off)
Dec 18 21:49:27 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.400
Dec 18 21:49:27 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.400 switched to on (from off)
Dec 18 21:49:27 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.500
Dec 18 21:49:28 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.500 switched to on (from off)
Dec 18 21:49:28 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.300
Dec 18 21:49:29 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.300 switched to on (from off)
Dec 18 21:49:40 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.500
Dec 18 21:49:41 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.500 switched to on (from off)
Dec 18 21:49:41 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.300
Dec 18 21:49:42 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.300 switched to on (from off)
Dec 18 21:49:44 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.400
Dec 18 21:49:45 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.400 switched to on (from off)
Dec 18 21:49:48 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.500
Dec 18 21:49:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.500 switched to on (from off)
Dec 18 21:49:49 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.300
Dec 18 21:49:50 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.300 switched to on (from off)
Dec 18 21:49:53 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.400
Dec 18 21:49:54 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.400 switched to on (from off)
Dec 18 21:49:56 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.500
Dec 18 21:49:57 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.500 switched to on (from off)
Dec 18 21:49:57 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.300
Dec 18 21:49:58 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.300 switched to on (from off)
Dec 18 21:50:01 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.400
Dec 18 21:50:02 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.400 switched to on (from off)
Dec 18 21:50:05 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.500
Dec 18 21:50:05 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.500 switched to on (from off)

## Possible Solution
Clicking Save & Apply (or rebooting) will temporarily reset the system to functional.

## Steps to Reproduce the Problem
It will eventually degrade to this condition, usually within 8 hours.

## Context (Environment)
This prevents aggregation, as `post-tracking` cycles each interface every ~10 seconds.

## Specifications

  - OpenMPTCProuter version: `v0.60beta1-6.1 r0+24041-74e7f8ebbd`
  - OpenMPTCProuter VPS version: `0.1029-test 6.1.0-15-amd64`

[image]

Ysurac, 2023-12-19:
Duplicate of #3068 
As indicated in latest comment, did you update to latest version of the 15th december  that should give more log ?

xzjq, 2023-12-19:
Thanks. The other bug was getting to be multifaceted (and I wasn't the one who opened that one), so I forked this off independently.

Per your comment in that ticket, I had updated the VPS and it was a brand new version from the server-test script.

However, the OMR image name had not changed (`openmptcprouter-v0.60beta1-6.1-r0+24041-74e7f8ebbd-x86-64-generic-ext4-combined.vdi`), so I did not download a another copy. After you replied, I noticed the date was different on the file on the server, so I downloaded it and found the shasum is different as well. I did not anticipate that file name did not contain the actual version. I am now segregating the OMR downloads in timestamped directories so I can keep track.

Here is the equivalent output from a brand new OMR VM:
<hr>

Dec 19 11:54:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.400
Dec 19 11:54:21 OpenMPTCProuter user.notice MPTCP: Disabling MPTCP on interface bonding_masters not found in enabled multipath list
Dec 19 11:54:21 OpenMPTCProuter user.notice MPTCP: Disabling MPTCP on interface eth0 not found in enabled multipath list
Dec 19 11:54:21 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.400 (wan2) switched to on (from off)
Dec 19 11:54:22 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Reload MPTCP config for eth0.300
Dec 19 11:54:22 OpenMPTCProuter user.notice MPTCP: Disabling MPTCP on interface bonding_masters not found in enabled multipath list
Dec 19 11:54:22 OpenMPTCProuter user.notice MPTCP: Disabling MPTCP on interface eth0 not found in enabled multipath list
Dec 19 11:54:22 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Multipath eth0.300 (wan1) switched to on (from off)

Ysurac, 2023-12-20:
I've found the issue. I test the fix and this will be added to git in a few hours.

closed 2023-12-20 by Ysurac