Ysurac/openmptcprouter #3086: OMR Connectivity Unstable if more than hundreds user connected

ccmks, 2023-12-20:
## Expected Behavior
The OMR has 4 WAN that will have stable connectivity although have more than 100 users connected behind it

## Current Behavior
I am running mini ISP, where serving around 25 houses, that total user easily above 100 users. However, when OMR is got introduced, the connection become unstable and the ping is constantly high (above 500 ms), however if only several user using OMR (below 10), the connection is OK, the ping is constantly at 50 ms. This affects the entire user experience if they are connected behind OMR

This behavior is also exhibited on VPS site, when the OMR is connected for all users, ping to 8.8.8.8 is constantly above 300 ms, the connection become unstable

Does OMR doesn't support that many user? Especially on ISP environment?

## Specifications

  - OpenMPTCProuter version: 0.60beta1-6.1
  - OpenMPTCProuter VPS version: 0.1029-test 6.1.0-16-amd64
  - OpenMPTCProuter VPS provider: local VPS in Indonesia
  - OpenMPTCProuter platform: x86-64 Hyper-V

Ysurac, 2023-12-20:
You mean ping from client to internet via a single OMR router ? ping use a VPN between between router and VPS, what is the VPN used ?
For the second part, you mean ping from the VPS to 8.8.8.8 ? This may explain why you get a high ping for client. On VPS ping shouldn't increase. Check VPS CPU usage (with for example, ```apt-get install htop && htop```) and if it's low, try another VPS if possible.

ccmks, 2023-12-20:
> You mean ping from client to internet via a single OMR router ? ping use a VPN between between router and VPS, what is the VPN used ?

Ping to 8.8.8.8 from client to internet using OMR, and also from VPS ping to 8.8.8.8 using OMR

I forgot to mention that HTOP doesn't display CPU being full during the usage, also the connection is not even utilize full bandwidth, even just running 10 Mbps, the OMR already exhibit unstable connection

I am using Shadowsocks, V2RAY, X2RAY with VPN Glorytun TCP & UDP, OpenVPN TCP, they are all still display same issue

Ysurac, 2023-12-21:
What is the result of ```awk  </proc/net/tcp 'BEGIN{t=0};{if ($4 == "01") {t++;}};END{print t}'``` on VPS and router ?

docblocxxx, 2023-12-25:
> ## Expected Behavior
> The OMR has 4 WAN that will have stable connectivity although have more than 100 users connected behind it
>
> ## Current Behavior
> I am running mini ISP, where serving around 25 [...]

in my case, openvpn omr gave a large delay, I had to give it up.
Almost any VPN in beta significantly increases latency.

shadowsocks or rust, + working glorytun udp give me minimal latency.

ccmks, 2023-12-26:
I ended up move to different VPS and it looks like the problem disappear

I am using shadowsock + OpenVPN TCP

However, I am still seeing occasionally, it will display "Contact server admin script (no result)"

ccmks, 2023-12-26:
> What is the result of `awk </proc/net/tcp 'BEGIN{t=0};{if ($4 == "01") {t++;}};END{print t}'` on VPS and router ?

It is only 301

closed 2024-03-31 by github-actions[bot]