Ysurac/openmptcprouter #3090: What is limit on count WAN connections?

GermanAizek, 2023-12-25:
@Ysurac Hello, I am interested in question why openmptcprouter is limited to 4 wan connections, what explains this problem?
There are no restrictions themselves programmatically, but I read issues and everyone is talking about real limit 4 connections.

Ysurac, 2023-12-26:
There is currently a limit of 8 addresses.

GermanAizek, 2023-12-26:
> There is currently a limit of 8 addresses.

Is it possible to manually fixed and increase to 12, is this due to the unstable network operations in kernel?

Ysurac, 2023-12-26:
It's due to MPTCP internal Path Manager limit of subflows numbers.
The number is already increased in current 5.4 kernel release to 16 subflows for x86 and x86_64.
It's not yet increased for 6.1 snapshot release.

GermanAizek, 2023-12-26:
> It's due to MPTCP internal Path Manager limit of subflows numbers. The number is already increased in current 5.4 kernel release to 16 subflows for x86 and x86_64. It's not yet increased for 6.1 snapshot release.

thanks for detailed answer, I built 6.6 kernel so there should be a maximum 16 wan connections, but I already see that you have version 6.7 available, what are major differences between them?

Ysurac, 2023-12-26:
Even for 6.6 or 6.7 it's still limited to 8 subflows.
The current 6.7 is for test only, based on MPTCP_net-next branch. I'm using it to test BPF.
I've not yet checked changelog between 6.6 and 6.7 as 6.7 is still a Release Candidate.

GermanAizek, 2023-12-26:
> Even for 6.6 or 6.7 it's still limited to 8 subflows.

this is bad, but I hope something will change in this direction.

closed 2023-12-26 by GermanAizek